Sloeber/arduino-eclipse-plugin #1578: Errors After Upgrading ESP32 Library

emintz, 2023-08-05:
Note: I'd like to lend a hand with this and need help getting started.

I had to reinstall after I upgraded from Ubuntu 20.4 to 22.04. Because the upgrade broke the Arduino IDE, Eclipse, and Sloeber, I think the upgrade messed things up.  However, when I reinstalled Sloeber, I upgraded my runtime from 2.0.10 to 2.0.11, my builds remained stuck on 2.0.10 even after I changed the platform directory to 2.0.11.

Working around this is hard. I was able to fix some projects by changing "2.0.10" in their .project file while the project was closed, opening the project, selecting the 2.0.11 platform path, reattaching the libraries, cleaning the project, deleting the Release subdirectory, and building. Sometimes I needed to do a bit more fiddling. When I got absolutely wedged, I would delete and recreate the project, leaving the source files on disk, configure the platform path, board, and port, and rebuilding. 

I also had to reinstall some libraries; I probably lost them when I deleted the prior Sloeber install. I expect to reinstall libraries after I upgrade, so this is no big deal.

I'd like to contribute to Sloeber, so if you could help me get started with this, I'll get right on it.

Thanks so much for all you do. I think that Sloeber rocks.

I look forward to hearing from you.

-- Eric

jantje, 2023-08-05:
Hi Eric
I have a hard time understanding what you are trying to say.
When you say:
>I upgraded my runtime from 2.0.10 to 2.0.11, my builds remained stuck on 2.0.10 even after I changed the platform directory to 2.0.11.

Do you mean you installed a new version of the ESP platform? Are you aware you can have multiple platforms at the same time?
In the project properties->sloeber(in the latests version ; before it was arduino) you select the platform and that one includes the platform version.

I hope this helps.

emintz, 2023-08-06:
Hi, Jantje,

Yes, that's right. When I reinstalled Sloeber (no complaint there!), I upgraded my platform from 2.0.10 to 2.0.11. Over the decades, I've learned to keep libraries up to date. You can ask me how I know! ;-).

Unfortunately, my projects remained stuck on 2.0.10 after I switched the platform. I had to edit their .platform files to get to the current platform version.

If you don't mind, I'd like to try fixing this, so that Sloeber automatically updates the .project file when I change platforms. Do you think this to be a good idea? If so, can you give me a bit of background (especially links to documentation) to help me get started?

If you'd rather that I not bother, I'll look for other opportunities to contribute.

Thank you again. I'm your and Sloeber's biggest fan.

-- Eric

jantje, 2023-08-06:
>Unfortunately, my projects remained stuck on 2.0.10 after I switched the platform. I had to edit their .platform files to get to the current platform version.

Because Sloeber supports multiple versions of a platform you can/need to specify the platform you want to use in the project properties->Sloeber. If you do not do so Sloeber will keep on using the platform it was last set to use.

>If you don't mind, I'd like to try fixing this, so that Sloeber automatically updates the .project file when I change platforms.

Nice to hear you want to help; but.
I still do not understand the problem you are facing. and "automatically updates the .project" makes me think you want to change the linked folders (core libraries) but that is only part of the changes needed.
As to"links to documentation"; everything is in this repository. That is apart from the Arduino framework documentation; CDT documentation; Eclipse documentation; java documentation,....

As you may know Sloeber has a API itself. This API should allow you to do things like "Change all the project configurations that reference platform X version Y to platform Z versin T".
Have a look at the regression tests here:
https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.tests/src/io/sloeber/core/RegressionTest.java

closed 2024-02-20 by jantje