Sloeber/arduino-eclipse-plugin #1580: Sloeber menu not visible on first launch

Alex-Kent, 2023-08-08:
When launching Sloeber for the first time (when built from source) the **Sloeber** menu does not appear in the menu bar:

[image]

Once the following menu item is selected the **Sloeber** menu becomes visible:

    Window → Perspective → Open Perspective → C/C++

[image]

Alex-Kent, 2023-08-08:
I tried adding `org.eclipse.ui/defaultPerspectiveId=org.eclipse.cdt.ui.CPerspective` to `io.sloeber.product/plugin_customization.ini` followed by a build using `mvn clean verify -Plinux64,latest,NOSDK -DskipTests=true` but it didn't fix things.

The test was run under a test user (`test`) with `SLOEBER_HOME` set to `/home/test` and no `~/.eclipse` or `~/Documents/sloeber-workspace` directories.

jantje, 2023-08-08:
The automagic selecting of the Sloeber perspective is something I never got to work.
There are several places that it says the Sloeber perspective should be selected by default; however it is not.
To me this is just one of these things that are supposedly simple but are not.

Alex-Kent, 2023-08-08:
Not a big deal, just something I noticed.  If I'm reading the docs correctly the fix I tried should have worked...

jantje, 2023-08-09:
I wish it worked. It is most problematic for people not being used to the eclipse perspective as they do not see any Sloeber stuff

jantje, 2023-11-27:
FYI: the Sloeber menu is only visible in C/C++ perspective or Sloeber perspective. That is "as designed".

closed 2024-02-20 by jantje