Sloeber/arduino-eclipse-plugin #1582: ESP32 Core 2.0.11 Broke Sloeber

gfvalvo, 2023-08-28:
I just added ESP32 Core 2.0.11 to my "Platforms and Boards". I'm running this version of the Sloeber Nightly:
[https://github.com/Sloeber/arduino-eclipse-plugin/actions/runs/3659904477](url)

This basic code fails to compile:
```
#include "Arduino.h"

//The setup function is called once at startup of the sketch
void setup()
{
// Add your initialization code here
}

// The loop function is called in an endless loop
void loop()
{
//Add your repeated code here
}
```

The error message from the console is below. Please Advise. Thanks.
```
08:52:59 **** Build of configuration Release for project ESP32_2_0_11 ****
"C:\\SloeberNightly127\\arduinoPlugin\\tools\\make\\make" all 
C:/SloeberNightly127/arduinoPlugin/tools/make/make --no-print-directory pre-build
cmd /c if exist "C:\SloeberNightly127_Workspace\ESP32_2_0_11\partitions.csv" COPY /y "C:\SloeberNightly127_Workspace\ESP32_2_0_11\partitions.csv" "C:\SloeberNightly127_Workspace\ESP32_2_0_11\Release\partitions.csv"
cmd /c if not exist "C:\SloeberNightly127_Workspace\ESP32_2_0_11\Release\partitions.csv" if exist "C:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11\variants\esp32\partitions.csv" COPY "C:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11\variants\esp32\partitions.csv" "C:\SloeberNightly127_Workspace\ESP32_2_0_11\Release\partitions.csv"
cmd /c if not exist "C:\SloeberNightly127_Workspace\ESP32_2_0_11\Release\partitions.csv" COPY "C:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11\tools\partitions\default.csv" "C:\SloeberNightly127_Workspace\ESP32_2_0_11\Release\partitions.csv"
cmd /c IF EXIST "C:\SloeberNightly127_Workspace\ESP32_2_0_11\bootloader.bin" ( COPY /y "C:\SloeberNightly127_Workspace\ESP32_2_0_11\bootloader.bin" "C:\SloeberNightly127_Workspace\ESP32_2_0_11\Release\ESP32_2_0_11.bootloader.bin" ) ELSE ( IF EXIST "C:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11\variants\esp32\bootloader.bin" ( COPY "C:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11\variants\esp32\bootloader.bin" "C:\SloeberNightly127_Workspace\ESP32_2_0_11\Release\ESP32_2_0_11.bootloader.bin" ) ELSE ( "C:\SloeberNightly127\arduinoPlugin\packages\esp32\tools\esptool_py\4.5.1/esptool.exe" --chip esp32 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o "C:\SloeberNightly127_Workspace\ESP32_2_0_11\Release\ESP32_2_0_11.bootloader.bin" "C:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11\tools\sdk\esp32\bin\bootloader_qio_80m.elf" ) )
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
cmd /c if exist "C:\SloeberNightly127_Workspace\ESP32_2_0_11\build_opt.h" COPY /y "C:\SloeberNightly127_Workspace\ESP32_2_0_11\build_opt.h" "C:\SloeberNightly127_Workspace\ESP32_2_0_11\Release\build_opt.h"
cmd /c if not exist "C:\SloeberNightly127_Workspace\ESP32_2_0_11\Release\build_opt.h" type nul > "C:\SloeberNightly127_Workspace\ESP32_2_0_11\Release\build_opt.h"
' '
C:/SloeberNightly127/arduinoPlugin/tools/make/make --no-print-directory main-build
'Building file: ..\ESP32_2_0_11.cpp'
'Starting C++ compile'
"C:\SloeberNightly127\arduinoPlugin\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0/bin/xtensa-esp32-elf-g++" -DHAVE_CONFIG_H "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE -DIDF_VER="v4.4.5" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS   "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/newlib/platform_include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/freertos/include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/freertos/include/esp_additions/freertos" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/freertos/port/xtensa/include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/freertos/include/esp_additions" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/esp_hw_support/include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/esp_hw_support/include/soc" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/esp_hw_support/include/soc/esp32" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/esp_hw_support/port/esp32" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/esp_hw_support/port/esp32/private_include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/heap/include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/log/include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/lwip/include/apps" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/lwip/include/apps/sntp" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/lwip/lwip/src/include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/lwip/port/esp32/include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/lwip/port/esp32/include/arch" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/soc/include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/soc/esp32" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/soc/esp32/include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/hal/esp32/include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/hal/include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/hal/platform_port/include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/esp_rom/include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/esp_rom/include/esp32" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/esp_rom/esp32" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/esp_common/include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/esp_system/include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/esp_system/port/soc" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/esp_system/port/public_compat" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/esp32/include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/xtensa/include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/xtensa/esp32/include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/driver/include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/driver/esp32/include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/esp_pm/include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/esp_ringbuf/include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/efuse/include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/efuse/esp32/include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/vfs/include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/esp_wifi/include" "-IC:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11/tools/ [... cut ...] F"ESP32_2_0_11.cpp.d" -MT"ESP32_2_0_11.cpp.o" -D__IN_ECLIPSE__=1 "..\ESP32_2_0_11.cpp" -o "ESP32_2_0_11.cpp.o"
xtensa-esp32-elf-g++: error: C:\SloeberNightly127_Workspace\ESP32_2_0_11\Release/file_opts: No such file or directory
make[1]: *** [subdir.mk:20: ESP32_2_0_11.cpp.o] Error 1
make: *** [makefile:88: all] Error 2
"C:/SloeberNightly127/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

08:53:01 Build Failed. 2 errors, 0 warnings. (took 1s.739ms)
```

jantje, 2023-08-28:
see https://github.com/Sloeber/arduino-eclipse-plugin/discussions/1576

gfvalvo, 2023-08-28:
> see #1576
Do the two lines need to go anywhere specific in platform.txt?:
C:\SloeberNightly127\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11\platform.txt

I put them at the end:
recipe.hooks.prebuild.7.pattern=bash -c "[ -f "{build.path}"/file_opts ] || touch "{build.path}"/file_opts"
recipe.hooks.prebuild.7.pattern.windows=cmd /c if not exist "{build.path}\file_opts" type nul > "{build.path}\file_opts"

Then deleted the project, restarted Sloeber and made a new project. No joy. Same error.

jantje, 2023-08-28:
That is not the correct process
https://github.com/Sloeber/arduino-eclipse-plugin/discussions/1533

jantje, 2023-08-29:
From my point of view this is a problem in ESP32 2.0.11 or some new(?) feature in the arduino IDE I am not aware of.

ArminJo, 2023-10-15:
It is a new arduino feature used by esp32 since arduino tries to build the core separately.

recipe.hooks.prebuild.**set_core_build_flag**.pattern.windows=cmd /c type nul > "{file_opts.path}"

It would help if it could be included just as a plain **recipe.hooks.prebuild** entry into platform.sloeber.txt as done in #1576

jantje, 2023-10-15:
Sloeber implements the Arduino framework which -amongst others- specifies the platform 
https://arduino.github.io/arduino-cli/0.34/platform-specification/#pre-and-post-build-hooks-since-arduino-ide-165
I do not find set_core_build_flag on the platform specification page.
If the esp32 people are serious about this option they should create a PR to Arduino to update the platform specification page.
IMHO  it is a better idea to use the standard pre/post commands which perfectly fit the job.

jantje, 2023-12-01:
Other workaround use version 3

closed 2023-12-01 by jantje

ArminJo, 2023-12-01:
But ESP version 3 is incompatible with a lot of libraries, they changed too much in this version https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html.

jantje, 2023-12-01:
no problem mate.
I think you handled this issue very well 👍

gfvalvo, 2023-12-01:
> Other workaround use version 3

Where does "Version 3" come from ???
The latest version is see offered in the Arduino IDE is 2.0.14!!!

[image]

ArminJo, 2023-12-01:
https://github.com/espressif/arduino-esp32/releases

jantje, 2023-12-01:
>OK, but unusable until Arduino IDE and/or Sloeber offer that as a choice in their Board Managers.

As it is a release there should be a json to install it somewhere.

erdemuncuoglu, 2023-12-01:
> As it is a release there should be a json to install it somewhere.

v3 is a pre-release and still in development. And as documented [here](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#installing-using-arduino-ide) development JSON is

https://espressif.github.io/arduino-esp32/package_esp32_dev_index.json

GitHubLionel, 2024-01-13:
Hi,

I have the same error with esp2.0.11 installed (no other choices in Platforms selection) :
`xtensa-esp32-elf-g++: error: L:\Sloeber\workspace\ESP32_Test\Release/file_opts: No such file or directory
make[1]: *** [subdir.mk:20: sloeber.ino.cpp.o] Error 1
make: *** [makefile:102: all] Error 2`
I try to add the two lines at the end of platform.txt but it doesn't work.

What I missing ?

Note that in the file sloeber.ino.cpp generated by Sloeber, it seem's that __IN_ECLIPSE__ is not defined because this define part is greyed.

Regards,
Lionel

jantje, 2024-01-13:
Have you used the correct process?
https://github.com/Sloeber/arduino-eclipse-plugin/issues/1582#issuecomment-1696367722

GitHubLionel, 2024-01-13:
I think that yes but I am a little confuse with the .txt file since I have no .txt file in my test project. 
So I have no .txt file to delete despite the "How to modify a .txt file #1533" guide say.
I have just copied the 2 lines in the platform.txt, clean the project, delete the release directory, close Sloeber and restart it.
Shall I add this 2 lines in "platform.sloeber.txt" ?
Regards
Lionel

jantje, 2024-01-13:
Your confusion is confusing me.
You stated:
>I try to add the two lines at the end of platform.txt but it doesn't work.

Which means you did find the platform.txt. 

>So I have no .txt file to delete despite the "How to modify a .txt file https://github.com/Sloeber/arduino-eclipse-plugin/discussions/1533" guide say.

The platform.sloeber.txt is in the same folder as the platform.txt so you should be able to delete it.

>Shall I add this 2 lines in "platform.sloeber.txt"

Did you or did you not find the sloeber.platform.txt file?

GitHubLionel, 2024-01-14:
Hi,
Sorry, I thought that there where other specific .txt file for each project like I read in the guide.
So if I summarize, I must do : 
1- add the 2 lines to platform.txt
2- delete sloeber.platform.txt
That what say gfvalvo https://github.com/Sloeber/arduino-eclipse-plugin/discussions/1533 and it work fine.
Thanks for your response,
Lionel

gfvalvo, 2024-01-25:
Is this bug fixed in Release 4.4.2 from last November?

jantje, 2024-01-25:
> Is this bug fixed in Release 4.4.2 from last November?

As you can see from the label this is a package issue not a sloeber issue
[image]
So there is no fix in Sloeber and there never will be. Next esp32 package release is supposed to fix this issue
There is a workaround (one time fix).

gfvalvo, 2024-01-25:
> As you can see from the label this is a package issue not a sloeber issue

I guess I don't understand the nuance there since the ESP32 package works without any problems in the Arduino IDE. It requires no "workaround". And Sleober is supposed to be better than Arduino IDE.

jantje, 2024-01-25:
>I guess I don't understand the nuance there since the ESP32 package works without any problems in the Arduino IDE.
I think this comment explains is quite well:
https://github.com/Sloeber/arduino-eclipse-plugin/issues/1582#issuecomment-1836310745

>And Sloeber is supposed to be better than Arduino IDE.

I for one never said Sloeber is better in each and every way than Arduino IDE.
I always stated that the Arduino IDE focuses on beginners (but has moved on somewhat) where Sloeber focuses on the experienced programmer providing a full complex IDE.
There is no doubt Arduino IDE is the gold standard for the Arduino framework. 
But if the Arduino team state they will not support something (and even consider changing their code to make it impossible) they are saying "This may no longer work in the next Arduino IDE version".
As a consequence ESP32 changed their package.