Sloeber/arduino-eclipse-plugin #1585: Failed to install Arduino AVR Boards

aatanas, 2023-10-22:
---

**Describe your environment**   
eclipse.buildId=4.29.0.20230907-1200
java.version=17.0.8.1
java.vendor=Eclipse Adoptium
BootLoader constants: OS=linux, ARCH=x86_64, WS=gtk, NL=en_US
Framework arguments:  -product org.eclipse.epp.package.cpp.product
Command-line arguments:  -os linux -ws gtk -arch x86_64 -product org.eclipse.epp.package.cpp.product
  

**Describe the bug**   
Plugin installed, on restart i got this error:

Failed to install Arduino AVR Boards
Failed to extract tar.bz2. "/home/l/Documents/eclipse/arduinoPlugin/downloads/avr-1.8.6.tar.bz2".
java.io.IOException: Unexpected end of stream
	at org.apache.commons.compress.compressors.bzip2.BZip2CompressorInputStream.bsR(BZip2CompressorInputStream.java:139)
	at org.apache.commons.compress.compressors.bzip2.BZip2CompressorInputStream.getAndMoveToFrontDecode(BZip2CompressorInputStream.java:508)
	at org.apache.commons.compress.compressors.bzip2.BZip2CompressorInputStream.initBlock(BZip2CompressorInputStream.java:636)
	at org.apache.commons.compress.compressors.bzip2.BZip2CompressorInputStream.setupNoRandPartA(BZip2CompressorInputStream.java:884)
	at org.apache.commons.compress.compressors.bzip2.BZip2CompressorInputStream.setupNoRandPartB(BZip2CompressorInputStream.java:891)
	at org.apache.commons.compress.compressors.bzip2.BZip2CompressorInputStream.read0(BZip2CompressorInputStream.java:724)
	at org.apache.commons.compress.compressors.bzip2.BZip2CompressorInputStream.read(BZip2CompressorInputStream.java:695)
	at org.apache.commons.compress.archivers.tar.TarArchiveInputStream.read(TarArchiveInputStream.java:688)
	at java.base/java.io.InputStream.read(InputStream.java:218)
	at io.sloeber.core.tools.PackageManager.copyStreamToFile(PackageManager.java:359)
	at io.sloeber.core.tools.PackageManager.extract(PackageManager.java:253)
	at io.sloeber.core.tools.PackageManager.processArchive(PackageManager.java:84)
	at io.sloeber.core.tools.PackageManager.downloadAndInstall(PackageManager.java:74)
	at io.sloeber.core.tools.PackageManager.downloadAndInstall(PackageManager.java:465)
	at io.sloeber.core.api.BoardsManager.install(BoardsManager.java:230)
	at io.sloeber.core.api.BoardsManager.startup_Pluging(BoardsManager.java:704)
	at io.sloeber.core.Activator$2.run(Activator.java:254)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)

jantje, 2023-10-22:
delete /home/l/Documents/eclipse/arduinoPlugin/downloads/avr-1.8.6.tar.bz2 and try again.
Probably the download failed.

aatanas, 2023-10-23:
I uninstalled, deleted the whole arudinoPlugin folder, and installed again, that fixed it.
But now i have another problem i cant select anything in "Platforms and Boards" I click on checkboxes but nothing happens.
Also I cannot unselect Arduino AVR Boards, that was selected by default. Everything on that menu page is completely unresponsive, i think it could be related to this error, i have a few of those in the log.

eclipse.buildId=4.29.0.20230907-1200
java.version=17.0.8.1
java.vendor=Eclipse Adoptium
BootLoader constants: OS=linux, ARCH=x86_64, WS=gtk, NL=en_US
Framework arguments:  -product org.eclipse.epp.package.cpp.product
Command-line arguments:  -os linux -ws gtk -arch x86_64 -product org.eclipse.epp.package.cpp.product

org.eclipse.ui
Error
Mon Oct 23 03:15:40 CEST 2023
Unhandled event loop exception

java.lang.IllegalArgumentException: Argument not valid
	at org.eclipse.swt.SWT.error(SWT.java:4899)
	at org.eclipse.swt.SWT.error(SWT.java:4833)
	at org.eclipse.swt.SWT.error(SWT.java:4804)
	at org.eclipse.swt.graphics.Image.init(Image.java:1219)
	at org.eclipse.swt.graphics.Image.<init>(Image.java:209)
	at org.eclipse.equinox.internal.p2.ui.discovery.util.GradientCanvas.onPaint(GradientCanvas.java:168)
	at org.eclipse.equinox.internal.p2.ui.discovery.util.GradientCanvas.lambda$0(GradientCanvas.java:81)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:5855)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1529)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1555)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1538)
	at org.eclipse.swt.widgets.Control.gtk_draw(Control.java:3928)
	at org.eclipse.swt.widgets.Scrollable.gtk_draw(Scrollable.java:365)
	at org.eclipse.swt.widgets.Composite.gtk_draw(Composite.java:506)
	at org.eclipse.swt.widgets.Canvas.gtk_draw(Canvas.java:174)
	at org.eclipse.swt.widgets.Widget.windowProc(Widget.java:2482)
	at org.eclipse.swt.widgets.Control.windowProc(Control.java:6884)
	at org.eclipse.swt.widgets.Display.windowProc(Display.java:6163)
	at org.eclipse.swt.internal.gtk3.GTK3.gtk_main_do_event(Native Method)
	at org.eclipse.swt.widgets.Display.eventProc(Display.java:1597)
	at org.eclipse.swt.internal.gtk3.GTK3.gtk_main_iteration_do(Native Method)
	at org.eclipse.swt.browser.WebKit$Webkit2AsyncToSync.execAsyncAndWaitForReturn(WebKit.java:1282)
	at org.eclipse.swt.browser.WebKit$Webkit2AsyncToSync.runjavascript(WebKit.java:1030)
	at org.eclipse.swt.browser.WebKit$Webkit2AsyncToSync.evaluate(WebKit.java:1000)
	at org.eclipse.swt.browser.WebKit.evaluate(WebKit.java:1299)
	at org.eclipse.swt.browser.WebKit.close(WebKit.java:865)
	at org.eclipse.swt.browser.WebKit.onDispose(WebKit.java:1792)
	at org.eclipse.swt.browser.WebKit.lambda$4(WebKit.java:768)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:5855)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1529)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1555)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1534)
	at org.eclipse.swt.widgets.Widget.release(Widget.java:1337)
	at org.eclipse.swt.widgets.Control.release(Control.java:4804)
	at org.eclipse.swt.widgets.Widget.dispose(Widget.java:543)
	at org.eclipse.epp.internal.mpc.ui.wizards.AbstractMarketplaceDiscoveryItem.computeBrowserAvailable(AbstractMarketplaceDiscoveryItem.java:521)
	at org.eclipse.epp.internal.mpc.ui.wizards.AbstractMarketplaceDiscoveryItem.createInfoLink(AbstractMarketplaceDiscoveryItem.java:394)
	at org.eclipse.epp.internal.mpc.ui.wizards.AbstractMarketplaceDiscoveryItem.createDescription(AbstractMarketplaceDiscoveryItem.java:341)
	at org.eclipse.epp.internal.mpc.ui.wizards.AbstractMarketplaceDiscoveryItem.createContent(AbstractMarketplaceDiscoveryItem.java:241)
	at org.eclipse.epp.internal.mpc.ui.wizards.AbstractMarketplaceDiscoveryItem.createContent(AbstractMarketplaceDiscoveryItem.java:203)
	at org.eclipse.epp.internal.mpc.ui.wizards.DiscoveryItem.createContent(DiscoveryItem.java:123)
	at org.eclipse.epp.internal.mpc.ui.wizards.AbstractMarketplaceDiscoveryItem.<init>(AbstractMarketplaceDiscoveryItem.java:168)
	at org.eclipse.epp.internal.mpc.ui.wizards.DiscoveryItem.<init>(DiscoveryItem.java:102)
	at org.eclipse.epp.internal.mpc.ui.wizards.MarketplaceViewer.createDiscoveryItem(MarketplaceViewer.java:501)
	at org.eclipse.epp.internal.mpc.ui.wizards.MarketplaceViewer.doCreateViewerItem(MarketplaceViewer.java:435)
	at org.eclipse.equinox.internal.p2.ui.discovery.wizards.CatalogViewer$3.doCreateItem(CatalogViewer.java:344)
	at org.eclipse.equinox.internal.p2.ui.discovery.util.ControlListViewer.createNewItem(ControlListViewer.java:244)
	at org.eclipse.equinox.internal.p2.ui.discovery.util.ControlListViewer.refreshAll(ControlListViewer.java:486)
	at org.eclipse.equinox.internal.p2.ui.discovery.util.ControlListViewer.inputChanged(ControlListViewer.java:358)
	at org.eclipse.jface.viewers.ContentViewer.setInput(ContentViewer.java:282)
	at org.eclipse.jface.viewers.StructuredViewer.setInput(StructuredViewer.java:1647)
[... 36 lines cut ...]
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1529)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:5065)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:4517)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1155)
[... cut ...]
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:568)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:651)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:588)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1459)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1432)

jantje, 2023-10-23:
I see you installed Sloeber in an existing eclipse. ("eclipse.buildId=4.29.0.20230907-1200" is newer than latest Sloeber build)
I also see no references to Sloeber nor to CDT in the stack trace.
This leads me to believe there is a plugin incompatibility somewhere.
I remember hearing that there are incompatibilities between CDT and the "latest eclipse" after the latest Sloeber release. If CDT had incompatibilities it is extremely likely there are also incompatibilities between Sloeber and the latest eclipse.

These potential incompatibilities are the very reason we bothered to make a product that contain a set of compatible plugins and is also the reason why we do not support adding the plugin to an existing eclipse (we simply do not have the resources).

If you really want the greatest and latest I propose to build Sloeber yourself. It ain't hard.
See the readme for instructions
https://github.com/Sloeber/arduino-eclipse-plugin#build-from-source

aatanas, 2023-10-23:
Ok now this is very weird, i just installed standalone sloeber IDE separately and i have the same problem, clicking on anything in the "Platform and Boards" does not work, except the "Hide 3rd party json files" checkbox that only changes the view. I cannot select or unselect anything. Only one error in the log:

eclipse.buildId=unknown
java.version=17.0.3
java.vendor=Eclipse Adoptium
BootLoader constants: OS=linux, ARCH=x86_64, WS=gtk, NL=en_US
Framework arguments:  -perspective io.sloeber.application.perspective
Command-line arguments:  -os linux -ws gtk -arch x86_64 -perspective io.sloeber.application.perspective

org.eclipse.core.resources
Error
Mon Oct 23 16:30:43 CEST 2023
Workspace was not properly initialized or has already shutdown.

java.lang.Exception: Workspace was not properly initialized or has already shutdown.
	at org.eclipse.core.internal.resources.ResourceException.provideStackTrace(ResourceException.java:42)
	at org.eclipse.core.internal.resources.ResourceException.<init>(ResourceException.java:38)
	at org.eclipse.core.internal.resources.Workspace.getWorkManager(Workspace.java:1817)
	at org.eclipse.core.internal.resources.Workspace.endOperation(Workspace.java:1481)
	at org.eclipse.core.internal.events.AutoBuildJob$AutoBuildOffJob.run(AutoBuildJob.java:333)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)

aatanas, 2023-10-23:
I did the build from source, same thing, i installed on another machine and there i figured out it was working all the time, you just have to open the checkbox tree all the way down and select a specific version 🙃

jantje, 2023-10-23:
Lol
It flashed through my mind that maybe that was the case but I rejected it. Sorry for that.
I guess it is not obvious as a newbee as Arduino IDE does not support installing multiple versions.

aatanas, 2023-10-24:
Oh I used sloeber in the past, back then it was the only proper arudino setup i could find, it helped a lot of with my work on freertos library among other things, its just that i never had the need to mess with that menu before. Anyway, I installed esp8266 and now i have another problem. On create new project i changed the "Platform folder" from esp to avr and now when i try it doesn't fill the defaults for the selected board anymore 

[image]

aatanas, 2023-10-24:
Ok, sloeber standalone version works, same thing did happen but it cleared on its own
Btw i did clean reinstall of eclipse and i can confirm https://github.com/Sloeber/arduino-eclipse-plugin/issues/1587#issue-1957676714

jantje, 2023-12-01:
As #1587 is fixed I close this issue. Please reopen if you think the issue is not fixed.

closed 2023-12-01 by jantje