Sloeber/arduino-eclipse-plugin #1587: Eclipse Plugin will not install

GreyCon, 2023-10-23:
When attempting to install Sloeber 4 from the Eclipse (2023-09) marketplace, it fails:
Cannot complete the install because one or more required items could not be found.
  Software being installed: Sloeber, the Eclipse Arduino IDE (plugin version) 4.4.1.202207231052 (io.sloeber.feature.feature.group 4.4.1.202207231052)
  Missing requirement: Arduino eclipse plugin 4.4.1.202207231052 (io.sloeber.core 4.4.1.202207231052) requires 'osgi.bundle; org.slf4j.api 1.7.2' but it could not be found
  Cannot satisfy dependency:
    From: Sloeber, the Eclipse Arduino IDE (plugin version) 4.4.1.202207231052 (io.sloeber.feature.feature.group 4.4.1.202207231052)
    To: org.eclipse.equinox.p2.iu; io.sloeber.core [4.4.1.202207231052,4.4.1.202207231052]

jantje, 2023-10-24:
I think I have to change Sloeber marketplace settings that it does not install with the latest eclipse.

GreyCon, 2023-10-24:
OK cool, thanks. I'll try to install it with an older Eclipse later
tonight. Meantime, as reported elsewhere, I have it working under Win10
now, so very happy indeed ;-)

jantje, 2023-10-24:
Due to mac security changes there are issues with mac install.
Surest way to get Sloeber on mac is to build it yourself (like in the good old days 😉 )

GreyCon, 2023-10-24:
I tried my friend, I'm a dev, so I figured I'd give it a go, so I could at
least add more logging around the Spawn command, where the issue appears to
be. But the build failed, and I had already spent too many hours in this
little rabbit hole :-) I needed to get back to the job at hand and get my
ESP8266 flashed. So windows for now. I'll try to get the build working
again on OSX Ventura when I can. But I know (As you do) that this is like
playing whack-a-mole, with so many changing pieces of open source software.
If I have any success I will, of course, post back here.

And thanks again for a very useful tool.

jantje, 2023-10-24:
I think I know what is going on.
I've been working on a change for Sloeber on another branch for over a year now (and still no end in sight 😒) . I had to do some changes due to eclipse changes; I did not yet port these to the main thread.
As main builds with main and the latest CDT/Eclipse version it is probably broken now.

FrankHovis, 2023-10-30:
Still not working, any way around this issue?

> Cannot complete the install because one or more required items could not be found.
  Software being installed: Sloeber, the Eclipse Arduino IDE (plugin version) 4.4.1.202207231052 (io.sloeber.feature.feature.group 4.4.1.202207231052)
  Missing requirement: Arduino eclipse plugin 4.4.1.202207231052 (io.sloeber.core 4.4.1.202207231052) requires 'osgi.bundle; org.slf4j.api 1.7.2' but it could not be found
  Cannot satisfy dependency:
    From: Sloeber, the Eclipse Arduino IDE (plugin version) 4.4.1.202207231052 (io.sloeber.feature.feature.group 4.4.1.202207231052)
    To: org.eclipse.equinox.p2.iu; io.sloeber.core [4.4.1.202207231052,4.4.1.202207231052]

jantje, 2023-10-30:
Workaround:
Use older eclipse install or install the Sloeber product (which includes an older eclipse)
http://eclipse.baeyens.it/stable.php?OS=Windows

jantje, 2023-11-27:
@GreyCon 
#1593 should fix the build issue. Note you need java 17 and a recent maven (more recent that maven that comes with the latest ubuntu 😞  ) to build