Sloeber/arduino-eclipse-plugin #1588: Sloeber 4.4 missing build.opt on Windows

GreyCon, 2023-10-23:
_Before continuing._   
_Click on the preview button so the text is more readable and the links work_

**Before creating a bug report check for existing issues**  
Make sure you are on the latest and greatest Sloeber version. We do not support older version.  
Here is a list of [known limitations](https://github.com/Sloeber/arduino-eclipse-plugin/issues?q=is%3Aissue++label%3A%22status%3A+known+limitation%22+)   
Here is a list of [common issues ](https://github.com/Sloeber/arduino-eclipse-plugin/issues?q=is%3Aissue+label%3A%22Hot+issue%22+)   
Here is the list of all [open issues](https://github.com/Sloeber/arduino-eclipse-plugin/issues?q=is%3Aissue+is%3Aopen)  
Here is a list of issues that have been [fixed but are not yet available in a release](https://github.com/Sloeber/arduino-eclipse-plugin/issues?q=label%3A%22status%3A+fixed+in+nightly%22+)  

---

**Describe your environment**   
Do you use the Sloeber product or plugin?   Sloeber Product for Windows
Sloeber version number  4.4.1
OS: [e.g. windows 10]   Windows 10
The url of the json that you used to download the platform  
A image of the project properties arduino showing the board settings   

**Describe the bug**   
Installed on Windows. No other Eclipse installed. Created a new ESP8266 project, which fails to compile, complaining about a missing build.opt file

**To Reproduce**   
Steps to reproduce the behavior:   
Install 4.4.1 on windows
Add support for ESP8266, 3.12
Create new empty Arduino project for ESP8266, and build.

provide the name of the board and the platform version you are using.  
Provide the json url and a screenshot of project properties->arduino  

**Expected behavior**   
A clear and concise description of what you expected to happen.  

**Data to back up your claim**   
If applicable, add screenshots, console output and/or logs to help explain your problem.   

**Additional context**   
Add any other context about the problem here.

jantje, 2023-10-23:
Cool you found a solution yourself 👍

GreyCon, 2023-10-25:
Hey!
  all going well here but I've a peculiar problem with an ESP32 project
that's killing me. It shows 19 validation errors - it's not finding the
include files in "Arduino.h" (Such as string.h, etc) The code compiles
fine, but the little error markers are annoying. A separate ESP8266 project
does not have this issue. Do you have any clue where the validator picks up
its include path? I can't find it on the project properites. Obviously this
isn't important enough to raise as an issue, but I'd dearly love to fix it.