Sloeber/arduino-eclipse-plugin #1589: sloeber 4.4.1 missing environment variables, build fails

Emmanuel-DUFOUR, 2023-11-07:
-

**Describe your environment** 
 Sloeber product
Sloeber version number  
4.4.1
windows 7
**arduino settings as required**
[image]

**Describe the bug**   
the compiler path references A.RUNTIME.TOOLS.ARM-NONE-EABI-GCC.PATH, and that variable isnt set and can't be set (all env variables are overwritten on build)
[image]
as a consequence of not having the entire compiler path, the build fails:
[image]

Emmanuel-DUFOUR, 2023-11-07:
I was wrong, the environment variables can be fixed provided that "apply" is clicked before "apply and close"
[image]

Emmanuel-DUFOUR, 2023-11-07:
but I still have no idea why these variables got deleted in the 1st place (upong rebuilding, in 2 different projects, using different boards)

jantje, 2023-11-07:
Note this is an old sloeber.
Normally you do not need to modify the environment variables directly. The newest Sloeber does not allow to modify them.
Selecting the project properties->arduino (change something) and select apply close will overwrite all environment variables.
If you need to change A. environment variables It is better to change the *.txt files and force the changes through the caches as described here https://github.com/Sloeber/arduino-eclipse-plugin/discussions/1533

jantje, 2023-12-01:
I just now noticed that you claim to be running 4.4.1 which is strange as 4.4.1 does not use the A.XX environment variables anymore.
Sloeber has code to clean up these A.XX variables. So I guess Sloeber works as intended.

Emmanuel-DUFOUR, 2023-12-01:
I had 2 installs of sloeber, a recent one and an old one, somehow when launching, both would show the splash sceen showing the version of the most recent one. I therefore gave you the wrong version number.

Emmanuel-DUFOUR, 2023-12-03:
splash shows 4.4.1 but plugin is 4.3.1 (that is probably because I installed both, Im surprised that they share data though, perhaps a common windows registry key ?):
[image]

jantje, 2023-12-03:
>perhaps a common windows registry key ?):

no such thing in Sloeber.

The only things I can think of are
1) the product has been upgraded without the plugins
2) use of bundle pools (default install option when using oomph)
3) manual action.

Any ways 4.3.1 is 5 years old so No longer supported.
The A.xx approach has been replaced in V4.4