Sloeber/arduino-eclipse-plugin #1590: Data type declarations not copied from sketch to Sloeber.ino

emintz, 2023-11-09:
Data type (e.g. a struct) declarations are not copied from a sketch to Slober.ino, which causes the build to fail.  For example, given this:

```
#include "Arduino.h"

struct DeclaredStruct {
  uint32_t value1;
};
typedef struct {
  uint32_t value2;
} TypedefedStruct;
DeclaredStruct declared_struct;
TypedefedStruct typedefed_struct;

static void use_declared_struct(DeclaredStruct *value) {}
static void use_typedefed_struct(TypedefedStruct *value) {}
void setup() {}
void loop() {}
```
which compiles correctly, Sloeber generates:

```
#ifdef __IN_ECLIPSE__
//This is a automatic generated file
//Please do not modify this file
//If you touch this file your change will be overwritten during the next build
//This file has been generated on 2023-11-09 12:37:18

#include "Arduino.h"
#include "Arduino.h"

static void use_declared_struct(DeclaredStruct *value) ;
static void use_typedefed_struct(TypedefedStruct *value) ;
void setup() ;
void loop() ;

#include "typedef_bug.ino"

#endif
```
whose compilation fails because DeclaredStruct and TypedefedStruct are not defined. The solution is to move their declarations into a header and include it in the sketch.

This might not be worth fixing, but mentioning it in the documentation could be helpful.

I absolutely love Slober and recommend it to my friends. Thanks for all your hard work. It shows.

jantje, 2023-11-09:
This is known issue #775

emintz, 2023-11-09:
Sorry I missed that. Thank you for your kind help.

closed 2023-11-09 by jantje