Sloeber/arduino-eclipse-plugin #1591: uploadWithProgrammerWithoutBuild

theduskin, 2023-11-22:
I'll try my best to describe 
Win10, sloeber 441 complete download.
project setup:
[image]
the uploaded code gets corrupted. it works partially but with errors.
avrdude mentions: 5710 bytes of flash verified

when i manually upload with both, avrdude 5.8(winavr latest) or 7.2(github latest), flashsize is 5650 bytes
and the Tiny works as expected.
selected programmer for both is stk500v2. physically a DIAMEX AVR-ISP.
I'm not sure where the difference to DIAMEX USB-ISP is, if any.

the sloeber command line:
-pattiny85 -cstk500v2 -PCOM11 "-Uflash:w:D:\Eclipse_Workspaces\sloeber44\Tiny_NTC_Neo\Release/Tiny_NTC_Neo.hex:i"
manually:
-C D:\WinAVR-20100110\avrdude-v7.2-windows-x64\avrdude.conf -p t85 -P com11 -c stk500v2  -U flash:w:D:\Eclipse_Workspaces\sloeber44\Tiny_Volt_Neo\Release\Tiny_Volt_Neo.hex:a 

that's what I noticed, please let me know what further information is needed.
Thanks for reading and Sloeber itself btw
Chris

jantje, 2023-11-22:
1) It is not clear to me whether you want to do a upload or a program
2) can you provide the console content from the failed command
3) can you provide the full command that works.

theduskin, 2023-11-23:
I want to program via ISP without Bootloader
[image]

this is the Sloeber console content / the failed command:

>      C:\Users\chris>"E:\Tools\ecliSloeber441\arduinoPlugin\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude" "-CE:\Tools\ecliSloeber441\arduinoPlugin\packages\ATTinyCore\hardware\avr\1.5.2/avrdude.conf" -pattiny85 -cstk500v2 -PCOM11 "-Uflash:w:D:\Eclipse_Workspaces\sloeber44\Tiny_NTC_Neo\Release/Tiny_NTC_Neo.hex:a"
> 
>     avrdude: AVR device initialized and ready to accept instructions
> 
>     Reading | ################################################## | 100% 0.02s
> 
>     avrdude: Device signature = 0x1e930b (probably t85)
>     avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
>     To disable this feature, specify the -D option.
>     avrdude: erasing chip
>     avrdude: reading input file "D:\Eclipse_Workspaces\sloeber44\Tiny_NTC_Neo\Release/Tiny_NTC_Neo.hex"
>     avrdude: input file D:\Eclipse_Workspaces\sloeber44\Tiny_NTC_Neo\Release/Tiny_NTC_Neo.hex auto detected as Intel Hex
>     avrdude: writing flash (5710 bytes):
> 
>     Writing | ################################################## | 100% 3.33s
> 
>     avrdude: 5710 bytes of flash written
>     avrdude: verifying flash memory against D:\Eclipse_Workspaces\sloeber44\Tiny_NTC_Neo\Release/Tiny_NTC_Neo.hex:
>     avrdude: load data flash data from input file D:\Eclipse_Workspaces\sloeber44\Tiny_NTC_Neo\Release/Tiny_NTC_Neo.hex:
>     avrdude: input file D:\Eclipse_Workspaces\sloeber44\Tiny_NTC_Neo\Release/Tiny_NTC_Neo.hex auto detected as Intel Hex
>     avrdude: input file D:\Eclipse_Workspaces\sloeber44\Tiny_NTC_Neo\Release/Tiny_NTC_Neo.hex contains 5710 bytes
>     avrdude: reading on-chip flash data:
> 
>     Reading | ################################################## | 100% 2.98s
> 
>     avrdude: verifying ...
>     avrdude: 5710 bytes of flash verified
> 
>     avrdude: safemode: Fuses OK (E:FF, H:D4, L:E2)
> 
>     avrdude done. Thank you.

this is a working example via cmd konsole

>    C:\Users\chris>D:\WinAVR-20100110\bin\avrdude.exe -C D:\WinAVR-20100110\bin\avrdude.conf -p t85 -P com11 -c stk500v2 -U flash:w:D:\Eclipse_Workspaces\sloeber44\Tiny_Volt_Neo\Release\Tiny_Volt_Neo.hex:a
> 
>     avrdude.exe: AVR device initialized and ready to accept instructions
> 
>     Reading | ################################################## | 100% 0.03s
> 
>     avrdude.exe: Device signature = 0x1e930b
>     avrdude.exe: NOTE: FLASH memory has been specified, an erase cycle will be performed
>     To disable this feature, specify the -D option.
>     avrdude.exe: erasing chip
>     avrdude.exe: reading input file "D:\Eclipse_Workspaces\sloeber44\Tiny_Volt_Neo\Release\Tiny_Volt_Neo.hex"
>     avrdude.exe: input file D:\Eclipse_Workspaces\sloeber44\Tiny_Volt_Neo\Release\Tiny_Volt_Neo.hex auto detected as Intel Hex
>     avrdude.exe: writing flash (5650 bytes):
> 
>     Writing | ################################################## | 100% 3.16s
> 
>     avrdude.exe: 5650 bytes of flash written
>     avrdude.exe: verifying flash memory against D:\Eclipse_Workspaces\sloeber44\Tiny_Volt_Neo\Release\Tiny_Volt_Neo.hex:
>     avrdude.exe: load data flash data from input file D:\Eclipse_Workspaces\sloeber44\Tiny_Volt_Neo\Release\Tiny_Volt_Neo.hex:
>     avrdude.exe: input file D:\Eclipse_Workspaces\sloeber44\Tiny_Volt_Neo\Release\Tiny_Volt_Neo.hex auto detected as Intel Hex
>     avrdude.exe: input file D:\Eclipse_Workspaces\sloeber44\Tiny_Volt_Neo\Release\Tiny_Volt_Neo.hex contains 5650 bytes
>     avrdude.exe: reading on-chip flash data:
> 
>     Reading | ################################################## | 100% 2.12s
> 
>     avrdude.exe: verifying ...
>     avrdude.exe: 5650 bytes of flash verified
> 
>     avrdude.exe: safemode: Fuses OK
> 
>     avrdude.exe done. Thank you

theduskin, 2023-11-23:
please mark as closed. 
the working projekt was build on a linux machine and after that the code was changed.
so the projects did not share the same code as i thought. I'm sorry, my fault.
I made a new project and copied the sources, everything works as expected.

jantje, 2023-11-23:
No problem. Thanks for letting us know your problem is resolved was not a Sloeber issue.

closed 2023-11-23 by jantje