Sloeber/arduino-eclipse-plugin #1592: Missing icons & Sloeber perspective

mr-miky, 2023-11-24:
I compiled the plugin by downloading it with git and then passed it on to maven.

Everything went well but when I install the plugin on Eclipse 4.29 does not appear the perspective sloeber among those selectable and also the icon is replaced with a red square as if it did not find the icon but I checked and the png files of the icons there are all.

If instead of installing the plugin in a completely new version I install it in one that had already sloeber installed the perspective remains but here also the icon disappears but it is as if it were the residue of the old configuration and highlights it by placing sloeber between  < >

```
eclipse.buildId=4.29.0.I20230903-1000
eclipse.commands=-os
linux
-ws
gtk
-arch
x86_64
```

[image]

jantje, 2023-11-24:
It is strange that the compile went well because I am currently fixing the "the compile does not work" in issue #1587 .

mr-miky, 2023-11-25:
> It is strange that the compile went well because I am currently fixing the "the compile does not work" in issue #1587 .

I don’t know what to say maybe we’re building different things.

```
git clone https://github.com/Sloeber/arduino-eclipse-plugin sloeber
cd sloeber
mvn clean verify -Plinux64,latest,NOSDK -DskipTests=true
```

[image]

jantje, 2023-11-25:
Strange as it doesn't work for me and has not worked for others
[image]

Anyway the latest eclipse is one way or another not compatible with the latest released Sloeber. As Sloeber hasn't been released for some time I don't know when the incompatibility started.
That is why I'm working on a new release.

mr-miky, 2023-11-27:
> Anyway the latest eclipse is one way or another not compatible with the latest released Sloeber. As Sloeber hasn't been released for some time I don't know when the incompatibility started. That is why I'm working on a new release.

Last Friday I used a laptop that was unused for a while and sloeber works in the sense that it has the Sloeber perspective. If I remember correctly it uses Eclipse 2023.06 and java 17 anyway tomorrow I can verify.
I surely remember that at the beginning of the current year everything worked then after an update of eclipse maybe 2023.03 the sloeber menu had disappeared but I accidentally discovered that if I use java 17 instead of 11 the menu reappears.

jantje, 2023-11-27:
> I accidentally discovered that if I use java 17 instead of 11 the menu reappears.

Java17 is a must because Sloeber is based on CDT which requires Java17. If you run from Java11 all menu's and toolbars provided by CDT and Sloeber should disappear.

jantje, 2023-12-01:
Closing this issue as fixed. Please reopen if you disagree

closed 2023-12-01 by jantje