Sloeber/arduino-eclipse-plugin #1601: Unable to build plugin

mr-miky, 2023-12-10:
Every now and then I like to hurt myself and I agreed to update eclipse to the latest version of 2023-12 to see if the problem of missing some plugin icons had disappeared.
The problem persists so I decided to recompile the sloeber plugin to see if anything changes.
Contrary to what happened up to 2 weeks ago, the plugin no longer compiles.

[image]

The java version is 17 

```
java.class.version=61.0
java.home=/usr/lib/jvm/java-17-openjdk-17.0.8.0.7-1.fc38.x86_64
java.io.tmpdir=/tmp
java.library.path=/usr/java/packages/lib:/usr/lib64:/lib64:/lib:/usr/lib
java.runtime.name=OpenJDK Runtime Environment
java.runtime.version=17.0.8+7
java.security.manager=allow
java.specification.name=Java Platform API Specification
java.specification.vendor=Oracle Corporation
java.specification.version=17
java.vendor=Red Hat, Inc.
java.vendor.url=https://www.redhat.com/
java.vendor.url.bug=https://bugzilla.redhat.com/enter_bug.cgi?product=Fedora&component=java-17-openjdk-portable&version=38
java.vendor.version=(Red_Hat-17.0.8.0.7-1.fc38)
java.version=17.0.8
java.version.date=2023-07-18
java.vm.compressedOopsMode=Zero based
java.vm.info=mixed mode, sharing
java.vm.name=OpenJDK 64-Bit Server VM
java.vm.specification.name=Java Virtual Machine Specification
java.vm.specification.vendor=Oracle Corporation
java.vm.specification.version=17
java.vm.vendor=Red Hat, Inc.
java.vm.version=17.0.8+7
jdk.debug=release
```

jantje, 2023-12-10:
I just tried the build on windows and it worked for me. So it is a linux issue or a hiccup.

### Why did the build fail?
The error logs states "cannot resolve dependencies ..." and then the specific dependency X uses Y; can't find Y
Sloeber is dependent on many other plugins. To find these plugins sloeber uses a target file.
https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.product/sloeber.target
As you can see the target file points to several download repositories
`<repository location="https://download.eclipse.org/releases/latest"/>`
 and the maven repository
`<location includeDependencyDepth="none" includeDependencyScopes="compile" includeSource="true" label="DirectFromMaven" missingManifest="error" type="Maven">`
If these repositories do not contain the java library/plugin Sloeber is dependent on the build "cannot resolve dependencies ..."

### Why can there be hiccups?
If any of these repositories can not be reached (firewall; network connection issues ..) or the repository is not available (it is being updated or server down or corrupted ...) the build will fail.
Waiting a while should fix this issue (can take a couple of days when personnel is needed to fix the issue)

### What if the build fail is more persistent?
Sloeber points to the latest versions of the repositories.
That means that with every eclipse release the latest release repository gets replaced with a completely new one. If a plugin (or a dependent java library) that Sloeber is dependent on is no longer in the repository the build is broken.
This happened with 4.4.1 where eclipse stopped using java libraries Sloeber used. That is why the maven repository was added in 4.4.2.

Technically it would be better to point to explicit versions of the repositories (they do not get replaced and do not need updates) but that would mean way more work to keep the target file up to date.
It would also mean the build does not build against the latests and greatest eclipse release.

jantje, 2023-12-10:
As I myself struggled a lot with "I don't really understand what it doesn't find." I'm converting this to a question.