Sloeber/arduino-eclipse-plugin #1605: Automatic Library handling problems

DocAlex, 2023-12-26:
Hello,

I am having a problem with the automatic library handling mechanism. 

### First of all I understood the following:

I put a library that I want to use tn the sloeber workspace. Name the folder as I want bit put the source code into the src folder. While compiling sloeber copies the a library that I wnat to use by #include into the libray folder in the project. 

If I dont use the library by removing the #include, sloeber automatically deletes the library from the library folder in the project. 

If I understood this correctly it does not work like this. 

### Here the settings:

[image]

I installed the libraries here:

[image]

### Then I load my project

[image]

this currently uses the library "nonblocking Dallas.h". It is correctly copied in the libraries folder.

In the next step I remove this library from the .ino file

[image]

but the library file stays in the library folder.

I tried to delete all libraries because I believe they are automatically coming back

[image]

In fact they do - including the unused one. "nonblocking Dallas.h"

To evaluate this further I have copied the complete code into a text editor and copied the text content of the blink.ino example in the sloeber editor.

[image]

The libraries stayed there. I again deleted all of them and this time they stayed deleted since I am not using any library in the code. 

[image]

A was able to compile and upload the project

Now I took back the code that I stored meanwhile in the external editor and replaced the blink code in Sloeber

[image]

compilation brought back the libraries, and to my big surprise, without the unneccessary library  "nonblocking Dallas.h". This is now the correct set of libraries. 

**But I think there is an error in sloeber handling the libraries correctly.** 

### Here another example:

Now I am including another library in the code: "DS18B20_INT.h"

[image]

This library is stored correctly in "my libraries"

[image]

Now Sloeber does NOT copy the library from MyLibraries into the project libraries:

[image]

and consequently I am getting an include error

[image]

In some other cases I copied the missing library by hand into "libraries" but after a second compile obviously sloeber copied something somewhere and I cot a double definition error. In this case I was only able to rescue the project by creation of a new project and copying the content of the old project into the editor. 
### 
Fixes I tried:

I tried to use the refresh button (right clik on libraries), I tried to restart sloeber but nothing did help. 

Can you help?

Happy new year 

Alexander

jantje, 2023-12-30:
It seems to me that you have done your homework 👍 
However there are some misunderstandings about how things are supposed to work.
When I read through what you have written everything matches with how I think Sloeber works and It seems it work even better than I expected 😉 

As to your understanding:
>While compiling Sloeber **copies** the a library 

Sloeber does not copy the libraries. Sloeber adds the library folder with a eclipse level link. In other words the Libraries folder in your workspace is empty. You can see the difference between a real resource (resource is eclipse speak for project file or folder) and a eclipse level linked resource by looking at the icon. A linked resource has this arrowy thing on the bottom right.

>While **compiling** Sloeber copies the a library 
It is not at compile time (though it does look like it because a build quite often triggers the indexer) but it is when the indexer is ready.

>If I dont use the library by removing the #include, sloeber automatically deletes the library from the library folder in the project.

Nope. Sloeber does not remove libraries on it's own (except for some rare cases like changing the configuration)

>If I understood this correctly it does not work like this.

You understood correctly that it does not work like this.

>In fact they do - including the unused one. "nonblocking Dallas.h"

Dallas.h is probably still in the indexer cache. You should remove the includes; save all files; wait for the indexer to finish (and hope the indexer worked as expected) then delete the libraries folder.

>But I think there is an error in sloeber handling the libraries correctly.

Errors for sure; but nothing I can see here. But the libraries inclusion is rough and absolutely not as smooth as Arduino  ide.
That is why there is an option to turn it off.

>Now I am including another library in the code: "DS18B20_INT.h"

Sloeber assumes the library in a folder that matches the include name -.h. There is some extra magic that got you pretty far.
As we can not support magic and this library does not match the rule; it works as designed.

Tips from what I see:
You should put sloeber closer to the root. You will get into "command line to long" issue if you don't.
Maybe have a look at
https://github.com/Sloeber/arduino-eclipse-plugin/discussions/1548
https://github.com/Sloeber/arduino-eclipse-plugin/discussions/1555

jantje, 2024-02-20:
Can this be closed?

DocAlex, 2024-02-20:
Yes. Thank you

closed 2024-02-20 by jantje