Sloeber/arduino-eclipse-plugin #1606: Minor suggestions

DocAlex, 2023-12-26:
Here are some minor suggestions/propbems

### 1. The humbling new user

When you start with 1 project loaded in sloeber and hit the upload button - you get an error messagethat no project is selected.
I understand that in the case of multiple projects this does make sense to avoid accidential compilation. 

But in the case of a single project? I would expect that this is automatically selected

### 2. I understood that the new upload button shall be used only. 

[image]

Is there a possibility to get rid of the "old arduino upload" button? It is really annoying when you accidentially hit it. 

### 3. Console

In some cases (I do not understand how this can happen) The very important console view is not showing up. I would like to suggest that a new menue item is added to the "Windows" menu called "show Console". The only way I found was going tho the "help menu" searching for "console" and click on the item. 

Thanks in advance

Alexander

DocAlex, 2023-12-26:
Forgotten:

4. Properties

for a beginner it is confusing, that there are multiple places for "Properties".

And even depending on what is selected the properties are different:

When libraries are selected - there is no "sloeber" menu

[image]

When the project is selected - there is a "sloeber" menu.

[image]

This is confusing me often.

jantje, 2023-12-30:
1) that could be an option. Feel free to contribute it.
2) you understood wrong
3) are you in the sloeber or the CDT perspective? It should be there. Are you sure you didn't minimize the console or something like that?
4) Eclipse is a professional development environment. That brings a very high confusion level.

closed 2024-12-20 by jantje