Sloeber/arduino-eclipse-plugin #1607: linked resource not found for #include

MasteringTheMess, 2024-01-04:
Hi,
I use 4.4.2 on windows

We have multiple projects that use some common files that are linked to out of the corresponding projects. Common files are just a folder, no cdt/sloeber/eclipse project. However all three projects are grouped under one empty root project. (Don't know if this is relevant so I guess I better mention it)
Now, to add confusion, we changed HW in the development process and now have to support two different targets/board types.
To do so, I created a separate new project and just link to the existing code. So in this project folder there exist only eclipse and sloeber config files, and the build files.

For a reason we needed to include a project-specific file in the common codebase. Which works fine within the original setup by only adding the corresponding project path to the included folders list in project -> properties ->C/C++ build -> Settings

But it does not work within the project that only links to the source code. I get a "fatal error - file or directory not found" here, which I dont understand why this happens ...

MasteringTheMess, 2024-01-05:
Thanks for the clarification!
I currently ran in some other problems so cannot investigate this
I assume I have a command-lie length problem too :/
28.998 chars compile fine
33.295 give me an error 87 - which could indicate a problem with the length according to some quickly googled comments

jantje, 2024-01-05:
>this happens only with make -clean and it is the rm command,

In new Sloeber clean will be delete the build folder. (Don't hold your breath)

MasteringTheMess, 2024-01-05:
new Sloeber?

Now that's exciting news!
Any guess on the release date so far?

jantje, 2024-01-05:
Been working on it for over a year now.
I'm rewriting managed build (that is the CDT part Sloeber is build on) to be able to fix some long lasting issues in regards with archiving libraries before linking.
Lets hope this year (2024) we'll have a release.

jantje, 2024-02-20:
Any progress on this?

jantje, 2024-02-21:
That makes this issue in my book a "one of" and a duplicate.

MasteringTheMess, 2024-02-21:
yeah, can be closed and you can forget about it and move on to other fun things :)

closed 2024-02-21 by jantje