Sloeber/arduino-eclipse-plugin #1608: Can we have ${SLOEBER_HOME} in .sproject and sloeber.cfg files please

MasteringTheMess, 2024-01-04:
This would add to portability of the projects around systems

All other project settings like include paths, linked resources etc are using env. variables.

The path in .sproject and sloeber.cfg is the only setting that has absolute path left

jantje, 2024-01-04:
Strange.
This is how it should work.

> Config.mega.board.BOARD.ID=mega
> Config.mega.board.BOARD.MENU.cpu=atmega2560
> Config.mega.board.BOARD.TXT=${SLOEBER_HOME}/arduinoPlugin/packages/arduino/hardware/avr/1.8.6/boards.txt
> Config.mega.compile.sloeber.extra.all=
 
Looking at the code I see it is a string replace which means potential / \ issues
https://github.com/Sloeber/arduino-eclipse-plugin/blob/b1b8e70e088f4d7054a825b2ecddb8eb47b2631e/io.sloeber.core/src/io/sloeber/core/common/Common.java#L244
https://github.com/Sloeber/arduino-eclipse-plugin/blob/b1b8e70e088f4d7054a825b2ecddb8eb47b2631e/io.sloeber.core/src/io/sloeber/core/common/Common.java#L241

MasteringTheMess, 2024-01-05:
In this context I really think I should have mentioned that I am on Windows system...

and that slashes are backslashes ( \ )

jantje, 2024-01-05:
I'm on windows to right now.
I'm only trying to say you need an exact string match for it to work.
You do realize you can define SLOEBER_HOME at the OS level?

MasteringTheMess, 2024-01-05:
Nope, wasn't aware of that...

jantje, 2024-08-13:
The nightly has a way better implementation to get ${SLOEBER_HOME} in sloeber.cfg
Still looking how to get  ${SLOEBER_HOME} in .sproject