Sloeber/arduino-eclipse-plugin #1609: missing slashes for archiver results in build error

MasteringTheMess, 2024-01-05:
I have a somewhat larger project) and am in the process of porting from AVR to STM32
Not sure if this is sloeber or toolchain related ??
sloeber 4.4.2 on windows

build runs through fine, only after the archiver step it fails with a strange message
```
Finished building: C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.7.1\cores\arduino\main.cpp
 
Finished building: C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.7.1\cores\arduino\wiring_pulse.cpp
 
Starting archiver
"C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\tools\xpack-arm-none-eabi-gcc\12.2.1-1.2/bin/arm-none-eabi-gcc-ar" rcs  "C:\sloeber_fs-Test\TestHead_STM32\Release\arduino.ar"  .\core\core\stm32\usb\hid\usbd_hid_composite.c.o .\core\core\stm32\usb\hid\usbd_hid_composite_if.c.o  .\core\core\stm32\usb\cdc\cdc_queue.c.o .\core\core\stm32\usb\cdc\usbd_cdc.c.o .\core\core\stm32\usb\cdc\usbd_cdc_if.c.o  .\core\core\stm32\usb\usb_device_core.c.o .\core\core\stm32\usb\usb_device_ctlreq.c.o .\core\core\stm32\usb\usb_device_ioreq.c.o .\core\core\stm32\usb\usbd_conf.c.o .\core\core\stm32\usb\usbd_desc.c.o .\core\core\stm32\usb\usbd_ep_conf.c.o .\core\core\stm32\usb\usbd_if.c.o  .\core\core\stm32\OpenAMP\virtio\virtio.c.o .\core\core\stm32\OpenAMP\virtio\virtqueue.c.o  .\core\core\stm32\OpenAMP\open-amp\rpmsg\rpmsg.c.o .\core\core\stm32\OpenAMP\open-amp\rpmsg\rpmsg_virtio.c.o  .\core\core\stm32\OpenAMP\open-amp\remoteproc\remoteproc_virtio.c.o  .\core\core\stm32\OpenAMP\libmetal\generic\cortexm\sys.c.o  .\core\core\stm32\OpenAMP\libmetal\generic\condition.c.o .\core\core\stm32\OpenAMP\libmetal\generic\generic_device.c.o .\core\core\stm32\OpenAMP\libmetal\generic\generic_init.c.o .\core\core\stm32\OpenAMP\libmetal\generic\generic_io.c.o .\core\core\stm32\OpenAMP\libmetal\generic\generic_shmem.c.o .\core\core\stm32\OpenAMP\libmetal\generic\time.c.o  .\core\core\stm32\OpenAMP\libmetal\device.c.o .\core\core\stm32\OpenAMP\libmetal\init.c.o .\core\core\stm32\OpenAMP\libmetal\io.c.o .\core\core\stm32\OpenAMP\libmetal\log.c.o .\core\core\stm32\OpenAMP\libmetal\shmem.c.o  .\core\core\stm32\OpenAMP\mbox_ipcc.c.o .\core\core\stm32\OpenAMP\openamp.c.o .\core\core\stm32\OpenAMP\rsc_table.c.o .\core\core\stm32\OpenAMP\virt_uart.c.o .\core\core\stm32\OpenAMP\virtio_buffer.c.o .\core\core\stm32\OpenAMP\virtio_log.c.o  .\core\core\stm32\startup_stm32yyxx.S.o  .\core\core\avr\dtostrf.c.o  .\core\core\HardwareSerial.cpp.o .\core\core\IPAddress.cpp.o .\core\core\Print.cpp.o .\core\core\RingBuffer.cpp.o .\core\core\Stream.cpp.o .\core\core\Tone.cpp.o .\core\core\USBSerial.cpp.o .\core\core\VirtIOSerial.cpp.o .\core\core\WInterrupts.cpp.o .\core\core\WMath.cpp.o .\core\core\WSerial.cpp.o .\core\core\WString.cpp.o .\core\core\abi.cpp.o .\core\core\board.c.o .\core\core\core_debug.c.o .\core\core\hooks.c.o .\core\core\itoa.c.o .\core\core\main.cpp.o .\core\core\pins_arduino.c.o .\core\core\wiring_analog.c.o .\core\core\wiring_digital.c.o .\core\core\wiring_pulse.cpp.o .\core\core\wiring_shift.c.o .\core\core\wiring_time.c.o 
c:\sloeber\arduinoplugin\packages\stmicroelectronics\tools\xpack-arm-none-eabi-gcc\12.2.1-1.2\bin\../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ar.exe: .corecorestm32usbhidusbd_hid_composite.c.o: No such file or directory
make[1]: *** [makefile:156: C:\sloeber_fs-Test\TestHead_STM32\Release\arduino.ar] Error 1
make: *** [makefile:141: all] Error 2
"C:/Sloeber/arduinoPlugin/tools/make/make -j12 all" terminated with exit code 2. Build might be incomplete
```

it seems it is looking for a file named `.corecorestm32usbhidusbd_hid_composite.c.o` which resembles a lot the first file with path `.\core\core\stm32\usb\hid\usbd_hid_composite.c.o`

maybe this is a slash/backslash - problem?

jantje, 2024-01-05:
The command line is simply wrong.
I'm not sure how you created/modified the project but this does not look like "default Sloeber behavior"
Let me explain why:
When Sloeber starts the build process it starts it in the build folder (here C:\sloeber_fs-Test\TestHead_STM32\Release).
So all files must be referenced as FQN from there or as relative paths "as they are on the file system" (not like Sloeber represents them as the Eclipse linking is not available on the OS).
So when the command specifies .\core\core\stm32\usb\hid\usbd_hid_composite.c.o the archiver will look for
C:\sloeber_fs-Test\TestHead_STM32\Release\core\core\stm32\usb\hid\usbd_hid_composite.c.o
Which should not exists in a Sloeber install.

MasteringTheMess, 2024-01-05:
these files are created during build with STM32duino for STM32 targets
I just checked, the specified file exists

jantje, 2024-01-05:
I just realized I made a mistake. Indeed the command seems fine.
The command is 10820 long though so it may be a command to long thing
Can you experiment on the command line to see how it behaves there (making it shorter and so on)

jantje, 2024-02-20:
I assume this is a duplicate of #1053

closed 2024-02-20 by jantje