Sloeber/arduino-eclipse-plugin #1610: Sloeber V4.4.2 ESP32 Index bug

GitHubLionel, 2024-01-18:
Hi,

I have a problem with the update of the Index.
When I change something in Sloeber properties of my ESP32 project, for example in Compile Options and then after "Apply and Close" I try to rebuild the Index, nothing is happening and all the lines of code is tagged red.
I need to close and reopen Sloeber many times to finally have the rebuild of the Index.
Any idea to solve this problem ?
thanks,
Lionel

jantje, 2024-01-18:
You are probably experiencing problems with the discovery not the indexer.
You can verify this as there should be errors in the error log and your projects should have an error about failing to run (something)
The only working work around I know is to go to project properties->C/C++ General->preprocessor include paths ...->Providers->Arduino Compiler Settings
Modify the "command to get compiller specs" (FI add space at end or replace the $COMMAND with the actual command and fix it forever for this config)
Select apply and close
[image]

GitHubLionel, 2024-01-19:
Hi Jantje,
No, this not solve the problem.
To reproduce the problem I create a simple ESP32 project with only this code:
```
#include "Arduino.h"

void setup()
{
}

void loop()
{
}
```
Then I change something in properties, for example compile options. After, I try to rebuild Index many times with open and close several times Sloeber. Here is the log :
```

!SESSION 2024-01-19 16:40:58.845 -----------------------------------------------
eclipse.buildId=unknown
java.version=17.0.9
java.vendor=Eclipse Adoptium
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=fr_FR
Framework arguments:  -perspective io.sloeber.application.perspective
Command-line arguments:  -os win32 -ws win32 -arch x86_64 -perspective io.sloeber.application.perspective

!ENTRY org.eclipse.equinox.app 0 0 2024-01-19 16:41:00.547
!MESSAGE Product io.sloeber.product.product could not be found.

!ENTRY org.eclipse.egit.ui 2 0 2024-01-19 16:41:07.923
!MESSAGE Warning: The environment variable HOME is not set. The following directory will be used to store the Git
user global configuration and to define the default location to store repositories: 'C:\Users\Lionel'. If this is
not correct please set the HOME environment variable and restart Eclipse. Otherwise Git for Windows and
EGit might behave differently since they see different configuration options.
This warning can be switched off on the Team > Git > Confirmations and Warnings preference page.

!ENTRY io.sloeber.core 8 0 2024-01-19 16:41:15.994
!MESSAGE Index of project changed :ESP32_Test

!ENTRY org.eclipse.cdt.core 1 0 2024-01-19 16:41:16.045
!MESSAGE Indexed 'ESP32_Test' (0 sources, 0 headers) in 0,013 sec: 0 declarations; 0 references; 0 unresolved inclusions; 0 syntax errors; 0 unresolved names (0 %)

!ENTRY io.sloeber.core 8 0 2024-01-19 16:41:16.063
!MESSAGE Looking for libraries for project :ESP32_Test

!ENTRY io.sloeber.core 8 0 2024-01-19 16:41:16.087
!MESSAGE libraries added for project ESP32_Test

!ENTRY org.eclipse.cdt.managedbuilder.core 4 0 2024-01-19 16:41:21.962
!MESSAGE Error running Builtin Specs Detector
!STACK 1
org.eclipse.core.runtime.CoreException: Error running Builtin Specs Detector
	at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector.runForLanguage(AbstractBuiltinSpecsDetector.java:744)
	at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector.runForEachLanguage(AbstractBuiltinSpecsDetector.java:601)
	at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector$1.runInWorkspace(AbstractBuiltinSpecsDetector.java:527)
	at org.eclipse.core.internal.resources.InternalWorkspaceJob.run(InternalWorkspaceJob.java:43)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)
Caused by: org.eclipse.core.internal.resources.ResourceException: Project 'ESP32_Test' is not open.
	at org.eclipse.core.internal.resources.Project.checkAccessible(Project.java:186)
	at org.eclipse.core.internal.resources.Resource.checkAccessibleAndLocal(Resource.java:247)
	at org.eclipse.core.internal.resources.Resource.getSessionProperty(Resource.java:1228)
	at org.eclipse.cdt.internal.core.BuildRunnerHelper.prepareStreams(BuildRunnerHelper.java:163)
	at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector.runForLanguage(AbstractBuiltinSpecsDetector.java:728)
	... 4 more
Caused by: java.lang.Exception: Project 'ESP32_Test' is not open.
	at org.eclipse.core.internal.resources.ResourceException.provideStackTrace(ResourceException.java:42)
	at org.eclipse.core.internal.resources.ResourceException.<init>(ResourceException.java:38)
	... 9 more
!SUBENTRY 1 org.eclipse.cdt.managedbuilder.core 4 0 2024-01-19 16:41:21.963
!MESSAGE Error running Builtin Specs Detector
!STACK 1
[... 484 lines cut ...]
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:568)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:651)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:588)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1459)

!ENTRY io.sloeber.arduino.core 1 0 2024-01-19 16:54:27.210
!MESSAGE The folder "C:\Users\Lionel\Documents\Arduino\hardware" is empty.

!ENTRY io.sloeber.core 8 0 2024-01-19 16:55:00.897
!MESSAGE Index of project changed :ESP32_Test

!ENTRY org.eclipse.cdt.core 1 0 2024-01-19 16:55:00.920
!MESSAGE Indexed 'ESP32_Test' (0 sources, 0 headers) in 0,017 sec: 0 declarations; 0 references; 0 unresolved inclusions; 0 syntax errors; 0 unresolved names (0 %)

!ENTRY io.sloeber.core 8 0 2024-01-19 16:55:00.935
!MESSAGE Looking for libraries for project :ESP32_Test

!ENTRY io.sloeber.core 8 0 2024-01-19 16:55:00.949
!MESSAGE libraries added for project ESP32_Test
```

Sometimes, when I close Sloeber with the project open and then restart Sloeber, I have this window that freeze Sloeber then I need to kill Sloeber :

[image]

jantje, 2024-01-19:
Apart from 
A) you working from the "unsupported plugin" instead of the "supported product" (Not that I think this will have an impact)
B) the project not open issue (which I think is a CDT issue)  https://github.com/eclipse-cdt/cdt/issues/634
I do not see any problems in the log. 

>Sometimes, when I close Sloeber with the project open and then restart Sloeber, I have this window that freeze Sloeber then I need to kill Sloeber :

I noticed freezes when starting Sloeber as well but I never found the root cause. Note that in your case the freeze is at what I described before https://github.com/Sloeber/arduino-eclipse-plugin/issues/1610#issuecomment-1898956147
 I hope #1126 will fix this as it will work completely differently in regards of open and closing projects.

GitHubLionel, 2024-01-20:
Hi,
I try with ESP32 2.0.11 : same problem
I try to go back Sloeber V4.4.1 : same problem
Note that I don't have problem with ESP8266, so the problem comes with ESP32. Maybe because we delete the file platform.sloeber.txt ?
The fact is that in the log file, we can see this lines :
```
!ENTRY org.eclipse.cdt.core 1 0 2024-01-20 15:04:03.612
!MESSAGE Indexed 'ESP32_Test' (0 sources, 0 headers) in 0,005 sec: 0 declarations; 0 references; 0 unresolved inclusions; 0 syntax errors; 0 unresolved names (0 %)
```
Why he write : 0 sources, 0 headers ?

jantje, 2024-01-20:
>Why he write : 0 sources, 0 headers ?

I have seen this quite often. I do not know why CDT does that. 

After the logging you see 
```
!ENTRY org.eclipse.cdt.core 1 0 2024-01-19 16:41:16.045
!MESSAGE Indexed 'ESP32_Test' (0 sources, 0 headers) in 0,013 sec: 0 declarations; 0 references; 0 unresolved inclusions; 0 syntax errors; 0 unresolved names (0 %)

!ENTRY io.sloeber.core 8 0 2024-01-19 16:41:16.063
!MESSAGE Looking for libraries for project :ESP32_Test

!ENTRY io.sloeber.core 8 0 2024-01-19 16:41:16.087
!MESSAGE libraries added for project ESP32_Test
```
`!MESSAGE libraries added for project ESP32_Test` tells you the index is not empty.

```
!ENTRY org.eclipse.cdt.managedbuilder.core 4 0 2024-01-19 16:41:21.962
!MESSAGE Error running Builtin Specs Detector
!STACK 1
org.eclipse.core.runtime.CoreException: Error running Builtin Specs Detector
```
The entry above clearly states the discovery phase was not successfull

GitHubLionel, 2024-01-21:
Hi,
Well, for the moment, I close and re-open Sloeber many times until Indexed work. 
Note that it is not the only perturbation I noticed with the version 4.4.2 :
- no splash-screen at the beginning of Sloeber (I have one with older version)
- no icon for the Sloeber program in the program bar of Window
- sometimes there is no log in the console when I compile a sketch (I need to close it and re-open with Window/Show View/Console)
- same with the problems view

But these problems are not blocking.
Regards,
Lionel

GitHubLionel, 2024-04-09:
Hi,
I just installed Sloeber V4.4.3. Always the same problem with the Index. The only way I found when index not work is :
- clean build
- make build
- close Sloeber
- re-open Sloeber
- then the rebuild index work

Regards,
Lionel

jantje, 2024-04-09:
I'm currently working on #1126 that should fix this issue.
Don't hold your breath as I've been working on it for over a year now.

GitHubLionel, 2024-04-09:
Hi
Many thanks Jantje for your great job :)
Best regards,
Lionel

msebas, 2024-06-18:
Hello, I encounter the same problem (Errors from missing imports for some of the framework/core components like in the screenshot below), but with slightly different symptoms. 
[image]

I created a new ESP32 Project (Default Template) using the Arduino ESP32 Core 3.0.1 and when opening the Arduino.h  many includes are reported as missing (with their symbols) in the IDE. 

Then I tried to apply the suggestions here, but they did not work out, so I dug a little bit deeper and when enabling the console output I got
```
17:51:20 **** Running scanner discovery: Arduino Compiler Settings ****
${recipe.cpp.o.CODAN} -E -P -v -dD -D__IN_ECLIPSE__ /home/Sebastian/workspace/test/.metadata/.plugins/org.eclipse.cdt.managedbuilder.core/spec.cpp 
Cannot run program "${recipe.cpp.o.CODAN}": Unknown reason

Error: Program "${recipe.cpp.o.CODAN}" not found in PATH
PATH=[/home/Sebastian/programme/eclipse/sloeber/Sloeber/arduinoPlugin/packages/esp32/tools/esp-x32/2302/bin/::/usr/local/cuda/bin/:/home/Sebastian/bin:/usr/local/bin:/usr/bin:/bin:/usr/lib/mit/bin:/usr/lib/mit/sbin]

17:51:20 Build Failed. 1 errors, 0 warnings. (took 1ms)
```

When I resolve "${recipe.cpp.o.CODAN}" in the environment I get 
"/home/Sebastian/programme/eclipse/sloeber/Sloeber/arduinoPlugin/packages/esp32/tools/esp-x32/2302/bin/xtensa-esp32-elf-gcc" which exists and is a compiler.
If I get some idea where to look for I would try to find a workaround and document it or fix the problem.

The bash output of sloeber during the command is 
```
Unable to find full path for "${recipe.c.o.CODAN}"
Unable to find full path for "${recipe.cpp.o.CODAN}"
```

If I try to compile it it fails as described in #1649 and I worked around that as I described in #1652. After that the project builds and uploads just fine. Its only the IDE code index that suggests errors.

The same process works for Arduino Nano without showing these errors.

jantje, 2024-06-19:
@msebas 
The problems you report on the includes are typical problems of the indexer not having indexed properly.
The logging you show is showing the indexer gets the "wrong" (that is not expanded environment variable) commands from Sloeber.
Therefore it is safe to conclude that the root cause for the indexing problem is in Sloeber not having expanded the environment variables before the indexer requests them.
The fact the build also fails because of the non-expanded variables this you are not simply experiencing a indexer bug (which this issue is about)

If I have behavior like that the first thing I do is: go to the project properties->sloeber and verify the data is correct  and then select apply/close.

msebas, 2024-06-21:
I took a look into the Sloeber configuration as suggested, but all options look correct. 

So I compared the options for a Arduino ESP32 Core 2.0.16 Test Project (which works fine out of the box) and a Arduino ESP32 Core 3.0.1 with the fixes mentioned in #1652 done to its environment.

For the 2.0.16 Project everything works fine, but the 3.0.1 shows the result above.
I attached a screenshot of the 3.0.1 project as I do not know what might result in the exact same configuration working for 2.0.16 and not working for 3.0.1.

[image]

[image]

I did the following stepts to set up my test projects in an isolated Sloeber 4.4.3 on opensuse linux 15.5 (Kernel 5.14.21):

* Create Project
* Check Sloeber Board Page and Apply
* Check Sloeber Option Page and Apply
* Add ENV Variables (only for 3.0.1, see #1652)
* Add ESP32 Lib (Otherwise when build the FreeRTOS-Lib is added and the compile fails)
* Verify Project (e.g. build it to create release folder)
* Update indexer configuration to print to console

As  I said for 2.0.16 it works and for 3.0.1 it fails.

[image]

jantje, 2024-07-06:
Have you seen #1652

jantje, 2024-08-19:
Seems the indexer bug is fixed in V5

closed 2024-08-19 by jantje