Sloeber/arduino-eclipse-plugin #1611: Teensy JSON fails Board Platform install due to unsupported tar.zst archives

mattbaker-digital, 2024-01-19:
Sloeber 4.4.2 Product Bundle
OS: Windows 10  
Platform JSON: https://www.pjrc.com/teensy/package_teensy_index.json

Project properties arduino showing the board settings:  
[image]

Sloeber settings:  
[image]

**Describe the bug**   
The archive files in the Teensy package JSON are not supported (tar.zst file)  
[image]

**To Reproduce**   
Steps to reproduce the behavior:   
1. Go to `Window` -> `Preferences`
2. Click on `Sloeber` -> `Third party index url's` section
3. Add `https://www.pjrc.com/teensy/package_teensy_index.json`
4. Click `Apply and Close` to force the board list update.
5. Go to `Window` -> `Preferences`
6. Click on `Sloeber` -> `Platforms and Boards` section
7. Tick `teensy` -> `Teensy` -> `1.58.1`
8. Click `Apply`

**Expected behavior**   
Board to be installed correctly

**Data to back up your claim**   
If applicable, add screenshots, console output and/or logs to help explain your problem.   

**Additional context**   
As a workaround I tried the outdated (now installed to %LOCALAPPDATA% not Program files) instructions here:  
http://eclipse.baeyens.it/how_to.shtml#/i

This failed as the precompile_helper needs the path backslashes escaping.  
This is the default command which fails with permission denied error 126: `C:\Users\matt.baker\AppData\Local\Arduino15\packages\teensy\tools\teensy-tools\1.58.0\precompile_helper.exe -v "C:\Users\matt.baker\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.58.1\cores\teensy4" "C:\Users\matt.baker\Documents\sloeber-workspace\Test\Release" "/arm/bin/arm-none-eabi-g++" -x c++-header -O2 -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=158 -DARDUINO=10812 -DARDUINO_TEENSY41 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_UNITED_KINGDOM "-IC:\Users\matt.baker\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.58.1\cores\teensy4" "C:\Users\matt.baker\Documents\sloeber-workspace\Test\Release\pch\Arduino.h" -o "C:\Users\matt.baker\Documents\sloeber-workspace\Test\Release\pch\Arduino.h.gch"`

This is the modified version which works: `"C:\\Users\\matt.baker\\AppData\\Local\\Arduino15\\packages\\teensy\\tools\\teensy-tools\\1.58.0/precompile_helper" "C:\\Users\\matt.baker\\AppData\\Local\\Arduino15\\packages\\teensy\\hardware\\avr\\1.58.1/cores/teensy4" "C:\Users\matt.baker\Documents\sloeber-workspace\Test\Release" "C:\\Users\\matt.baker\\AppData\\Local\\Arduino15\\packages\\teensy\\tools\\teensy-compile\\11.3.1/arm/bin/arm-none-eabi-g++" -x c++-header -O2 -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=158 -DARDUINO=10607 -DARDUINO_TEENSY41 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Users\\matt.baker\\AppData\\Local\\Arduino15\\packages\\teensy\\hardware\\avr\\1.58.1/cores/teensy4" "C:\Users\matt.baker\Documents\sloeber-workspace\Test\Release\pch\Arduino.h" -o "C:\Users\matt.baker\Documents\sloeber-workspace\Test\Release\pch\Arduino.h.gch"`

jantje, 2024-01-20:
I confirm this is an issue.
It looks as if @PaulStoffregen made a new version of https://www.pjrc.com/teensy/package_teensy_index.json using a archive format that Sloeber does not (yet) supports and in one go changed all old platform versions (which you should not do) making it impossible to add any Teensy platform version to Sloeber.
I will add the archive format to Sloeber but I'm currently caught up in other work 😒

PaulStoffregen, 2024-01-20:
Yes, confirm, we switched to zstd compression.  Arduino IDE has supported zstd since version 2.0.4.

It gives significantly smaller files to download and decompresses much faster, especially on Windows.

PaulStoffregen, 2024-01-20:
Teensy will also soon start using Arduino's new pre_uninstall script feature.

https://github.com/arduino/arduino-cli/issues/2277

So far no stable Arduino IDE release has this feature, but it is in the nightly builds.

I hope for your long term planning, consider checking for this script and run it before uninstall.  The script allows cleanup to be done.  In the short term, Teensy will use this to cause any lingering Teensy Loader application to automatically shut down.  If the installed files are to be deleted, this especially helps for Windows where the file can not be deleted if any instance of the program is still running.

jantje, 2024-01-20:
@PaulStoffregen 
Thanks for the confirmation and early warning.

jantje, 2024-01-20:
I'm in a bit of a pickle here.
Teensy install worked with commit #1612 in branch support_for_tar_zst_#1611
However the tests for the pull request failed due to "Sloeber needs a newer maven than github build uses" 😒 
_Due to this issue the github actions for Sloeber build has been down for a couple of months now._
The only way I can think of to get this build right now is to build the branch support_for_tar_zst_#1611 yourself

PaulStoffregen, 2024-01-21:
Is using an older version of the plugin an option?  On [this release notes file](https://github.com/eclipse-tycho/tycho/blob/main/RELEASE_NOTES.md), looks like version 3.0.5 might be the last before is started requiring maven 3.9.

jantje, 2024-01-21:
After some more investigation in this issue:
from https://github.com/actions/runner-images/discussions/8034

>As of version 4.0.0, the Eclipse Tycho plugin for Maven requires Maven 3.9.0 or later:

So basically github actions is very aware and the situations for Sloeber is:

1. Eclipse started a cleanup to remove packages from their dependency list and repositories (Which they have every right to do)
2. Sloeber installs no longer work due to missing packages.
3. I fixed this by referencing maven repositories in a .target file=>Sloeber maven build needs tycho 4.0.0 or later to build
4. Tycho  4.0.0  requires maven 3.9 or later
5. Maven 3.9 breaks many github actions (due to removal of dependencies to "old packages" -Where did I see this before😒 - )
6. Github actions have bad experiences with upgrading to maven 3.9 (due to point 5) so they do not feel like doing it again.
7. I find myself in the situation where I need to bet on waiting for github actions to upgrade maven or on investing time learning the proposed alternative solutions.

=>I'm in a bit of a pickle here.

PaulStoffregen, 2024-01-21:
Maybe I'm reading too much into this situation, but kinda feels like Github doesn't care much for Java?

jantje, 2024-01-21:
or eclipse?
or actions?
or ...
but .... that thought did cross my mind when reading about this issue.

mattbaker-digital, 2024-01-22:
@jantje thanks for looking into this. This is my first time using the plugin in several years.

mattbaker-digital, 2024-01-22:
Is there a way for me to temporarily fix this by pointing to the files installed by the Arduino IDE and somehow correcting the path to `/arm/bin/arm-none-eabi-g++`?

jantje, 2024-01-22:
>Is there a way for me to temporarily fix this by pointing to the files installed by the Arduino IDE and somehow correcting the path to /arm/bin/arm-none-eabi-g++?

There probably is but I think the easiest way it to compile sloeber yourself

You can find good instructions to build the Master branch and find the results here: https://github.com/Sloeber/arduino-eclipse-plugin/tree/master

You need change the build a bit to build the branch "support_for_tar_zst_#23161"
So the build command becomes
```
git clone https://github.com/Sloeber/arduino-eclipse-plugin sloeber
cd sloeber
git branch support_for_tar_zst_#231611
git switch support_for_tar_zst_#231611
mvn clean verify -DskipTests=true
```

mattbaker-digital, 2024-01-23:
@jantje thanks for the info. Unfortunately the latest branch is still failing for me when using the Teensy JSON file. I built the 64-bit product using `mvn clean verify -Pwin64,latest,NOSDK -DskipTests=true` after checking out the `support_for_tar_zst_#231611` branch.

mattbaker-digital, 2024-01-23:
I hacked about with the Status text on failure and it failed with the following package:  
[image]

I was unable to build the SDK version to develop (improve) the plugin properly as it reported missing the following dependencies: 
```
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: io.sloeber.product 4.4.2.qualifier
[ERROR]   Missing requirement: io.sloeber.product 4.4.2.qualifier requires 'org.eclipse.equinox.p2.iu; ilg.gnumcueclipse.debug.gdbjtag.openocd.feature.feature.group 0.0.0' but it could not be found
```

mattbaker-digital, 2024-01-23:
> There probably is but I think the easiest way it to compile sloeber yourself
>
> You can find good instructions to build the Master branch and find the results here: [...]

Ignore previous error about the package still not being supported. I followed the instructions to the letter which of course doesn't checkout the actual branch.

jantje, 2024-01-23:
@mattbaker-digital 

Did you succeed in building and installing/upgrading Sloeber ?
And if so did it fix the issue?

mattbaker-digital, 2024-01-24:
@jantje I managed to build the Sloeber for Windows in the end. As you probably already know from my deleted posts, I used the instructions above verbatim which didn't checkout the fixed branch. Using `git checkout support_for_tar_zst_#1611` worked (not sure where the 23 prefix to the branch name in the original comes from?).  

It fixed the issue for me. Thank you for the quick patch.  

Hope you manage to get he automated builds to work again without too much work - it sounds like a real pain 😕.  

On an unrelated note I did encounter missing dependency errors when attempting to build the SDK version for development / improvements to Sloeber.

jantje, 2024-01-24:
>As you probably already know from my deleted posts

That would have been an assumption and only me knowing that.
I asked to confirm so everybody who sees this issue knows you got it to work.

>it sounds like a real pain 

It is. Thanks for noticing.

>On an unrelated note I did encounter missing dependency errors when attempting to build the SDK version for development / improvements to Sloeber.

I confirm the issue. I should make SDK part of the normal build so it doesn't get left out.

jantje, 2024-01-25:
I got the actions build back up and running.

closed 2024-01-26 by jantje

jantje, 2024-02-04:
FYI
I'm currently release testing this version https://github.com/Sloeber/arduino-eclipse-plugin/actions/runs/7764103167
So people who have this issue can download the product (and update repository) from there.