Sloeber/arduino-eclipse-plugin #1624: MacOS 4.4.3 Latest Stable Not Working

mkumar2789, 2024-02-07:
**Environment**  
Sloeber Product version number : V4.4.3
OS: macOS Sonoma (14.3), M1 MacBook Air (ARM64)
URL (Download): https://eclipse.baeyens.it/stable.php?OS=MacOS  
Image attached, during installation, with comparison to previous version V4.4.2

**Bug Description**   
After download "sloeber-ide-V4.4.3-macosx.cocoa.aarch64.zip", followed the below steps:
i. unpacked the sloeber ide zip file with the macOS native Archive Utility. Not in the download folder!!!
ii. in the folder where you extracted Sloeber run: sudo xattr -r -d com.apple.quarantine Sloeber.app

When I clicked to open V4.4.3, got an error "You can’t open the application “Sloeber.app” because it may be damaged or incomplete." and after extracting "Sloeber.app” its icon also showing like not valid icon, show in attached photo, PFA.
I already highlighted comparison of latest and previous version's icon.

[image]

jantje, 2024-02-07:
As a test I just downloaded the windows 4.4.3 version and it unpacked and ran fine.
I also downloaded sloeber-ide-V4.4.3-macosx.cocoa.aarch64.zip.
It extracted fine and as far as I can see (which is not far ;-) )all seems fine.
I do not own a mac so I can not test this.
Mac has been a trouble to install due to all kind of security issues and my guess is you hit one of these.

What I think is really strange is that you have the icon for 4.4.2 which it should not have due to a bug fixed in 4.4.3.

Here is the thread that lead to the mac instructions #1141 it contains some alternative ways.
Alternatively you can build Sloeber locally (which I fear will be the default way one day)
Look here https://github.com/Sloeber/arduino-eclipse-plugin?tab=readme-ov-file#build-from-source for instructions.
You do need the latest maven which is probably the hardest part.

jantje, 2024-02-08:
I just updated the update site so you should be able to update your 4.4.2 version to 4.4.3
Do not forget to take backups of you eclipse and workspace.
Please let me know if this worked for you.

mkumar2789, 2024-02-08:
No, Sloeber.app 4.4.3 is still not working for me; same issue as described in the image.

closed 2024-02-20 by jantje