Sloeber/arduino-eclipse-plugin #1625: Sloeber create more than double size of .elf than Arduino IDE for STM32F103C8 BluePill

mkumar2789, 2024-02-07:
Environment
Sloeber Eclipse Plugin  V4.4.2
OS: macOS Sonoma (14.3), M1 MacBook Air (ARM64)

Bug Description  
I was trying to switch from Arduino IDE to Sloeber.
I followed the given tutorial for this:
https://github.com/ArminJo/SloeberTutorial

My configuration is in the screenshot:
1.
[image]

2.
[image]

3.
[image]

I got an error: (.elf section `.text' will not fit in region `FLASH'; region `FLASH' overflowed by 44808 bytes)
Part-1
[image]

Part-2
[image]

But when I change the existing MCU in Sloeber from BluePill F103C8 (64KB) to BluePill F103CB (128KB), the output shows no error but the size of the code!
[image]

Now look at Arduino IDE configuration and code size with BluePill F103C8 (64KB)
Config:
[image]

Output Part-1
[image]

Output Part-2
[image]

So why is Sloeber taking up more space than the Arduino IDE? Is there any setting left or any missing steps to do?
As of now, I have to work on the Arduino IDE, but I am willing to work with Sloeber.

Thanks

jantje, 2024-02-07:
>So why is Sloeber taking up more space than the Arduino IDE? 

I have created a Q&A discussion to explain all possible reasons I can think of.
https://github.com/Sloeber/arduino-eclipse-plugin/discussions/1626

The -DDebug in image number 3 may explain it.

I'll add a discussion to describe what to do if all seems equal but there are Arduino IDE versus Sloeber differences.

mkumar2789, 2024-02-07:
Thanks for your reply, Jantje.

I just use -DDEBUG in the "append to" options, as explained in the tutorial, which can be avoided in my case.
And I already deleted unneeded libraries and unneeded boards as well.
One more thing to notice: when I upload code with BluePill F103CB (128K) in "Properties" to the BluePill F103C8 (64K) board, even though code was not responding properly, code uploaded with no error console.
And I'm using the only board URL, officially supported by STMicroelectronics, to minimise random behaviour of code.

ArminJo, 2024-02-19:
The reason for being bigger than Arduino is that arduino introduced new debug target.
so the platform.txt looks now:
```
#compiler.optimization_flags=${compiler.optimization_flags}
compiler.optimization_flags.release=${build.flags.optimize} ${build.flags.debug}
compiler.optimization_flags.debug=-Og -g
```
If you change it to 
```
compiler.optimization_flags=${build.flags.optimize} ${build.flags.debug}
compiler.optimization_flags.release=${build.flags.optimize} ${build.flags.debug}
compiler.optimization_flags.debug=-Og -g
```
the `-Os` contained in build.flags.optimize are now recognized by sloeber and the size is smaller (as in Arduino if all unnecessary libraries are disabled 😀 ).
See also: https://github.com/stm32duino/Arduino_Core_STM32/issues/2159

jantje, 2024-02-20:
Thanks for the info.
I had a watcher on the platform specification but it didn't pick this change up. 😞 
Seems I had a watcher on a specific version 🤡 
This seems to have been introduced 4 year ago. 
I guess I should have a look and see what else Sloeber is missing.

As a **workaround** you have the option to change the txt file and add
compiler.optimization_flags=${compiler.optimization_flags.release}
or you can create the environment variable compiler.optimization_flags in the windows->preferences with the value ${compiler.optimization_flags.release}

MangyDogUK, 2024-08-06:
Can confirm this issue is still happening as of August 2024.
I had to ad -Os to the compile append to C and C++ line to get it to optimise my code. It appears to be ignoring the optimise level argument in the stm32duino project settings.

jantje, 2024-08-06:
I know.
Currently trying to get things back working on Linux.
Then This is on the agenda.
Please join the test effort if you can
https://github.com/Sloeber/arduino-eclipse-plugin/discussions/1660

jantje, 2024-08-13:
Thinking about this.
The easiest workaround is adding ${compiler.optimization_flags.release} to the  the compile append to C and C++ line

jantje, 2024-08-13:
This should be fixed in the nightly (I say should as I don't have a reliable way to test) I tried both V 2.7.1 and V2.8.1
Please join the test effort; more info about the test effort for V5 is here https://github.com/Sloeber/arduino-eclipse-plugin/discussions/1660

closed 2024-08-13 by jantje