Sloeber/arduino-eclipse-plugin #1629: unable to flash attiny84 without bootloader (STK500)

ChristianBrandtner, 2024-02-14:
my enviroment:

[image]
[image]
[image]

there are two problem in the commandline for avrdude:

/bin/avrdude -C/home/cebra/Arduino/ATTinyCore-master/avr/avrdude.conf -pattiny84 **-cstk500v1** -P/dev/ttyACM0 -b115200 **-D** -Uflash:w:/home/cebra/Arduino_workspace/attiny84_blink_v1/Release/attiny84_blink_v1.hex:i 

**-cstk500v1**
in this case, programmer did not respond.

Arduino IDE is using stk500 and is running.
I changed in the preferences the value to stk500, stk500v2 is also running

[image]

after that the result is a verification error:
[image]

the reason for this is the parameter **-D** in the avrdude comandline.

Arduino IDE did not use -D

the comparable Arduino commandline from the Arduino IDE:

/home/cebra/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18/bin/avrdude -C/home/cebra/.arduino15/packages/ATTinyCore/hardware/avr/1.5.2/avrdude.conf -v -pattiny84 -cstk500 -P/dev/ttyACM0 -Uflash:w:/tmp/arduino_build_719538/attiny_V1.ino.hex:i 

Arduino IDE use avrdude 6.3, Sloeber the "system" avrdude 7.1, the problem occurs with both versions.

I did not find a solution to delete this parameter in the projectconfig, sloeberconfig, board.txt or programmer.txt

jantje, 2024-02-14:
1)
[image]
You should not reference arduino installed boards.
Install the boards using the boardsmanger in Sloeber

2) Please provide the project properties->Sloeber 
So I can see the upload configuration you use

3) when you provide log output please provide the whole log not just a part.
You may have pressed the wrong button and I can not tell as the action that is executed is at the start.

ChristianBrandtner, 2024-02-14:
I've been trying to find the problem for 3 days.
I have now found a solution for myself.
1. in enviroment properties, changed stk500v1 to stk500.

2. in the sloeber.platform.txt and platform.txt have the line

**tools.avrdude.upload.pattern="${tools.avrdude.cmd.path}" "-C${tools.avrdude.config.path}" ${upload.verbose} -p${build.mcu} -c${upload.protocol} -P${serial.port} -b${upload.speed} -D "-Uflash:w:${build.path}/${build.project_name}.hex:i"**

changed, deleted -D there.

[Untitled.epf.zip](https://github.com/Sloeber/arduino-eclipse-plugin/files/14284907/Untitled.epf.zip)

My sloeber ide no longer works, so I switched to another Eclipse version with the Sloeber plugin installed. Of which is the preferences file.
This version has the same problem.

I have now installed the boards with the board manager, there was no improvement

jantje, 2024-02-14:
please provide this info
[image]

ChristianBrandtner, 2024-02-14:
[image]

jantje, 2024-02-14:
If you want to use the uploader the programmer protocol should be empty

ChristianBrandtner, 2024-02-15:
it is not possible to leave the programmer protocol empty.

[image]

This is also illogical, an Atmega without a bootloader is usually only flashed with an ISP programmer with STK500 protocol.

**-D** prevents flash memory from being erased before programming.
This is only used with ATxmega.
For all other ATmega, the flash memory must be deleted before programming. Therefore -D cannot be used there.
Why is it configured like this in the platform.txt of Arduino? no idea

But anyway, I'm using my workaround now.
Thanks for the support

closed 2024-02-20 by jantje