Sloeber/arduino-eclipse-plugin #1631: Serial monitor is not cleared after upload on STM32 Arduino core

MasteringTheMess, 2024-02-20:
**Describe your environment** 

latest sloeber 4.4.3 product
OS: Win 11
STM32 Arduino Core 2.7.1: https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json 
[image]

**Describe the bug** 

After upload, the serial monitor is not cleared, even when the checkbox is ticket in _Sloeber_ -> _Preferences_
[image]

I have seen this on a custom board with SWD upload and on a Nucleo32 with mass storage upload

**To Reproduce** 

1. install STM32 Arduino Core
2. create new project with serial output
3. compile, download
4. activate serial port
5.  see output
6.  check _Clean Serial monitor after upload_ and Switch to _Serial monitor after upload_ in the menu _Sloeber_ -> _Preferences_
7. upload project again
8. previous contents is still in serial monitor

jantje, 2024-02-20:
I'll have a look after I finished #1126 (don't hold your breath)

MasteringTheMess, 2024-03-01:
on a fresh install of sloeber & workspace, this seems to not happen anymore...

jantje, 2024-03-01:
I could reproduce it so something is off.

PerennialNovice, 2024-06-03:
I now can confirm that the serial monitor is randomly* cleared or not after upload on windows and on linux installations...

* at least I was not able to find out a rule as to when it is cleared or not. Sometimes it is cleared 5 times in a row, sometimes not at all during maaany uploads. Only thing: it seems to be cleared far less often then not.

jantje, 2024-06-03:
Last times I checked it worked fine.
It looks like this one will be hard to fix.