Sloeber/arduino-eclipse-plugin #1633: The Latest Sloeber Update Removes the Executable

emintz, 2024-02-20:
The latest Sloeber update deletes sloeber-ide. This is a real problem for [Metuchen Momentum](https://www.metuchenmomentum.org/), our local [FIRST Robotics](https://www.firstinspires.org/robotics/frc) team. Competition is two weeks away, so this is urgent.

To reproduce on Ubuntu 22.04:

1.  Download and install the latest Sloeber on Ubuntu 22.04
2.  Move the resulting Sloeber directory into the home directory
3.  Go to the Sloeber directory
4.  ./sloeber-ide
5.  Help>Check for Updates
6.  Leave both updates checked and press Next
7.  Review and confirm the updates and press Next
8.  Accept the license agreement and press Finish
9.  Verify that slober-ide remains in ~/Slober (it does on my system)
10.  On the resulting Trust page, press select all, then Trust Selected 
11.  The "Trust Signers" panel appears. sloeber-ide remains in ~/Slober (I checked). Click Select All then Trust Selected.
12.  sloeber-ide is deleted from ~/Slober

Please advise soonest, and let me know if I can assist in any way.

emintz, 2024-02-20:
PS -- the update also removes slober-ide.ini

jantje, 2024-02-20:
Why do you download and then check for updates?

jantje, 2024-02-20:
Looking at the history I see that the update site didn't work so I had to fix that resulting in the update site having a newer timestamp which triggers the completely unnecessary update.
Why this update deletes the executable and ini file is a different story but I have seen that before. Updating a product has never been without risk and that is also why the update documentation states to take backups of everything before trying to update http://eclipse.baeyens.it/how_to.shtml#/n

jantje, 2024-02-21:
I tried the install update here on windows and that worked.
> I have found that many installs, no matter how scrupulously maintained

I tend to forget to update the update site when making a release. As the nightly is dead by now you are 100% sure there is no need to run the update.

closed 2024-08-19 by jantje