Sloeber/arduino-eclipse-plugin #1637: 4.4.3 compile  error: get a file_opts error (No such file or directory)

guanpengcn, 2024-03-09:
Hi,

I used Sloeber IDE 4.4.3 and added a sketch with empty setup and loop funktion and get a file_opts error (No such file or directory) as well.
My device is ESP32 WROOM and I used ESP32 DEV Module as board in Sloeber IDE.
If I compile I get following error:

... IC:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/qio_qspi/include" -mlongcalls -ffunction-sections -fdata-sections -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -ggdb -freorder-blocks -Wwrite-strings -fstack-protector -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu++11 -fexceptions -fno-rtti  -MMD -c -w -Os -DF_CPU=240000000L -DARDUINO=10812 -DARDUINO_ESP32S3_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32S3_DEV\"" -DARDUINO_VARIANT="esp32s3" -DARDUINO_PARTITION_default  -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1  -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=0 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0 "@C:\Users\Thomas\Documents\sloeber441workspace\Test\Release/file_opts"   -I"C:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14\variants\esp32s3" -I"C:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14\cores\esp32" -I"C:\Sloeber441\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14\libraries\ESP32\src" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"   -o "sloeber.ino.cpp.o" **xtensa-esp32s3-elf-g++: error: C:\Users\Thomas\Documents\sloeber441workspace\Test\Release/file_opts: No such file or directory** make[1]: *** [subdir.mk:20: sloeber.ino.cpp.o] Error 1 make: *** [makefile:97: all] Error 2 "C:/Sloeber441/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

What I am missing please?

jantje, 2024-03-09:
see #1582
As you are using version 2.0.14 I would have thought this would have been fixed.

DiyParts, 2024-03-14:
Hi,
I use Windows 11, newest Update. Sloeber 4.4.3.202402080117. All Eclipse Updates installed. 
**I get the same Make Error with a Blink Excample for Arduino Nano:** avr-gcc: error: .CDC.cpp.o: No such file or directory
**Now the funny part comes: The same statement works on CMD prompt in project Release folder** (see below) 8-o
What to do?

17:21:39 **** Incremental Build of configuration Release for project Test_Sloeber_Blink2 ****
"C:\\Sloeber\\arduinoPlugin\\tools\\make\\make" all 
Building file: ..\Blink.cpp
Starting C++ compile
"C:\Sloeber\arduinoPlugin\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   -I"C:\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\cores\arduino" -I"C:\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\variants\eightanaloginputs" -MMD -MP -MF"Blink.cpp.d" -MT"Blink.cpp.o" -D__IN_ECLIPSE__=1 "..\Blink.cpp" -o "Blink.cpp.o"
Finished building: ..\Blink.cpp
 
Building file: C:\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\cores\arduino\CDC.cpp
Starting C++ compile
...
Finished building: C:\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\cores\arduino\CDC.cpp
...
Starting archiver
"C:\Sloeber\arduinoPlugin\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc-ar" rcs  "C:\sloeber-workspace\Test_Sloeber_Blink2\Release\arduino.ar"  .\core\core\CDC.cpp.o .\core\core\HardwareSerial.cpp.o .\core\core\HardwareSerial0.cpp.o .\core\core\HardwareSerial1.cpp.o .\core\core\HardwareSerial2.cpp.o .\core\core\HardwareSerial3.cpp.o .\core\core\IPAddress.cpp.o .\core\core\PluggableUSB.cpp.o .\core\core\Print.cpp.o .\core\core\Stream.cpp.o .\core\core\Tone.cpp.o .\core\core\USBCore.cpp.o .\core\core\WInterrupts.c.o .\core\core\WMath.cpp.o .\core\core\WString.cpp.o .\core\core\abi.cpp.o .\core\core\hooks.c.o .\core\core\main.cpp.o .\core\core\new.cpp.o .\core\core\wiring.c.o .\core\core\wiring_analog.c.o .\core\core\wiring_digital.c.o .\core\core\wiring_pulse.S.o .\core\core\wiring_pulse.c.o .\core\core\wiring_shift.c.o 
**c:\sloeber\arduinoplugin\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\bin\../lib/gcc/avr/7.3.0/../../../../avr/bin/ar.exe: .corecoreCDC.cpp.o: No such file or directory**
make: *** [makefile:109: C:\sloeber-workspace\Test_Sloeber_Blink2\Release\arduino.ar] Error 1
"C:/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

17:21:44 Build Failed. 1 errors, 4 warnings. (took 4s.337ms)

**CMD Prompt:**
_C:\sloeber-workspace\Test_Sloeber_Blink2\Release>_ del arduino.ar
_C:\sloeber-workspace\Test_Sloeber_Blink2\Release>_"C:\Sloeber\arduinoPlugin\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc-ar" rcs  "C:\sloeber-workspace\Test_Sloeber_Blink2\Release\arduino.ar"  .\core\core\CDC.cpp.o .\core\core\HardwareSerial.cpp.o .\core\core\HardwareSerial0.cpp.o .\core\core\HardwareSerial1.cpp.o .\core\core\HardwareSerial2.cpp.o .\core\core\HardwareSerial3.cpp.o .\core\core\IPAddress.cpp.o .\core\core\PluggableUSB.cpp.o .\core\core\Print.cpp.o .\core\core\Stream.cpp.o .\core\core\Tone.cpp.o .\core\core\USBCore.cpp.o .\core\core\WInterrupts.c.o .\core\core\WMath.cpp.o .\core\core\WString.cpp.o .\core\core\abi.cpp.o .\core\core\hooks.c.o .\core\core\main.cpp.o .\core\core\new.cpp.o .\core\core\wiring.c.o .\core\core\wiring_analog.c.o .\core\core\wiring_digital.c.o .\core\core\wiring_pulse.S.o .\core\core\wiring_pulse.c.o .\core\core\wiring_shift.c.o
_--> runs without error!_
_After running: Folder C:\sloeber-workspace\Test_Sloeber_Blink2\Release_

14.03.2024  17:32    <DIR>          .
14.03.2024  17:21    <DIR>          ..
14.03.2024  17:32           343.200 arduino.ar
...

jantje, 2024-03-14:
Strange; very strange.
Do you have a virus checker running?
I mean in the log you see Sloeber builds CDC.cpp and then it is gone. Why?

DiyParts, 2024-03-15:
I do have windows security running. No actions recorded.
CDC.cpp is not gone. It is just the first file in the string that is not possible to find for avr-gcc-ar.

jantje, 2024-03-15:
I mean:
I mean in the log you see Sloeber builds CDC.cpp to CDC.cpp.o and then CDC.cpp.o is gone. Why?

It is indeed the first file with a relative path and the output (without the slashes -or is that github-) is wierd.
It looks like or something strange is going on on your system (like with virus checkers) or you do something strange (though I see nothing in the information provided)
I just tested here with the latest Sloeber and the blink example. As you can see below it builds fine. The output looks very much like yours as well
[image]
`
14:52:10 **** Build of configuration Release for project issue1637 ****
"C:\\sloeber\\V4.4.3\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"C:\sloeber\V4.4.3\arduinoPlugin\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   -I"C:\sloeber\V4.4.3\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\cores\arduino" -I"C:\sloeber\V4.4.3\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\variants\eightanaloginputs" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 "..\sloeber.ino.cpp" -o "sloeber.ino.cpp.o"
'Finished building: ..\sloeber.ino.cpp'
' '
'Building file: C:\sloeber\V4.4.3\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\cores\arduino\CDC.cpp'
'Starting C++ compile'
"C:\sloeber\V4.4.3\arduinoPlugin\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   -I"C:\sloeber\V4.4.3\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\cores\arduino" -I"C:\sloeber\V4.4.3\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\variants\eightanaloginputs" -MMD -MP -MF"core\core\CDC.cpp.d" -MT"core\core\CDC.cpp.o" -D__IN_ECLIPSE__=1 "C:\sloeber\V4.4.3\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\cores\arduino\CDC.cpp" -o "core\core\CDC.cpp.o"
'Finished building: C:\sloeber\V4.4.3\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\cores\arduino\CDC.cpp'
' '
'Building file: C:\sloeber\V4.4.3\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\cores\arduino\HardwareSerial.cpp'
'Starting C++ compile'
"C:\sloeber\V4.4.3\arduinoPlugin\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   -I"C:\sloeber\V4.4.3\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\cores\arduino" -I"C:\sloeber\V4.4.3\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\variants\eightanaloginputs" -MMD -MP -MF"core\core\HardwareSerial.cpp.d" -MT"core\core\HardwareSerial.cpp.o" -D__IN_ECLIPSE__=1 "C:\sloeber\V4.4.3\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\cores\arduino\HardwareSerial.cpp" -o "core\core\HardwareSerial.cpp.o"
'Finished building: C:\sloeber\V4.4.3\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\cores\arduino\HardwareSerial.cpp'
' '
'Building file: C:\sloeber\V4.4.3\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\cores\arduino\HardwareSerial0.cpp'
'Starting C++ compile'
"C:\sloeber\V4.4.3\arduinoPlugin\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   -I"C:\sloeber\V4.4.3\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\cores\arduino" -I"C:\sloeber\V4.4.3\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\variants\eightanaloginputs" -MMD -MP -MF"core\core\HardwareSerial0.cpp.d" -MT"core\core\HardwareSerial0.cpp.o" -D__IN_ECLIPSE__=1 "C:\sloeber\V4.4.3\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\cores\arduino\HardwareSerial0.cpp" -o "core\core\HardwareSerial0.cpp.o"
'Finished building: C:\sloeber\V4.4.3\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\cores\arduino\HardwareSerial0.cpp'
' '
'Building file: C:\sloeber\V4.4.3\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\cores\arduino\HardwareSerial1.cpp'
'Starting C++ compile'
"C:\sloeber\V4.4.3\arduinoPlugin\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   -I"C:\sloeber\V4.4.3\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\cores\arduino" -I"C:\sloeber\V4.4.3\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\variants\eightanaloginputs" -MMD -MP -MF"core\core\HardwareSerial1.cpp.d" -MT"core\core\HardwareSerial1.cpp.o" -D__IN_ECLIPSE__=1 "C:\sloeber\V4.4.3\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\cores\arduino\HardwareSerial1.cpp" -o "core\core\HardwareSerial1.cpp.o"
'Finished building: C:\sloeber\V4.4.3\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\cores\arduino\HardwareSerial1.cpp'
[... cut ...]
.note.gnu.avr.deviceinfo      64         0
.debug_aranges               104         0
.debug_info                 3607         0
.debug_abbrev               2043         0
.debug_line                 1218         0
.debug_frame                 180         0
.debug_str                  1189         0
.debug_loc                  1167         0
.debug_ranges                 48         0
Total                      10570

'Finished building target: issue1637'
' '

14:52:15 Build Finished. 0 errors, 0 warnings. (took 5s.196ms)

`

DougMorgan, 2024-03-15:
I also get the same error with the 4.4.3 version.   I have not found a way around the error and have tested with several different versions of the ESP32 board package.    No problems with RP2040 or STM boards that I have tried.    

Is there a prior version that works with the ESP32?   

08:19:26 **** Build of configuration Release for project TestEsp32_again ****
"M:\\Sloeber\\arduinoPlugin\\tools\\make\\make" all 
cmd /c if exist "M:\Sloeber_Data\Projects\TestEsp32_again\partitions.csv" COPY /y "M:\Sloeber_Data\Projects\TestEsp32_again\partitions.csv" "M:\Sloeber_Data\Projects\TestEsp32_again\Release\partitions.csv"
cmd /c if not exist "M:\Sloeber_Data\Projects\TestEsp32_again\Release\partitions.csv" if exist "M:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14\variants\adafruit_feather_esp32s3_tft\partitions-4MB-tinyuf2.csv" COPY "M:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14\variants\adafruit_feather_esp32s3_tft\partitions-4MB-tinyuf2.csv" "M:\Sloeber_Data\Projects\TestEsp32_again\Release\partitions.csv"
        1 file(s) copied.
cmd /c if not exist "M:\Sloeber_Data\Projects\TestEsp32_again\Release\partitions.csv" COPY "M:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14\tools\partitions\default.csv" "M:\Sloeber_Data\Projects\TestEsp32_again\Release\partitions.csv"
cmd /c IF EXIST "M:\Sloeber_Data\Projects\TestEsp32_again\bootloader.bin" ( COPY /y "M:\Sloeber_Data\Projects\TestEsp32_again\bootloader.bin" "M:\Sloeber_Data\Projects\TestEsp32_again\Release\TestEsp32_again.bootloader.bin" ) ELSE ( IF EXIST "M:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14\variants\adafruit_feather_esp32s3_tft\bootloader-tinyuf2.bin" ( COPY "M:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14\variants\adafruit_feather_esp32s3_tft\bootloader-tinyuf2.bin" "M:\Sloeber_Data\Projects\TestEsp32_again\Release\TestEsp32_again.bootloader.bin" ) ELSE ( "M:\Sloeber\arduinoPlugin\packages\esp32\tools\esptool_py\4.5.1/esptool.exe" --chip esp32s3 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o "M:\Sloeber_Data\Projects\TestEsp32_again\Release\TestEsp32_again.bootloader.bin" "M:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14\tools\sdk\esp32s3\bin\bootloader_qio_80m.elf" ) )
        1 file(s) copied.
cmd /c if exist "M:\Sloeber_Data\Projects\TestEsp32_again\build_opt.h" COPY /y "M:\Sloeber_Data\Projects\TestEsp32_again\build_opt.h" "M:\Sloeber_Data\Projects\TestEsp32_again\Release\build_opt.h"
cmd /c if not exist "M:\Sloeber_Data\Projects\TestEsp32_again\Release\build_opt.h" type nul > "M:\Sloeber_Data\Projects\TestEsp32_again\Release\build_opt.h" ||:
M:/Sloeber/arduinoPlugin/tools/make/make --no-print-directory main-build
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"M:\Sloeber\arduinoPlugin\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0/bin/xtensa-esp32s3-elf-g++" -DHAVE_CONFIG_H "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE -DIDF_VER="v4.4.6-dirty" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS   "-IM:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/newlib/platform_include" "-IM:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/freertos/include" "-IM:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/freertos/include/esp_additions/freertos" "-IM:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/freertos/port/xtensa/include" "-IM:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/freertos/include/esp_additions" "-IM:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/esp_hw_support/include" "-IM:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/esp_hw_support/include/soc" "-IM:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/esp_hw_support/include/soc/esp32s3" "-IM:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/esp_hw_support/port/esp32s3" "-IM:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/esp_hw_support/port/esp32s3/private_include" "-IM:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/heap/include" "-IM:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/log/include" "-IM:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/lwip/include/apps" "-IM:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/lwip/include/apps/sntp" "-IM:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/lwip/lwip/src/include" "-IM:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/lwip/port/esp32/include" "-IM:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/lwip/port/esp32/include/arch" "-IM:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/soc/include" "-IM:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/soc/esp32s3" "-IM:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/soc/esp32s3/include" "-IM:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/i [... cut ...] oeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14\cores\esp32" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 "..\sloeber.ino.cpp" -o "sloeber.ino.cpp.o"
xtensa-esp32s3-elf-g++: error: M:\Sloeber_Data\Projects\TestEsp32_again\Release/file_opts: No such file or directory
make[1]: *** [subdir.mk:20: sloeber.ino.cpp.o] Error 1
make: *** [makefile:104: all] Error 2
"M:/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

jantje, 2024-03-15:
for file_opts: No such file or directory see https://github.com/Sloeber/arduino-eclipse-plugin/issues/1582

DougMorgan, 2024-03-15:
Thanks for the quick response!!!
I had tried a bunch of the suggestions in that thread last night.  
This morning #1533 worked!

Thanks again !

DiyParts, 2024-03-27:
Hi,
finally... 
I built the same code with Arduino IDE and VSC / PlatformIO and it worked without problems. After a couple of days and booting the system I tried it again with Sloeber and it worked! 
CDC.cpp.o was not gone. The archiver just did not find it these days - what reason ever. With CMD Prompt and the same archiver string it worked. Again:

**CMD Prompt:**
C:\sloeber-workspace\Test_Sloeber_Blink2\Release> del arduino.ar
C:\sloeber-workspace\Test_Sloeber_Blink2\Release>"C:\Sloeber\arduinoPlugin\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc-ar" rcs "C:\sloeber-workspace\Test_Sloeber_Blink2\Release\arduino.ar" .\core\core\CDC.cpp.o .\core\core\HardwareSerial.cpp.o .\core\core\HardwareSerial0.cpp.o .\core\core\HardwareSerial1.cpp.o .\core\core\HardwareSerial2.cpp.o .\core\core\HardwareSerial3.cpp.o .\core\core\IPAddress.cpp.o .\core\core\PluggableUSB.cpp.o .\core\core\Print.cpp.o .\core\core\Stream.cpp.o .\core\core\Tone.cpp.o .\core\core\USBCore.cpp.o .\core\core\WInterrupts.c.o .\core\core\WMath.cpp.o .\core\core\WString.cpp.o .\core\core\abi.cpp.o .\core\core\hooks.c.o .\core\core\main.cpp.o .\core\core\new.cpp.o .\core\core\wiring.c.o .\core\core\wiring_analog.c.o .\core\core\wiring_digital.c.o .\core\core\wiring_pulse.S.o .\core\core\wiring_pulse.c.o .\core\core\wiring_shift.c.o
_--> runs without error!_

Best, Olaf

jantje, 2024-03-29:
Good to hear it works for you.
I'll close the issue

closed 2024-03-29 by jantje