Sloeber/arduino-eclipse-plugin #1638: After updating a core / board to new version, all projects must be manually updated too

ArminJo, 2024-03-14:
sloeber	4.4.3.202402080117

I have many projects with many build configurations using the same core / board version.
After loading a new core / board version, I have to manually change the `Platform Folder :` for each configuration.
Editing .sproject and .project files does not help, after opening the project the `Platform Folder :` still contains the old entry :-(
**Is there another file which must be edited?**
E.g. I want to upgrade from avr/1.8.5 to avr/1.8.6.

jantje, 2024-03-14:
>all projects must be manually updated too

That is actually by design.
Modifying the .sloeber file when the project is closed may work (but I think you wil still have to open the project properties and select apply/close for each configuration)

ArminJo, 2024-03-15:
Hello Jaintje,
thanks for the fast answer.

> That is actually by design.

Maybe you know where  the old Platform Folder value is stored in the design?
Are there some eclipse internal xml files, which contains the information?
I looked at `.metadata\.plugins\org.eclipse.core.resources\.projects\ ` but could not find any significant content 😞.

jantje, 2024-03-15:
Currently the info is stored in the .sproject file (which stands for sloeber project file)
If you share your configuration (see below) the info is also stored in sloeber.cfg (which overrules the .sproject file)
The problem is: A change in these files may -and in case of changing platform version it does- have impact on other areas (like the .project file) The code that does those changes needs to be triggered (which is typically done by a open the project properties->sloeber->apply and close action.
If you use the sloeber.cfg file you can simply edit and safe the file (at least that is the design; I doubt this is fully tested through for all cases) but changes on the .sproject are not picked up.

[image]

Thinking about this ... I think it is very likely that following scenario might work as well in current Sloeber
1) close the project
2) change the .sproject (I would do all configurations in one go)
3) reopen the project
4) change the configuration (this can be done by right clicking the project->Build configurations->set active. This triggers the above mentioned changes )
5) change back to your active configuration.

ArminJo, 2024-03-15:
I found the "bug" 😀 
**The content of a .sloeber.cfg is always taken, even if `share in version control` is not checked!.**
Removing all sloeber.cfg and editing the .sproject and .procect now works reliable at least after a reattach of libraries.

closed 2024-03-15 by ArminJo