Sloeber/arduino-eclipse-plugin #1639: hostname will not be resolved at upload

aber68, 2024-03-17:
hello,

using last sloeber version, ESP32 and OTA for upload ...

.......s\esp32\hardware\esp32\2.0.14/tools/espota.exe -r -i **BEwecker.local** -p 3232 .....

the above hostname will not be recognized from sloeber, but from arduino IDE 2.x 

whats wrong?

code snippet:

  ArduinoOTA.setPassword("1234");
  ArduinoOTA.setPort(3232);
  ArduinoOTA.setHostname("BEwecker");
  ArduinoOTA.setMdnsEnabled(true);
  ArduinoOTA.begin();

best regards
andreas

jantje, 2024-03-17:
The hostname resolution happens at the os level.
In other words for Sloeber "BEwecker.local" is only a string that is put in the command.
The command is run at the os level and the os needs to resolve "BEwecker.local" to a IP address

aber68, 2024-03-17:
thanks for reply, but where does sloeber take this string? i want to have the command -i set with the local IP eg 192.168.0.x

jantje, 2024-03-17:
project properties->sloeber?

aber68, 2024-03-18:
not really, the port shows the address of the ESP device. if you change the line to the IP only, espota will then switch to a local port, which is obviously not available, eg changing "BEwecker.local 192.168.0.50" to "192.168.0.50" will fail ...

but "BEwecker.local" **does** exist, bcs this can be entered in a browser and will be resolved to "192.168.0.50", which is correct

jantje, 2024-03-18:
Have you tried  "192.168.0.50  192.168.0.50"  in case of ip adress it is a 2 field one is the name one is the ip I don't recall the order.

aber68, 2024-03-18:
i think yes, but i will double check in the eve

aber68, 2024-03-18:
your tip did it, thanks! 

yesterday i tried **ArduinoOTA.setHostname("192.168.0.50")** which failed. but the following line is still needed: **ArduinoOTA.setPort(3232)** otherwise sloeber will use port 8266 as default in spite of ESP32 is choosen as plattform.

btw after changing any of the project properties this error comes up: **"...\Release/file_opts: No such file or directory"** (this can only be cleared after creating a "file_opts" file in the release directoray by hand - why is that?

br

jantje, 2024-03-18:
file_opts is a way ESP32 uses to provide the user to add build options to the build command. Something you can do in Sloeber as well.
Using the hooks the file_opts file should be created but Sloeber does not support all hook types and mutch more.
I prefer to remove the file_opts from the command line by modifying the platform.txt file
For more info see see https://github.com/Sloeber/arduino-eclipse-plugin/issues/1582