Sloeber/arduino-eclipse-plugin #1641: Support for International Characters in Serial Monitor View

guanpengcn, 2024-03-20:
Hi, I would like to report an issue with the Serial Monitor View in Sloeber IDE v4.4.3.202402080117. Currently, while receiving debug information from the serial port, the view seems to have limited support for international characters. English text displays correctly, but non-English characters, specifically Chinese characters in my case, appear as gibberish or garbled text.

I have attempted to adjust the encoding settings related to Eclipse (since Sloeber is based on Eclipse), ensuring that UTF-8 is used wherever possible. However, this did not resolve the issue. The Serial Monitor View still fails to properly render Chinese characters.

Could you suggest a workaround or provide a fix for this issue? It would be greatly appreciated if the Serial Monitor View could correctly display international characters, including Chinese, without any encoding issues.

Thank you in advance for your help and support!
[image]

jantje, 2024-03-20:
I have never tested non ASCII text with the serial monitor so it is verry likely to have issues.
One thing to look at though is the hide plotter data. This functionality will do some filtering and that may be the cause.
If it still doesn't work without hiding the plotter data you will have to wait as I'm working on some big changes.
[image]
Keep me posted on whether the hide plotter data works for you

guanpengcn, 2024-03-21:
Thank you for your feedback. I have tried to hide the plotter data in the Serial Monitor View, but the non-ASCII characters still cannot be displayed properly.

[image]

jantje, 2024-03-21:
ok thanks for the update

gayboter49, 2025-11-28:
It may seem an overly radical sugesstion, but perhaps it would be worth considering a re-implementation of the terminal view based on the Eclipse Terminal from the Eclipse Foundation's TM project (https://marketplace.eclipse.org/content/eclipse-terminal) ? It definitely incorporates not only support for all kinds of encodings (see screenshot below, bottom-most option) , but also VT-100 (and alike) terminal functionality which results in a much more "natural" feeling in terms of communication flow by doing away with the out-of-place separate input line above the communication window. Not to mention the possibilities emerging from the terminal emulator functionality which have their counterpart in a number of Arduino libraries that exist exactly for that purpose. 

Of course some (poetntially) minor modifications would be necessary, mainly
1. Adding on-connect and explicit (button-triggered) support for issuing target resets
2. Support for temporary suspension of pending serial connections during firmware uploads (by automatic disconnects/reconnects in order to avoid resource access conflicts on the serial port devices as already implemented in Sloeber's current terminal code)

What do do you think ?

[image]