Sloeber/arduino-eclipse-plugin #1645: Change Default .cpp File???

gfvalvo, 2024-04-24:
When an new Sloeber project is started with the "Default cpp file" option, the resultant code skeleton looks like this:
```
// Do not remove the include below
#include "New_Project.h"

//The setup function is called once at startup of the sketch
void setup()
{
// Add your initialization code here
}

// The loop function is called in an endless loop
void loop()
{
//Add your repeated code here
}
```

The only purpose of the .h file that's created is to `#include "Arduino.h"`. I would prefer to simply do that in the .cpp file. Also, I don't need the pointless comments and don't like the formatting style that puts the opening '{' on a new line. So, can the default cpp file be changed to look like this???
```
#include "Arduino.h"

void setup() {

}

Thanks.
void loop() {

}
```

jantje, 2024-04-28:
>The only purpose of the .h file that's created is to #include "Arduino.h".

I disagree on that one. Your project header should contain the headers you want to include for all your project level files.

>, I don't need the pointless comments

I agree but to some people these are helpful and it avoids issues/discussions for me and saves me time to work on improvements/issues.
Deleting them is easy with ctrl d which deletes the whole line in one go

>and very much dislike the formatting style putting the opening '{' on a new line. So, can I change the default cpp file to look like this???

This is the formatting style I like but eclipse allows you to select the formatting style you like (you can even create your own formatting style). Once you have selected the formatting style you like in the workspace preferences pressing ctrl+f will reformat the file to the formatting style of your choice.

You can also use your own template folder in Sloeber (a folder containing the files which will be copied in the project at the time of project creation)

gfvalvo, 2024-04-30:
Thanks for the reply @jantje. I'm not asking that the default Sloeber .cpp file be changed in the product. I'd just like to change it and eliminate the included .h file in **MY** installation.

Is that possible?

Will this "template folder" thing do it????
> You can also use your own template folder in Sloeber (a folder containing the files which will be copied in the project at the time of project creation)
If so, please let me know the procedure.

Thanks again.

jantje, 2024-05-03:
FYI here are the default template files in the repository
https://github.com/Sloeber/arduino-eclipse-plugin/tree/master/io.sloeber.core/src/io/sloeber/core/templates
You could consider forking the repository modifyingg these files and building your own sloeber version.
Unless you create a ton of new preojects: all of which will take more time than you could ever spend on modifying the "out of the box templates"

jantje, 2024-08-13:
Closing the issue as I think it is solved.
Please reopen if you disagree.

closed 2024-08-13 by jantje

jantje, 2024-08-18:
FYI
I did some changes in this area.
Sketch.cpp files in templates are now programmed to be renamed to [projName].cpp
Subfolders are also copied as part of project creation based on a template.
Fixed in V5.0 however the github build fails for reasons beyond my current knowledge. Should be fixed by the end of next week.