Sloeber/arduino-eclipse-plugin #1649: Build fails with latest arduino-esp32 3.0.0 RC1

kb1sph, 2024-05-14:
After setting up sloeber on a new laptop and installing arduino-esp32, I got the following error.

```
'Building file: C:\Users\kb1sp\OneDrive\Documents\GitHub\arduino-esp32\cores\esp32\chip-debug-report.cpp'
'Starting C++ compile'
"C:\Users\kb1sp\OneDrive\Documents\GitHub\arduino-esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -MMD -c "@C:\Users\kb1sp\OneDrive\Documents\GitHub\arduino-esp32/tools/esp32-arduino-libs/esp32/flags/cpp_flags" -w -Os -DF_CPU=240000000L -DARDUINO=10812 -DARDUINO_LOLIN32 -DARDUINO_ARCH_ARDUINO-ESP32 "-DARDUINO_BOARD=\"LOLIN32\"" -DARDUINO_VARIANT="lolin32" -DARDUINO_PARTITION_min_spiffs -DARDUINO_HOST_OS="" -DARDUINO_FQBN="" -DESP32 -DCORE_DEBUG_LEVEL=0    -DARDUINO_USB_CDC_ON_BOOT=0  "@C:\Users\kb1sp\OneDrive\Documents\GitHub\arduino-esp32/tools/esp32-arduino-libs/esp32/flags/defines" "-IC:\Sloeber\workspace\esp32test" -iprefix "C:\Users\kb1sp\OneDrive\Documents\GitHub\arduino-esp32/tools/esp32-arduino-libs/esp32/include/" "@C:\Users\kb1sp\OneDrive\Documents\GitHub\arduino-esp32/tools/esp32-arduino-libs/esp32/flags/includes" "-IC:\Users\kb1sp\OneDrive\Documents\GitHub\arduino-esp32/tools/esp32-arduino-libs/esp32/dio_qspi/include" -I"C:\Users\kb1sp\OneDrive\Documents\GitHub\arduino-esp32\cores\esp32" -I"C:\Users\kb1sp\OneDrive\Documents\GitHub\arduino-esp32\variants\lolin32" -MMD -MP -MF"core\core\chip-debug-report.cpp.d" -MT"core\core\chip-debug-report.cpp.o" -D__IN_ECLIPSE__=1 "@C:\Sloeber\workspace\esp32test\Release/file_opts" "C:\Users\kb1sp\OneDrive\Documents\GitHub\arduino-esp32\cores\esp32\chip-debug-report.cpp" -o "core\core\chip-debug-report.cpp.o"
C:\Users\kb1sp\OneDrive\Documents\GitHub\arduino-esp32\cores\esp32\chip-debug-report.cpp: In function 'void printSoftwareInfo()':
C:\Users\kb1sp\OneDrive\Documents\GitHub\arduino-esp32\cores\esp32\chip-debug-report.cpp:213:67: error: expected primary-expression before ')' token
  213 |   chip_report_printf("  Compile Host OS   : %s\n", ARDUINO_HOST_OS);
      |                                                                   ^
C:\Users\kb1sp\OneDrive\Documents\GitHub\arduino-esp32\cores\esp32\chip-debug-report.cpp: In function 'void printBoardInfo()':
<command-line>: error: 'lolin32' was not declared in this scope
C:\Users\kb1sp\OneDrive\Documents\GitHub\arduino-esp32\cores\esp32\chip-debug-report.cpp:223:52: note: in expansion of macro 'ARDUINO_VARIANT'
  223 |   chip_report_printf("  Arduino Variant   : %s\n", ARDUINO_VARIANT);
      |                                                    ^~~~~~~~~~~~~~~
C:\Users\kb1sp\OneDrive\Documents\GitHub\arduino-esp32\cores\esp32\chip-debug-report.cpp:225:64: error: expected primary-expression before ')' token
  225 |   chip_report_printf("  Arduino FQBN      : %s\n", ARDUINO_FQBN);
      |                                                                ^
make[1]: *** [core\core\subdir.mk:312: core\core\chip-debug-report.cpp.o] Error 1
make: *** [makefile:103: all] Error 2
"C:/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

00:58:29 Build Failed. 4 errors, 0 warnings. (took 43s.378ms)
```

I realized afterwards that it is due to a newer version of arduino-esp32.

It seems this was noticed by someone else and reported on another repository and fixed.  Perhaps the link to that could help.

https://github.com/plerup/makeEspArduino/issues/200

dimensionv, 2024-06-11:
Will there be an official patch for this? Because the two defines aren't the only things that needed to be "fixed". Until I got my project to compile, it required these defines to be fixed by adding them to the compile-options tab as parameters for the c and c++ compiler calls of the sloeber project options:

`-DARDUINO_HOST_OS="\"generic_os\"" -DARDUINO_FQBN="\"generic_board\"" -DARDUINO_VARIANT="\"adafruit_metro_esp32s3\""`

Please note that the last one is originally added, but without the escaped quotation marks, which leads the build-process to break, as the value of the define is not interpreted as a string, but as symbol that cannot be resolved.

Since the file boards.sloeber.txt, from which this value seems to originate, is autogenerated, I presume it needs to be fixed either there, or wherever the makefiles are created... I haven't analyzed this further, as I was glad that my project was finally compiling.

However, a proper official solution / patch would with the proper fixes in the proper places would be highly appreciated.

jantje, 2024-06-11:
It will be fixed but I'm currently working on a very big change (https://github.com/Sloeber/arduino-eclipse-plugin/tree/%231126_support_for_creating_library_archives) which is done but now I need to fix lots of regression (unrelated things that no longer work) issues.

emintz, 2024-07-01:
I have run into this as well. How can I assist.

Thank you.

jantje, 2024-12-21:
This should be fixed in the nightly. Please test V5 to confirm and reopen if not fixed in V5

closed 2024-12-21 by jantje