Sloeber/arduino-eclipse-plugin #1650: sloeber 4.4.3 - STM32 upload sketch fails

arnacre, 2024-05-18:
First of all, I appreciate very much the Sloeber development environment, but I got stuck while trying to upload a sketch on a STM32 BluePill  board (STM32F103C8T6 original STM chip :-)).

Hereafter a detailed description.

... and thanks for your support 

# Describe your environment
* Kubuntu 22.04 (up-to-date)
* java --version
  openjdk 17.0.10 2024-01-16
  OpenJDK Runtime Environment (build 17.0.10+7-Ubuntu-122.04.1)
  OpenJDK 64-Bit Server VM (build 17.0.10+7-Ubuntu-122.04.1, mixed mode, sharing)

* Installed STM32CubeProgrammer in the default installation directory 
  $HOME//STMicroelectronics/STM32Cube/STM32CubeProgrammer/ *

* Sloeber
  Downloaded Linux 4.4.3 Latest Stable, Product bundle
  sloeber	4.4.3.202402070201	io.sloeber.product	null

* Platform:
  Installed STM32 platform 
 https://raw.githubusercontent.com/stm32duino/BoardManagerFiles/main/package_stmicroelectronics_index.json  

# Project setup
* Created new sketch from template Blink
  _Platform folder_: STM32Microelectronics/hardware/stm32/2.7.1
  _Board_: Generich STM32F1 series
  _Port_: <empty>
  _U(S)ART support_: Enabled (no generic 'Serial')
  _Board part number_: BluePill F103C8
  _Optimize_: Debug (-og)
  _USB Speed_: Low/Full Speed
  _USB support_: None
  _Debug symbols and core logs_: Core logs and symbols enabled (-g)
  _Upload method_: SMT32CubeProgrammer (SWD)
  _C Runtime library_: Newlib Nano (Default)

# Describe the bug
**After successful compilation**
    Mouse Right-click project -> Build project

**Pressed: Upload**
At the pop-up question: "Do you want to build before upload?" pressed "YES"

**New pop-up appears:**
[image]

**On terminal running sloeber I get the following exception**
_!ENTRY io.sloeber.arduino.core 4 0 2024-05-18 11:54:47.170
!MESSAGE Failed to run the Upload recipe 
!STACK 0
java.io.IOException: Cannot run program "" (in directory "/home/user/sloeber-workspace"): error=2, No such file or directory
        at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1143)
        at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1073)
        at io.sloeber.core.tools.ExternalCommandLauncher.launch(ExternalCommandLauncher.java:211)
        at io.sloeber.core.tools.uploaders.UploadSketchWrapper$UploadJobWrapper.actualUpload(UploadSketchWrapper.java:253)
        at io.sloeber.core.tools.uploaders.UploadSketchWrapper$UploadJobWrapper.run(UploadSketchWrapper.java:177)
        at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)
Caused by: java.io.IOException: error=2, No such file or directory
        at java.base/java.lang.ProcessImpl.forkAndExec(Native Method)
        at java.base/java.lang.ProcessImpl.<init>(ProcessImpl.java:314)
        at java.base/java.lang.ProcessImpl.start(ProcessImpl.java:244)
        at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1110)
        ... 5 more

!ENTRY io.sloeber.arduino.core 4 0 2024-05-18 11:54:47.172
!MESSAGE failed to upload "blink2"_

**In the sloeber/eclipse console, the following message appears**
Starting upload
Uploading project "blink2" with "stm32CubeProg"

Launching:  sh /home/user/eclipse-sloeber/Sloeber/arduinoPlugin/packages/STMicroelectronics/tools/STM32Tools/2.2.1/stm32CubeProg.sh 0 /home/user/sloeber-workspace/blink2/Release/blink2.bin 0x0 
Output:
Cannot run program "" (in directory "/home/user/sloeber-workspace"): error=2, No such file or directory
failed to upload "blink2"

**From terminal I manually entered the command**
sh /home/user/eclipse-sloeber/Sloeber/arduinoPlugin/packages/STMicroelectronics/tools/STM32Tools/2.2.1/stm32CubeProg.sh 0 /home/user/sloeber-workspace/blink2/Release/blink2.bin 0x0

**... the upload run successfully**

**After resetting the BluePill the Blink program runs correctly (Led on / delay / off)**

jantje, 2024-05-18:
>Uploading project "blink2" with "stm32CubeProg"

It looks like you tried to upload using the programmer stm32CubeProg and that programmer doesn't have a program defined in your environment.
Can you provide this screenshot (notice the programmer should be empty)
[image]
or the file .sproject in the root of your project (Eclipse filters these in the gui)

arnacre, 2024-05-18:
Hi @jantje, thanks for your prompt reply ... 
The *PROGRAMMER" section in the .sproject file seems to be empty, as you suspected.

Here the **screenshot**

[image]

**and the content of th .sproject file**
_Config.Release.board.BOARD.ID=GenF1
Config.Release.board.BOARD.MENU.dbg=enable_all
Config.Release.board.BOARD.MENU.opt=ogstd
Config.Release.board.BOARD.MENU.pnum=BLUEPILL_F103C8
Config.Release.board.BOARD.MENU.rtlib=nano
Config.Release.board.BOARD.MENU.upload_method=swdMethod
Config.Release.board.BOARD.MENU.usb=none
Config.Release.board.BOARD.MENU.xserial=none
Config.Release.board.BOARD.MENU.xusb=FS
Config.Release.board.BOARD.TXT=${SLOEBER_HOME}/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.7.1/boards.txt
Config.Release.board.PROGRAMMER.NAME=
Config.Release.board.UPLOAD.PORT=
Config.Release.compile.sloeber.extra.all=
Config.Release.compile.sloeber.extra.archive=
Config.Release.compile.sloeber.extra.assembly=
Config.Release.compile.sloeber.extra.c.compile=
Config.Release.compile.sloeber.extra.compile=
Config.Release.compile.sloeber.extra.cpp.compile=
Config.Release.compile.sloeber.extra.link=
Config.Release.compile.sloeber.size.custom=
Config.Release.compile.sloeber.size.type=RAW_RESULT
Config.Release.compile.sloeber.warning_level=NONE
Config.Release.compile.sloeber.warning_level.custom=
Config.Release.other.IS_VERSION_CONTROLLED=false_

jantje, 2024-05-18:
There is clearly a inconsistency in the gui versus what is in the .sproject file.
Mostly this is because the version in the .sproject file is not available
The .sproject file states ${SLOEBER_HOME}/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.7.1/boards.txt
Best is to install version STMicroelectronics->stm32-> 2.7.1 
If it is installed try re selecting it in the gui.
Something is off on your system in regards to ${SLOEBER_HOME}/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.7.1/boards.txt

arnacre, 2024-05-19:
@jantje thank you again for your suggestion.
I tried both your suggestions, but it didn't work. Always the same issue.
The file "${SLOEBER_HOME}/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.7.1/boards.txt" is present.

I have no clue what it is wrong with the boards.txt file.
The file  "${SLOEBER_HOME}/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.7.1/boards.txt" is identical to the file in STM32duino repository on github.
I have no further ideas on what I should do.

jantje, 2024-05-19:
Is the json file you used to install available and parsed correctly?

arnacre, 2024-05-19:
I had to shutdown my PC and it seems, that my last comment was lost.
Just to summarize, the file in  .sproject is available and corresponds to the file (boards) on stm3duino github.
I tried both re-install and re-select but with no success.
I have no clue what can be wrong about the boards.txt file.

AFAIK the jason file (https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json) was correctly parsed.
All boards listed in the json file are available when asked to provide the Arduino information while generating a new sketch.
I made a couple of checks for the Board part number (and Generic F103, Blue Pill and F4411CCU x), for what I can see the information in the json file was interpreted and is available. Also the tools have been installed (I checked xpack-arm-none-eabi-gcc and xpack-openocd). By running them from a terminal/console also the versions are correct.

The code (always from the blink template), generated and especially the Arduino.h file correspond to the configuration. The compiled code  is correct and the executable can be successfully downloaded either by the command (see my starting comment) or directly using STM32CubeProgrammer; the code executes correctly.

arnacre, 2024-05-20:
@jantje just for your  information.
I succeeded downloading and running (debugging) the "blink" project by installing the Eclipse IDE for C/C++ Developers and following the [How to debug](https://github.com/stm32duino/Arduino_Core_STM32/wiki/How-to-debug)  of the Arduino_Core_STM32 Wiki with the installation of the Sloeber plug-in.

This time, and just to try, I used  the platform board from STMicrolectronics v 2.0.0 and not the  2.7.1, as I initially did.

After the first debug session, the "Upload sketch" **worked**.
**Next**: I will try to further investigate the differences between 2.0.0 and 2.7.1

**More info ....**
The sections v2.0.0 and 2.7.1 in the json file differ.
A part from "more boards" defined in v2.7.1 the main differences are in the tools used
[image]

Here the console log after having pressed the "Upload sketch":

_Starting upload
Uploading project "blink" with "stm32CubeProg"

Launching: /home/cre/eclipse/eclipse-cpp/eclipse/arduinoPlugin/packages/STMicroelectronics/tools/STM32Tools/2.0.0/linux/stm32CubeProg.sh 0 /home/cre/eclipse/workspace-eclipse-cpp/blink/Release/blink.bin -g 
Output:
      -------------------------------------------------------------------
                        STM32CubeProgrammer v2.16.0                  
      -------------------------------------------------------------------

ST-LINK SN  : 3B0308157216303030303032
ST-LINK FW  : V2J43S7
Board       : --
Voltage     : 3.18V
SWD freq    : 4000 KHz
Connect mode: Under Reset
Reset mode  : Hardware reset
Device ID   : 0x410
Revision ID : Rev X
Device name : STM32F101/F102/F103 Medium-density
Flash size  : 64 KBytes
Start Address : 8000000
Device type : MCU
Device CPU  : Cortex-M3
BL Version  : --

Memory Programming ...
Opening and parsing file: blink.bin
  File          : blink.bin
  Size          : 11.14 KB 
  Address       : 0x08000000 

Erasing memory corresponding to segment 0:
Erasing internal memory sectors [0 11]
Download in Progress:

File download complete
Time elapsed during download operation: 00:00:00.791

RUNNING Program ... 
  Address:      : 0x8000000
Application is running, Please Hold on...
Start operation achieved successfully
The execution of command "stm32CubeProg.sh" is done._

arnacre, 2024-05-20:
@jantje I think I finally got the problem.

The sections "# STM32CubeProgrammer upload" for 2.0.0 and 2.7.1 in the default platform.txt (from stm32duino packge) differ.
The paths (tools.stm32CubeProg.path.) are missing in the 2.7.1 (see pdf document below).

Just a question, is it enough if I simply modify the platform.sloeber.txt  by adding the missing paths and restart Sloeber?

Thank you again 

[stm32duino-platform-diff.pdf](https://github.com/Sloeber/arduino-eclipse-plugin/files/15375556/stm32duino-platform-diff.pdf)

jantje, 2024-05-20:
I do not think these differences can explain the issue you have.
You should not modify the Sloeber.txt files as any upgrade can delete your changes.
Here is how to do .txt changes
https://github.com/Sloeber/arduino-eclipse-plugin/discussions/1533

This is your problem
[image]
First of the boards.txt in tools should never be selected (I deleted this boards.txt file on my system)
If the information here is incomplete Sloeber is confused or not configured correctly. Typically this is caused by missing platform versions.
Easiest way out is to simply reenter the information in the dialog and select apply/close
It never ever happened to me that I fixed the setup and reopened the dialog and it was "wrong" again.

arnacre, 2024-05-20:
Hi @jantje 
The problem is clear now (see below). It is not a Sloeber's problem;  it lies in the configuration of the STM32duino environment.
Thank you again for your support.
Please, feel free to close this issue.

The platform **2.0.0** has the following environment variable set:

**_tools.stm32CubeProg.upload.pattern_**="${tools.stm32CubeProg.path}/${tools.stm32CubeProg.cmd}" ${upload.protocol} "${build.path}/${build.project_name}.bin" ${upload.options}

while platform **2.7.1** has:

**_tools.stm32CubeProg.upload.pattern_**=**"${tools.stm32CubeProg.busybox}" sh** "${tools.stm32CubeProg.path}/${tools.stm32CubeProg.cmd}" ${upload.protocol} "${build.path}/${build.project_name}.bin" ${build.flash_offset} ${upload.options}

with  
_tools.stm32CubeProg.busybox_=

In **2.7.1** by changing the environment variable to: 

**_tools.stm32CubeProg.upload.pattern_**=**sh** "${tools.stm32CubeProg.path}/${tools.stm32CubeProg.cmd}" ${upload.protocol} "${build.path}/${build.project_name}.bin" ${build.flash_offset} ${upload.options}

i.e. by removing **"${tools.stm32CubeProg.busybox}"**
The sketch upload works perfectly.

By removing also the call to "sh"  and by changing the rights of stm32CubeProgrammer.sh in the STM32Tools directory to u:rwx,g:r-x,o:w-x. The sketch upload also works.

closed 2024-05-23 by arnacre