Sloeber/arduino-eclipse-plugin #1651: fresh install on Linux gives Error: Program "make" not found in PATH

PerennialNovice, 2024-05-28:
I have downloaded the latest product and unzip on a fresh installation of Ubuntu

When I try to build a project, I always get "make not found" error
I have tried for STM32 and for AVR - same error

I don't understand what is wrong, because I recently did the very same on a fresh install of Debian 12 without problems

PerennialNovice, 2024-05-29:
Oh, missed that one...

Thank you very much!

closed 2024-05-29 by jantje

ArminJo, 2024-08-19:
A fresh install of V5 also results in:
```
10:43:03 **** Clean-only build of configuration Release for project JK-BMSToPylontechCAN_MightyCore ****
make clean 
Cannot run program "make": Launching failed

Error: Program "make" not found in PATH
```
On my 4.4.3 installations is runs seamlessly with:
```
10:56:00 **** Clean-only build of configuration Release for project JK-BMSToPylontechCAN_MightyCore ****
"E:\\Elektro\\Sloeber\\arduinoPlugin\\tools\\make\\make" clean 
```