Sloeber/arduino-eclipse-plugin #1652: Build Fails with Sloeber 4.3.3 Product and Arduino ESP32 Core 3.0.1.

gfvalvo, 2024-06-11:
Sigh … just added Adruino ESP32 Core 3.0.1 to my Sloeber 4.3.3 Product installation and building of a blank project (using “Default cpp file” option) fails. This seems related to [#200](https://github.com/plerup/makeEspArduino/issues/200) and [#1649](https://github.com/Sloeber/arduino-eclipse-plugin/issues/1649), but they don't provide enough proper instructions for fixing it.

Could someone please write down clear and complete instructions for a workaround until this is fixed in Sleober? Include:
* Exact locations of files to be modified
* Exact names of files to be modified
* Exact changes (deletions / additions) to those files

Thanks!!!

msebas, 2024-06-11:
Well I fixed my build by adding/changing the following build Variables (Project-RMB->Properties->C/C++ Build->Environment):

* `build.fqbn`: `\"ESP32:esp32:esp32\"`
* `build.fqbn`: `\"nodemcu-32s\"` (This one was present, I only added the `\"` to get a String after the preprocessoe inserted it
* `runtime.os`: `\"generic_os\"`

Hope this is what you were looking form.

gfvalvo, 2024-06-11:
> 
> Hope this is what you were looking form.

WOW! I don't know what any of that meant, but it worked. Thanks!!!

GitHubLionel, 2024-06-23:
Hi 
Same problems for me and the workaround of msebas work fine.
Just the `build.variant` is different for me : it is just `\"esp32\"`

emintz, 2024-07-06:
Ditto. I think (but am not sure) that these settings are only used to format diagnostic messages.

I used:

`build.fqbn` :  `\"ESP32:esp32:esp32\"`
`build.variant`: `\"esp32s\"`
`runtime.os`: `\"FreeRTOS\"`

Pro tips for ESP32 developers:

Use the library manager to hide the Arduino FreeRTOS  and I2C libraries. If they are present, Slober automatically includes them in projects (at least it does for me), and compilation fails due to undefined symbols.

emintz, 2024-07-18:
I opened the project environment by right clicking on the project and selecting Properties > C/C++ Build > Environment and found ARDUINO_FQBN and ARDIONO_HOST_OS are defined by:

-DARDUINO_HOST_OS="${runtime.os}" -DARDUINO_FQBN="${build.fqbn}" -DESP32 -DCORE_DEBUG_LEVEL=${build.code_debug} ${build.loop_core} ${build.event_core} ${build.defines} ${build.extra_flags.${build.mcu}} ${build.zigbee_mode}

ARDUINO_VARIANT is defined by:

"${compiler.path}${compiler.c.cmd}" ${compiler.S.extra_flags} ${compiler.S.flags} -DF_CPU=${build.f_cpu} -DARDUINO=${runtime.ide.version} -DARDUINO_${build.board} -DARDUINO_ARCH_${build.arch} "-DARDUINO_BOARD=\"${build.board}\"" -DARDUINO_VARIANT="${build.variant}" -DARDUINO_PARTITION_${build.partitions} ${build.extra_flags} ${compiler.cpreprocessor.flags} ${FLAGS} -D__IN_ECLIPSE__=1 "@${file_opts.path}" ${INPUTS} -o ${OUTPUT} ${sloeber.extra.assembly} ${sloeber.extra.all}

The compiler finds the following errors:

/home/eric/Sloeber/arduinoPlugin/packages/esp32/hardware/esp32/3.0.2/cores/esp32/chip-debug-report.cpp:215:67: error: expected primary-expression before ')' token
  215 |   chip_report_printf("  Compile Host OS   : %s\n", ARDUINO_HOST_OS);
      |                                                                   ^
/home/eric/Sloeber/arduinoPlugin/packages/esp32/hardware/esp32/3.0.2/cores/esp32/chip-debug-report.cpp: In function 'void printBoardInfo()':
<command-line>: error: 'node32s' was not declared in this scope
/home/eric/Sloeber/arduinoPlugin/packages/esp32/hardware/esp32/3.0.2/cores/esp32/chip-debug-report.cpp:225:52: note: in expansion of macro 'ARDUINO_VARIANT'
  225 |   chip_report_printf("  Arduino Variant   : %s\n", ARDUINO_VARIANT);
      |                                                    ^~~~~~~~~~~~~~~
/home/eric/Sloeber/arduinoPlugin/packages/esp32/hardware/esp32/3.0.2/cores/esp32/chip-debug-report.cpp:227:64: error: expected primary-expression before ')' token
  227 |   chip_report_printf("  Arduino FQBN      : %s\n", ARDUINO_FQBN);

These imply:

  ARDUINO_HOST_OS is undefined
  ARDUINO_VARIANT is defined as node32s. It needs to be \"node32s\"
  ARDUINO_FQBN is undefined.

Changint build.fqbn to \"\" fixed ARDUINO_FQBN
Changing build.variant to \"node32s\" defined ARDUINO_VARIANT properly
Changing runtime.os to \"FreeRTOS\" fixed ARDUINO_HOST_OS

Updating the environment automatically added the FreeRTOS and nI2C_master libraries to the project. Both libraries target AtMel processors, so they caused the build to fail. I used the library manager to remove the FreeRTOS library, but haven't yet been able to remove nI2C_master.

Searching the code for build.variant in hopes of finding where its value is set and found:

eric@mighty-wurlitzer:~/Downloads/from_git/sloeber$ grep -r build.variant *
io.sloeber.core/src/io/sloeber/core/api/Json/ArduinoPlatformVersion.java:                installPath.append(VARIANTS_FOLDER_NAME + "/{build.variant}")); //$NON-NLS-1$
io.sloeber.core/src/io/sloeber/core/txt/WorkAround.java:        platformTXT = platformTXT.replace(" -DVARIANT_H=\"{build.variant_h}\"",
io.sloeber.core/src/io/sloeber/core/txt/WorkAround.java:                " \"-DVARIANT_H=\\\"{build.variant_h}\\\"\"");

These show where the value is used. If I could find where it is set, I could probably fix all three.

Can you advise, please?

jantje, 2024-07-20:
Can you provide a screenshot of these environment variables (they absolutely make no sense to me) and off the project properties->sloeber?

emintz, 2024-07-21:
Will do, but a bit of background might make things even clearer. ESP made two breaking changes in version 3 of its libraries, one in the Non-Volatile Storage API, the other in the core library. The change to the core library broke Sloeber.

The newest library contains chip-debug-report.cpp, which was added at version 3. This file requires the following preprocessor symbols: 

  ARDUINO_VARIANT: the board name. This must be a string literal, e.g. "node32s". Sloeber currently defines it without the surrounding quotes.

  ARDUINO_FQBN: I have no idea what this means. Like ARDUINO_VARIANT, it must be a string literal.

  ARDUINO_HOST_OS: the operating system running on the ESP32. Like all the others, it must be a string literal.

I reverse engineered all this by backtracking from the compiler options to the environment variables. When I changed the environment as described above, the compile succeeded and everything worked just fine.

Please let me know if you want a screenshot of the environment. 

  

[image]
[image]
[image]

jantje, 2024-07-22:
Ok now I understand what you meant.
ESP32 has done some wierd stuff in the past and the latest version does not run "out of the box".

>ARDUINO_VARIANT: the board name. This must be a string literal, e.g. "node32s". Sloeber currently defines it without the surrounding quotes.

IMHO Having the surrounding quotes on the command line in Sloeber is going to be very hard.

emintz, 2024-07-22:
I managed to add it manually. If you could tell me where the code and point me to the relevant Eclipse internal documentation, I'll try to fix it this week. Otherwise, I'll need more time.

[image]

jantje, 2024-07-22:
> I'll try to fix it

This won't be that easy as I'm rewriting a part of Sloeber that is determining this behaviour and is not yet finished.
Changes applied to the current master won't be easily cherry picked and the new version is not 100% compatible and not yet finished.

emintz, 2024-07-22:
Oh. Perhaps I could prepare to help maintain the new version. Can you suggest any documentation I could read to get ready to assist?

jantje, 2024-07-22:
Help is always welcome.
However there is hardly any documentation. The only documentation of good quality is the Arduino framework documentation explaining how to define a product version the boardsmanager can handle; how to interpret the .txt files  ...
Nothing of this is sloeber nor eclipse specific but it does explain part of what Sloeber is trying to do.
The interesting pages are
https://arduino.github.io/arduino-cli/0.36/platform-specification/
https://arduino.github.io/arduino-cli/0.36/package_index_json-specification/
https://arduino.github.io/arduino-cli/0.36/library-specification/

jantje, 2024-08-19:
https://github.com/Sloeber/arduino-eclipse-plugin/issues/1652#issuecomment-2211816285

>Use the library manager to hide the Arduino FreeRTOS and I2C libraries. If they are present, Slober automatically includes them in projects (at least it does for me), and compilation fails due to undefined symbols.

Indeed sloeber does so.
The reason is the following line in the esp32 Arduino.h (and many other files)
`#include "freertos/FreeRTOS.h"`
https://github.com/espressif/arduino-esp32/blob/def319add8392d82892386a340f61580e708eca1/cores/esp32/Arduino.h#L33
Nor I nor the CDT indexer think the file is part of the buildpath (while it defeneatly is) and therefore FreeRTOS.h is marked as missing include and the FreeRTOS library is added.

Anyway; I hardcoded to ignore freertos when ESP32 unresolved includes are searched in V5.
Seems I2C is not included automagically on my system. Please advice if the lib is still added.

jantje, 2024-12-21:
Closing as fixed in the nightly

closed 2024-12-21 by jantje