Sloeber/arduino-eclipse-plugin #1653: Arduino CMSIS DSP library does not compile with SAMD51 / Adafruit ItsyBitsy M4

ElsomaUwe, 2024-06-22:
**Describe your environment**   
Using sloeber 4.4.3 product

OS: Win11

The url of the json that you used to download the platform   
https://adafruit.github.io/arduino-board-index/package_adafruit_index.json

**Describe the bug**   
I want to use CMSIS DSP functions.
after installing Arduino CMSIS DSP library build fails for SAMD51 boards like Adafruit ItsyBitsy M4 and Metro Grand Central also after clean, rebuild, reatach libraries...
doing the same with Arduino GUI workes fine.

14:16:29 **** Incremental Build of configuration Release for project IIM3k ****
"C:\\Sloeber\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\IIM3k.cpp'
'Starting C++ compile'
"C:\Sloeber\arduinoPlugin\packages\adafruit\tools\arm-none-eabi-gcc\9-2019q4/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g -Os -Werror=return-type -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -D__SKETCH_NAME__="""IIM3k""" -DF_CPU=120000000L -DARDUINO=10812 -DARDUINO_GRAND_CENTRAL_M4 -DARDUINO_ARCH_SAMD -DARDUINO_SAMD_ADAFRUIT  -D__SAMD51P20A__ -DADAFRUIT_GRAND_CENTRAL_M4 -D__SAMD51__ -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DUSB_VID=0x239A -DUSB_PID=0x8031 -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Grand Central M4\""  -g "-IC:\Sloeber\arduinoPlugin\packages\adafruit\hardware\samd\1.7.13/libraries/Adafruit_TinyUSB_Arduino/src/arduino" -DENABLE_CACHE -g -O2  -DVARIANT_QSPI_BAUD_DEFAULT=50000000 -D__SAMD51P20A__ -DADAFRUIT_GRAND_CENTRAL_M4 -D__SAMD51__ -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DUSB_VID=0x239A -DUSB_PID=0x8031 -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Grand Central M4\""  -g "-IC:\Sloeber\arduinoPlugin\packages\adafruit\hardware\samd\1.7.13/libraries/Adafruit_TinyUSB_Arduino/src/arduino" "-IC:\Sloeber\arduinoPlugin\packages\adafruit\tools\CMSIS\5.4.0/CMSIS/Core/Include/" "-IC:\Sloeber\arduinoPlugin\packages\adafruit\tools\CMSIS\5.4.0/CMSIS/DSP/Include/" "-IC:\Sloeber\arduinoPlugin\packages\adafruit\tools\CMSIS-Atmel\1.2.2/CMSIS/Device/ATMEL/" -I"C:\Sloeber\arduinoPlugin\packages\adafruit\hardware\samd\1.7.13\cores\arduino" -I"C:\Sloeber\arduinoPlugin\packages\adafruit\hardware\samd\1.7.13\variants\grand_central_m4" -I"C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src" -MMD -MP -MF"IIM3k.cpp.d" -MT"IIM3k.cpp.o" -D__IN_ECLIPSE__=1 "..\IIM3k.cpp" -o "IIM3k.cpp.o"
'Finished building: ..\IIM3k.cpp'
' '
'Building file: C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S'
'Starting S compile'
"C:\Sloeber\arduinoPlugin\packages\adafruit\tools\arm-none-eabi-gcc\9-2019q4/bin/arm-none-eabi-gcc" -c -g -x assembler-with-cpp -MMD -DF_CPU=120000000L -DARDUINO=10812 -DARDUINO_GRAND_CENTRAL_M4 -DARDUINO_ARCH_SAMD -DARDUINO_SAMD_ADAFRUIT  -D__SAMD51P20A__ -DADAFRUIT_GRAND_CENTRAL_M4 -D__SAMD51__ -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DUSB_VID=0x239A -DUSB_PID=0x8031 -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Grand Central M4\""  -g "-IC:\Sloeber\arduinoPlugin\packages\adafruit\hardware\samd\1.7.13/libraries/Adafruit_TinyUSB_Arduino/src/arduino" -DENABLE_CACHE "-IC:\Sloeber\arduinoPlugin\packages\adafruit\tools\CMSIS\5.4.0/CMSIS/Core/Include/" "-IC:\Sloeber\arduinoPlugin\packages\adafruit\tools\CMSIS\5.4.0/CMSIS/DSP/Include/" "-IC:\Sloeber\arduinoPlugin\packages\adafruit\tools\CMSIS-Atmel\1.2.2/CMSIS/Device/ATMEL/" -I"C:\Sloeber\arduinoPlugin\packages\adafruit\hardware\samd\1.7.13\cores\arduino" -I"C:\Sloeber\arduinoPlugin\packages\adafruit\hardware\samd\1.7.13\variants\grand_central_m4" -I"C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src" -MMD -MP -MF"libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S.d" -MT"libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S.o" -D__IN_ECLIPSE__=1 "C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S" -o "libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S.o"
C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S: Assembler messages:
C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S:145: Error: selected processor does not support `it LS' in Thumb mode
C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S:146: Error: thumb conditional instruction should be in IT block -- `bxls lr'
C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S:147: Error: cannot honor width suffix -- `push {r4-r9}'
C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S:151: Error: cannot honor width suffix -- `ldrh r8,[r1,#4]'
C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S:152: Error: cannot honor width suffix -- `ldrh r9,[r1,#2]'
C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S:154: Error: cannot honor width suffix -- `ldrh r12,[r1,#-2]'
C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S:157: Error: cannot honor width suffix -- `add r2,r0,r2'
C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S:159: Error: cannot honor width suffix -- `ldr r7,[r9,#0]'
C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S:160: Error: cannot honor width suffix -- `ldr r6,[r8,#0]'
C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S:162: Error: cannot honor width suffix -- `ldr r4,[r12,#0]'
C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S:163: Error: cannot honor width suffix -- `str r6,[r9,#0]'
C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S:164: Error: cannot honor width suffix -- `str r7,[r8,#0]'
C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S:165: Error: cannot honor width suffix -- `str r5,[r12,#0]'
C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S:167: Error: cannot honor width suffix -- `ldr r7,[r9,#4]'
C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S:168: Error: cannot honor width suffix -- `ldr r6,[r8,#4]'
C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S:170: Error: cannot honor width suffix -- `ldr r4,[r12,#4]'
C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S:171: Error: cannot honor width suffix -- `str r6,[r9,#4]'
C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S:172: Error: cannot honor width suffix -- `str r7,[r8,#4]'
C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S:173: Error: cannot honor width suffix -- `str r5,[r12,#4]'
C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S:178: Error: cannot honor width suffix -- `pop {r4-r9}'
C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S:185: Error: selected processor does not support `it LS' in Thumb mode
C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S:186: Error: thumb conditional instruction should be in IT block -- `bxls lr'
C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S:187: Error: cannot honor width suffix -- `push {r4-r9}'
C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S:191: Error: cannot honor width suffix -- `ldrh r8,[r1,#4]'
[... cut ...]
C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S:210: Error: cannot honor width suffix -- `pop {r4-r9}'
make: *** [libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\subdir.mk:145: libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S.o] Error 1
"C:/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

14:16:33 Build Failed. 38 errors, 0 warnings. (took 3s.970ms)

jantje, 2024-07-06:
I assume something is off with the assembly command.
There are some new features in Arduino IDE that have not yet made it into Sloeber which may explain this.
First step is to compare the compile command 
`"C:\Sloeber\arduinoPlugin\packages\adafruit\tools\arm-none-eabi-gcc\9-2019q4/bin/arm-none-eabi-gcc" -c -g -x assembler-with-cpp -MMD -DF_CPU=120000000L -DARDUINO=10812 -DARDUINO_GRAND_CENTRAL_M4 -DARDUINO_ARCH_SAMD -DARDUINO_SAMD_ADAFRUIT -D__SAMD51P20A__ -DADAFRUIT_GRAND_CENTRAL_M4 -D__SAMD51__ -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DUSB_VID=0x239A -DUSB_PID=0x8031 -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER="Adafruit"" "-DUSB_PRODUCT="Grand Central M4"" -g "-IC:\Sloeber\arduinoPlugin\packages\adafruit\hardware\samd\1.7.13/libraries/Adafruit_TinyUSB_Arduino/src/arduino" -DENABLE_CACHE "-IC:\Sloeber\arduinoPlugin\packages\adafruit\tools\CMSIS\5.4.0/CMSIS/Core/Include/" "-IC:\Sloeber\arduinoPlugin\packages\adafruit\tools\CMSIS\5.4.0/CMSIS/DSP/Include/" "-IC:\Sloeber\arduinoPlugin\packages\adafruit\tools\CMSIS-Atmel\1.2.2/CMSIS/Device/ATMEL/" -I"C:\Sloeber\arduinoPlugin\packages\adafruit\hardware\samd\1.7.13\cores\arduino" -I"C:\Sloeber\arduinoPlugin\packages\adafruit\hardware\samd\1.7.13\variants\grand_central_m4" -I"C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src" -MMD -MP -MF"libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S.d" -MT"libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S.o" -D__IN_ECLIPSE__=1 "C:\Users\Uwe\Documents\Arduino\libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S" -o "libraries\Arduino_CMSIS-DSP\src\Source\TransformFunctions\arm_bitreversal2.S.o"`

To what Arduino IDE produces

jantje, 2024-08-19:
I just tried to compile this board in V5 and that seems to work.
So I think my assumption in the previous post is correct

closed 2024-08-19 by jantje