Sloeber/arduino-eclipse-plugin #1656: Doxygen documentation and Sloeber - howto?

DocAlex, 2024-07-12:
Hi,

I see Doxygen as a very good possibility for code documentation. I have sloeber running, I have Doxygen running. 

Doxygen requires a folder of the project where all Source Code of the project can be found. 

When I look in The project Release folder, there are all used libraries but of course they are .o object files. Is there a possibility that Sloeber copies not only the object files but also the source files into the Release folder, or even better, into a Release/Sourcecode folder? The main programs *.ino should be included as well.

In this case it would be sufficient to set doxygen to this source code folder. *.o *.p files are obviously ignorde by doxygen.

Best regards

Alexander

jantje, 2024-07-12:
I do not really understand the problem.
Currently the sketch is in the root of the project and arduino core and libraries are in subfolders in the root.
So Why does the root of the project not work?
FYI the sloeber I'm working on provides a different folder structure and default the sketch is in a src folder.

DocAlex, 2024-07-13:
Here is an example:

D:\Benutzer\Alexander\Documents\sloeber-workspace\DS18B20_OOP_Test

is the root folder of the project.

This folder contains the Main file of the project: DS18B20_OOP_Test.ino

All the other folders of this root have only the .d or .o files. No sources like .h or .cpp are copied there. The libraries folder of the root folder is empty. 

[image]
[image]
[image]

[image]

What I would like to achieve, is that sloeber copies the .h and .cpp files into the ,now empty, folder 

D:\Benutzer\Alexander\Documents\sloeber-workspace\DS18B20_OOP_Test\libraries

Then I can tell Doxygen the folder D:\Benutzer\Alexander\Documents\sloeber-workspace\DS18B20_OOP_Test 

as the Project source folder and everything is working fine. 

Best regards 

Alexander

jantje, 2024-07-20:
OK I see.
First of ... I do not realy see why you would want to document the publicly provided code.
But if you really want to do so you will probably need to use the eclipse doxygen integration

DocAlex, 2024-07-24:
It is not only the publicly provided code, I write the most part of the code as libraries, that I use then later in my main program. The self written libraries stand just beside the publicly available ones. 

Well - there are two reasons for copying the source files into the release folder:

1. Reason is: using doxygen
2. Reason is: By copying all files used in this project I can access the files even years later in the state they had been when the target was programmed. Even if I have updated, changed or deleted the library files. 

Is the really no possibility to copy all source files that had been used for compilation into the realase folder? Or into another folder? Only the sloeber Project knows which files had been compiled for this project. I would like to have an archived copy of the source files being used in the project.

And I would like to build a doxygen documentation only for the project but with all files I used in the project. This is why I doubt that I can use another plugin. 

I would really appreciate that possibility.

Best regards

Alexander

jantje, 2024-07-24:
>The self written libraries stand just beside the publicly available ones.

Unless you use the private library way in Sloeber you should not have your personal libraries in the libraries folder.
If you use the private library way you should have a project in eclipse on which you can use doxygen

For reason 1 I would advice to use the eclipse doxygen integration
For reason 2 I strongly advice to keep to separate release management from documentation

uzi18, 2024-07-24:
Just use git and tag your releases, this will resolve your problems.

DocAlex, 2024-07-26:
Allright, mybe copying the files is not the best idea. 

I am trying to find a solution how the documentation can be automatically setup based on the sloeber project file. 

Within the .project file are all files that are included from eclipse/sloeber (see the example in the attachment). Is there a possibility to automatically include all files that had been included in the .project file into the doxyfile? 

[image]

DocAlex, 2024-07-26:
I found a solution:

attached you find a pyton script, that extracts all necessary information from the sloeber / eclipse structure and output the result into a text file. 

Then I select the paths I want to document and put a "." in front of the line for the current directory, separate them by spaces and copy the so generated line into the doxygen file as "INPUT". End and restart Sloeber and  run the doxygen plugin. 

Nearly half automatic documentation generation. :-)

I should mention that I hve created this file using chat GPT. 

Maybe it would be a nice idea to0 integrate this somewhere in sloeber? 

Greetings

Alexander
[extract_project_files.zip](https://github.com/user-attachments/files/16394214/extract_project_files.zip)

jantje, 2024-08-13:
Cool you solved the problem.
To me the eclipse doxygen is the way to go. It probably does something very similar.
As your problem is fixed I'm closing this issue.

DocAlex, 2024-08-13:
Maybe it would help to integrate my script into sloeber?

jantje, 2024-08-13:
I would only consider to do so if there is a issue on the eclipse doxygen plugin that the maintainers do not want to fix and that is making this script a workaround.

DocAlex, 2024-08-13:
There might be a misunderstanding: I use the Eclipse Doxygen plugin. The only question is which files/paths to be included. The only purpose for the script is to collect all paths/files that have been used during a compile respectively have been set in the project. What you use this information for is up to the user. You can copy the files, you can keep that as a documentation, or you can use it for doxygen. It is more „tell me which files have been included“…Greetings.

jantje, 2024-08-13:
>There might be a misunderstanding: I use the Eclipse Doxygen plugin.

There was a misunderstanding.
I might take a look at this but it is very low priority for me.
I propose you make a discussion show and tell.