Sloeber/arduino-eclipse-plugin #1659: Serial monitor only works after its "reset" button is pressed

jonbarril, 2024-07-25:
The target arduino is a ProMicro (Leonardo). I haven't used the plugin for many months. Recently I installed the very latest version. With this release it seems that the serial monitor will not display text from the target board unless I press the "reset" button on the serial monitor. Once pressed the serial monitor output works as expected.

Also, pressing the serial monitor "reset" button a second time does not seem to actually cause the target board to reset. It just continues to operate normally.

jantje, 2024-07-25:
I do not have a ProMicro but I can tell you the leonardo is a special beast and always has been troublesome for upload and serial monitor.

jonbarril, 2024-07-25:
Yeah, ProMicro/Leonardo is a bit different, however this is a new behavior that wasn't a problem in the older version of the plugin. In fact, I don't recall a serial monitor "reset" button (or if there was one I never had to use it -- the serial monitor just worked).

BTW: What is the reset button doing? As I mentioned, it doesn't seem to actually reset the target board.

jantje, 2024-07-25:
The reset button has always been there (it used to be the same code as for the upload -now there are multiple upload protocols and not all need a reset-).
The code pulls up a port; keeps it high for some time and then drops it back to low (or visa versa)
There is some hardware needed on the board for this to trigger a reset and Arduino has done changes in this area in the past.