Sloeber/arduino-eclipse-plugin #1665: V5.0 testing - ESP32 package shows up twice in "Platform and Boards" but V3.0.4 is missing in one - Also in Sloeber 4.4.3

ArminJo, 2024-08-19:
I use https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json as index URL for esp32.

I see 2 esp32 entries
[image]

but only in the first esp32 I can see the 3.0.4 version
[image]

The folder structure contains 3.0.4 :-)
[image]

In the file package_esp32_index.json, the 3.0.4 is included, in the file package_index.json the esp32 "section" starts with 3.0.3, 3.0.4 is missing.
Restarting Sloeber does not help.

jantje, 2024-08-19:
There will be 2 json files that provide esp32 hardware on you system.
Arduino IDE hides the json files so Sloeber hides them by default.
I think -when you use multiple json files- this page is way more readable if you do not hide the json files
[image]

ArminJo, 2024-08-19:
Thanks for the hint, but after deleting package_index.json, it is rebuild again on start of Sloeber, but again without the 3.0.4. 
And if I deselect the 2.0.14 from the package_index.json it is also deselected from package_esp32_index.json.
Looks like an inconsistency for me.

jantje, 2024-08-19:
>And if I deselect the 2.0.14 from the package_index.json it is also deselected from package_esp32_index.json.

I can not reproduce this
[image]

ArminJo, 2024-08-19:
Thanks for the elaborated explanation!
It seems, that Arduino included the ESP32 core up to 3.0.0 in their private package_index.json, since they claim to have a collaboration with ESP. 
All other 3. party cores like e.g. ESP8266 are still missing in the package_index.json.
**So the inconsistency is at the Arduino side and you just copy it.**

closed 2024-08-19 by ArminJo

jantje, 2024-08-19:
>So the inconsistency is at the Arduino side and you just copy it.

Not really; it is a combination of things
There is only 1 esp board (it doesn't work out of the box in Sloeber) in the arduino package as a result of the collaboration.
The problem is not really a problem but a combination of "could be betters" which will not be fixed
1) The Arduino IDE hides/hid(I don't know) the json file to simplify things. As the Arduino framework became more complex over time this simplification is no longer a valid simplification.
2) If you do not show the json files (The simplification which is still the default in Sloeber) the provider is hidden and 2 providers (Espressif and Arduino in this case) providing the same architecture (esp32 in this case) will both be under the architecture node (esp32). Most people won't see this but esp32 users will.
I think this is caused by how the arduino framework defines and uses "architecture" (I think they mean PU architecture). This is something that has bothered me from the beginning because it is not needed. Paul Stoffregen from Teensy proves this in his Teensy package where all boards are tagged avr but most boards in the package are not avr based.
3) The fact Espressif use esp32 instead of the provider name (Espressif) in the json file adds to the confusion.

To me in this case we are simply confronted with the complexity of life and a learning curve is all that can be done.
Except for changing the default of hiding the json files.