Sloeber/arduino-eclipse-plugin #1666: V5.0 testing - Fresh install on Windows gives Error: Program "make" not found in PATH

ArminJo, 2024-08-19:
A fresh Windows install of V5 results in:

```
10:43:03 **** Clean-only build of configuration Release for project JK-BMSToPylontechCAN_MightyCore ****
make clean 
Cannot run program "make": Launching failed

Error: Program "make" not found in PATH
```
typing make on the Windows CMD prompt also results in:
```
Der Befehl "make" ist entweder falsch geschrieben oder
konnte nicht gefunden werden.
```
so this is consistent 😀 .

On my 4.4.3 Windows installations it runs seamlessly with:

```
10:56:00 **** Clean-only build of configuration Release for project JK-BMSToPylontechCAN_MightyCore ****
"E:\\Elektro\\Sloeber\\arduinoPlugin\\tools\\make\\make" clean 
```
Therefore I can not test the 5.0 😢 

Duplicate of: https://github.com/Sloeber/arduino-eclipse-plugin/issues/1651#issuecomment-2296038497

jantje, 2024-08-19:
I think it is strange that in your test the clean in a freshly installed V5 uses make (= external make builder)
The default builder in V5 is the internal builder and there the clean works
Are you testing with a freshly installed V5 with a V5 created project?

jantje, 2024-08-19:
I just tried here both the internal builder and the external make builder use the same clean method which is internal (read nothing to be seen on the console) and simply removes all the content from the bin/[configname] folder
Not sure why the clean on your freshly installed V5 tries to use make to clean.

jantje, 2024-08-19:
When investigating I found that menu->project->clean does weird things.
I usually go right click project clean which works fine.
But I still haven't found a way to trigger make clean (which is how V5 is supposed to work)

ArminJo, 2024-08-19:
OK newly created projects work for me, but old ones are not migrated.

Printing size "raw Result" results in: 
```
Printing size:: test51.size
"E:\Sloeber5\arduinoPlugin\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-size" -A "E:\WORKSPACE_SLOEBER5\test51\bin\Release/test51.elf"
E:\WORKSPACE_SLOEBER5\test51\bin\Release/test51.elf  :
section                      size      addr
```
"AVR Alternative" also works, but "Arduino Way" (under Windows) results in:
```
Printing size:: test51.size
wsl -- "E:\Sloeber5\arduinoPlugin\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-size" -A "E:\WORKSPACE_SLOEBER5\test51\bin\Release/test51.elf" | "E:\Sloeber5\arduinoPlugin\tools\awk\awk" -f size.awk
/bin/bash: E:Sloeber5arduinoPluginpackagesarduinotoolsavr-gcc7.3.0-atmel3.6.1-arduino7/bin/avr-size: No such file or directory
/bin/bash: E:Sloeber5arduinoPlugintoolsawkawk: command not found
```

jantje, 2024-08-19:
There is no auto migration.
Opening project properties->sloeber apply close may work

ArminJo, 2024-08-19:
I now **always** get this error if I want to open ESP32 project properties->sloeber. On new as well as on old ESP32 projects.
[image]

After Clicking OK I see:
[image]

It gives no error for **new** AVR projects but for **old** AVR ones I see the same error and after clicking OK I see:
[image]
or 
[image]

So I give up testing Sloeber5 for now, since only new AVR Projects seems to work for me 😞 .

jantje, 2024-08-19:
Migration of projects is not going to work. If there is no easy way to get it to work I will have to write some conversion code.
For now please test with projects created in v5.

>I now always get this error if I want to open ESP32 project properties->sloeber. On new as well as on old ESP32 projects.

Can you zip up the V5 created ESP32 project so I can take a look?

ArminJo, 2024-08-19:
Here is the Zip:
[testESP32_C3.zip](https://github.com/user-attachments/files/16666664/testESP32_C3.zip)

jantje, 2024-08-24:
Seems there is a null case on my system after importing the project.
Once that one is fixed all worked "as expected"
"as expected" in this case means that the build reported the error 
` #error This library does not work with ESP32 core 3.x. Please use ESP 2.0.17 core. You are kindly invited to port and document the code to 3.x, to fix this problem!`
I also noted other build errors in esp32 related to missing "
I'm looking into it

jantje, 2024-08-24:
Seems the quoting issue does not happen in an empty sketch with the board in the provided zip.
It also didn't happen in my testing.
Seems I will have to wait until someone reports something.

ArminJo, 2024-08-25:
> I now **always** get this error if I want to open ESP32 project properties->sloeber. On new as well as on old ESP32 projects.

This seems to be fixed 😀 Thanks!

The error: `Error: Program "make" not found in PATH` still happens, when opening an old 4.3.3 configuration. 
Is there any plan to provide a configuration migration scrip or functionality?

jantje, 2024-08-25:
>The error: Error: Program "make" not found in PATH still happens, when opening an old 4.3.3 configuration.
Is there any plan to provide a configuration migration scrip or functionality?

I want V5 to work properly before looking at migration.
That is because migration is hard and migrating to a moving target is even harder.

ArminJo, 2024-10-01:
Hi Jaintje,
I took a new try with sloeber	5.0.0.202409240132.

I imported a single config project and tried to compile, but still get
```
Error: Program "make" not found in PATH
PATH=[E:/Sloeber5//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_17.0.12.v20240802-1518/jre/bin/server;E:/Sloeber5//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_17.0.12.v20240802-1518/jre/bin;E:\Elektro\Python311\Scripts\;E:\Elektro\Python311\;C:\Program Files\Common Files\Oracle\Java\javapath;E:\Elektro\PlatformIO\Python38\Scripts\;E:\Elektro\PlatformIO\Python38\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\ATI Technologies\ATI.ACE\Core-Static;C:\Program Files (x86)\Windows Live\Shared;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin\;C:\Program Files\IVI Foundation\VISA\Win64\Bin\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;C:\WINDOWS\System32\OpenSSH\;E:\Computer\Ultraedit;E:\Elektro\nodejs\;C:\Program Files (x86)\IVI Foundation\IVI\bin;C:\Program Files\IVI Foundation\IVI\bin;E:\Media\Pandoc\;C:\Users\Armin\AppData\Local\Microsoft\WindowsApps;E:\Elektro\PlatformIO\Microsoft VS Code\bin;C:\Users\Armin\AppData\Roaming\npm;E:\Elektro\PlatformIO\penv\Scripts;E:\Media\Graphviz\bin;E:\Sloeber5]
14:48:27 Build Failed. 1 errors, 0 warnings. (took 189ms)

**** Build of configuration Release for project Talkie_ESP32 ****

Nothing to build for project Talkie_ESP32
```
After deleting the build directory (RELEASE), I get 
```
**** Build of configuration Release for project Talkie_ESP32 ****
Nothing to build for project Talkie_ESP32
```
and still see 
[image]

**The file `.AutoBuildProject` is also still missing**

[image]

and languages.setting.xml starts still with
```
<project>
	<configuration id="io.sloeber.core.sketch.0" name="Release">
```
in contrast to a new project, where it starts with
```
<project>
	<configuration id="io.sloeber.autoBuild.configurationDescription.1887448826" name="Release">
```

This is independent, if source is located in /src directory or in root directory.

The zipped project is here: 
[Talkie_ESP32.zip](https://github.com/user-attachments/files/17209802/Talkie_ESP32.zip)

jantje, 2024-10-02:
Have you converted it to a sloeber project?
After importing a project you need to convert it to a sloeber project file->new->other->sloeber->convert to sloeber project

ArminJo, 2024-10-02:
Thanks!
After converting I observed, that I must add all libraries manually again and then delete the old /libraries directory

jantje, 2024-10-02:
>After converting I observed, that I must add all libraries manually again and then delete the old /libraries directory.

I think that is current implementation. I'll see wether I can improve

>And after adding a library all library examples are not excluded and compiled,

Strange they should