Sloeber/arduino-eclipse-plugin #1672: V5.0 testing - ESP32-C3 includes are missing in the Eclipse Project Explorer view

ArminJo, 2024-08-19:
In my old 4.3.3 project i see:
[image]

After restarting Sloeber5 with the new C3 project and opening the Sloeber project properties to add a port, I see:
[image]

But nevertheless the Includes still have only 11 entries.
[image]
and I get the error
```
Starting C++ compile: Esp.cpp.o
"E:\Sloeber5\arduinoPlugin\packages\esp32\tools\esp-rv32\2302/bin/riscv32-esp-elf-g++" -MMD -c "@E:\Sloeber5\arduinoPlugin\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-b6b4727c58\esp32c3/flags/cpp_flags" -Wall -Wextra -Os -DF_CPU=10000000L -DARDUINO=20302 -DARDUINO_ESP32C3_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD=\"ESP32C3_DEV\" -DARDUINO_VARIANT="esp32c3" -DARDUINO_PARTITION_app3M_fat9M_16MB -DARDUINO_HOST_OS="windows" -DARDUINO_FQBN="esp32:esp32:esp32c3:BOD=4v3,bootloader=uart0,CDCOnBoot=cdc,clock=16MHz_external,CPUFreq=10,DebugLevel=debug,eeprom=keep,EraseFlash=all,FlashFreq=40,FlashMode=dio,FlashSize=16M,JTAGAdapter=bridge,LTO=Os_flto,PartitionScheme=app3M_fat9M_16MB,pinout=standard,UploadSpeed=115200,variant=modelP,ZigbeeMode=default" -DESP32 -DCORE_DEBUG_LEVEL=4    -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=1  "@E:\Sloeber5\arduinoPlugin\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-b6b4727c58\esp32c3/flags/defines" "-IE:\WORKSPACE_SLOEBER5\testESP32_C3\src" -iprefix "E:\Sloeber5\arduinoPlugin\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-b6b4727c58\esp32c3/include/" "@E:\Sloeber5\arduinoPlugin\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-b6b4727c58\esp32c3/flags/includes" "-IE:\Sloeber5\arduinoPlugin\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-b6b4727c58\esp32c3/dio_qspi/include" "-IE:/WORKSPACE_SLOEBER/libraries/IRremote/src" "-IE:/Sloeber5/arduinoPlugin/libraries/FreeRTOS/11.1.0-2/src" "-IE:/Sloeber5/arduinoPlugin/packages/esp32/hardware/esp32/3.0.4/variants/esp32c3" "-IE:/Sloeber5/arduinoPlugin/packages/esp32/hardware/esp32/3.0.4/cores/esp32" -MMD -MP "-MFarduino/Release/core/Esp.cpp.d" "-MTarduino/Release/core/Esp.cpp.o" -D__IN_ECLIPSE__=1 "@E:\WORKSPACE_SLOEBER5\testESP32_C3\bin\Release/file_opts" "E:/Sloeber5/arduinoPlugin/packages/esp32/hardware/esp32/3.0.4/cores/esp32/Esp.cpp" -o "arduino/Release/core/Esp.cpp.o"     
In file included from E:\Sloeber5\arduinoPlugin\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-b6b4727c58\esp32c3/include/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:65,
                 from E:/Sloeber5/arduinoPlugin/packages/esp32/hardware/esp32/3.0.4/cores/esp32/Arduino.h:33,
                 from E:/Sloeber5/arduinoPlugin/packages/esp32/hardware/esp32/3.0.4/cores/esp32/Esp.cpp:20:
E:/Sloeber5/arduinoPlugin/libraries/FreeRTOS/11.1.0-2/src/FreeRTOSConfig.h:32:10: fatal error: avr/io.h: No such file or directory
```

jantje, 2024-08-19:
```
Starting C++ compile: Esp.cpp.o
"E:\Sloeber5\arduinoPlugin\packages\esp32\tools\esp-rv32\2302/bin/riscv32-esp-elf-g++" -MMD -c "@E:\Sloeber5\arduinoPlugin\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-b6b4727c58\esp32c3/flags/cpp_flags" -Wall -Wextra -Os -DF_CPU=10000000L -DARDUINO=20302 -DARDUINO_ESP32C3_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD=\"ESP32C3_DEV\" -DARDUINO_VARIANT="esp32c3" -DARDUINO_PARTITION_app3M_fat9M_16MB -DARDUINO_HOST_OS="windows" -DARDUINO_FQBN="esp32:esp32:esp32c3:BOD=4v3,bootloader=uart0,CDCOnBoot=cdc,clock=16MHz_external,CPUFreq=10,DebugLevel=debug,eeprom=keep,EraseFlash=all,FlashFreq=40,FlashMode=dio,FlashSize=16M,JTAGAdapter=bridge,LTO=Os_flto,PartitionScheme=app3M_fat9M_16MB,pinout=standard,UploadSpeed=115200,variant=modelP,ZigbeeMode=default" -DESP32 -DCORE_DEBUG_LEVEL=4    -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=1  "@E:\Sloeber5\arduinoPlugin\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-b6b4727c58\esp32c3/flags/defines" "-IE:\WORKSPACE_SLOEBER5\testESP32_C3\src" -iprefix "E:\Sloeber5\arduinoPlugin\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-b6b4727c58\esp32c3/include/" "@E:\Sloeber5\arduinoPlugin\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-b6b4727c58\esp32c3/flags/includes" "-IE:\Sloeber5\arduinoPlugin\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-b6b4727c58\esp32c3/dio_qspi/include" "-IE:/WORKSPACE_SLOEBER/libraries/IRremote/src" "-IE:/Sloeber5/arduinoPlugin/libraries/FreeRTOS/11.1.0-2/src" "-IE:/Sloeber5/arduinoPlugin/packages/esp32/hardware/esp32/3.0.4/variants/esp32c3" "-IE:/Sloeber5/arduinoPlugin/packages/esp32/hardware/esp32/3.0.4/cores/esp32" -MMD -MP "-MFarduino/Release/core/Esp.cpp.d" "-MTarduino/Release/core/Esp.cpp.o" -D__IN_ECLIPSE__=1 "@E:\WORKSPACE_SLOEBER5\testESP32_C3\bin\Release/file_opts" "E:/Sloeber5/arduinoPlugin/packages/esp32/hardware/esp32/3.0.4/cores/esp32/Esp.cpp" -o "arduino/Release/core/Esp.cpp.o"     
In file included from E:\Sloeber5\arduinoPlugin\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-b6b4727c58\esp32c3/include/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:65,
                 from E:/Sloeber5/arduinoPlugin/packages/esp32/hardware/esp32/3.0.4/cores/esp32/Arduino.h:33,
                 from E:/Sloeber5/arduinoPlugin/packages/esp32/hardware/esp32/3.0.4/cores/esp32/Esp.cpp:20:
E:/Sloeber5/arduinoPlugin/libraries/FreeRTOS/11.1.0-2/src/FreeRTOSConfig.h:32:10: fatal error: avr/io.h: No such file or directory
```

Is caused by the FreeRtos library being included in the project.

jantje, 2024-08-19:
I have to take a look for the missing libraries.

ArminJo, 2024-08-19:
> Is caused by the FreeRtos library being included in the project.

I knew this kind of errors from 4.3.3, therefore I disabled FreeRtos for this configuration, but this did not work here.
[image]

jantje, 2024-08-19:
>I knew this kind of errors from 4.3.3, therefore I disabled FreeRtos for this configuration, but this did not work here.

You found a regression in V5
` "-IE:/Sloeber5/arduinoPlugin/libraries/FreeRTOS/11.1.0-2/src"` should not be part of the compile command as the freeRtos lib is excluded.

jantje, 2024-08-24:
This should be fixed now in the action being build now.

ArminJo, 2024-08-25:
Now it works for me.
Thank you very much!

closed 2024-08-25 by ArminJo