Sloeber/arduino-eclipse-plugin #1673: V5.0 testing: Intro error

PerennialNovice, 2024-08-20:
with version 5.0.0.202408181423, right after first startup, I get the following error message:

[image]

PerennialNovice, 2024-08-20:
and sporadically I get an Index 0 out of bounds messag:

[image]

jantje, 2024-08-20:
Can you provide the logs?

PerennialNovice, 2024-08-20:
sure, hope it is the right thing ...

There are some lines about messed up project configurations in there too. As far as I can tell, this is unrelated to V5.0

[errorlog.log](https://github.com/user-attachments/files/16675320/errorlog.log)

jantje, 2024-08-20:
The error log tells me you tried to open a pre V5.0 workspace.
V5.0 is really 100% CDT project incompatible with previous versions and I have not yet considered the migration problem.
Only the error below might be Sloeber V5.0 related.
`java.io.FileNotFoundException: C:\Sloeber_Nightly\introContent.xml (The system cannot find the file specified)`

Please do not use existing V4.X.Y sloeber workspaces for testing right now. We are not there yet.

PerennialNovice, 2024-08-20:
in fact, I created a new workspace but imported an existing project from git.
With config files like .project, .sproject etc ...
I guess that is a problem too, then?

PerennialNovice, 2024-08-20:
allright, that's the path to follow then!

Thanks for enlightening me :)

jantje, 2024-08-24:
Closing this issue as user error.
Please reopen if you disagree

closed 2024-08-24 by jantje