Sloeber/arduino-eclipse-plugin #1679: not respecting platform choice with two versions of STM32 installed

PerennialNovice, 2024-09-03:
Since I ran into problems after upgrading from STM32core 2.7.1 to STM32core 2.8.1, I installed 2.7.1 again side by side with 2.8.1

[image]

When I switch to 2.7.1 in sloeber project properties, I get some "xyz undeclared" errors

[image]

and scrolling through the console output I see that several files are taken from 2.8.1 STM32core

[image]

I suspect this is the cause for the "undeclared" errors?

removing the 2.8.1 folder from the disk did not solve it.

I then noticed that several libraries where still pointing to the 2.8.1 version, but reattaching them via the "reattach all libraries" command did not work out. It attached libraries vor AVR arduino boards. I have configurations for both AVR and STM32 for this project.

I then manually removed and reattached the libraries via the "add a library" button, which did not help either. I checked all libraries and their path in the ressource properties dialog and they all point to 2.7.1
Still I got errors regarding Wire.h that would not go away.

Only way to get it to compile again, was to copy over .project, .cproject, .sproject and sloeber.cfg from a local backup I luckily did just before ...

---

**Describe your environment**
sloeber product 4.4.3 on Win11

PerennialNovice, 2024-09-03:
> Only way to get it to compile again, was to copy over .project, .cproject, .sproject and sloeber.cfg from a local backup I luckily did just before ...

But now I was on 2.8.1 again...
I switched to 2.7.1 and this time I immediately changed all the libraries ressource properties to point to 2.7.1 and it compiled at the first try :)
So the main issue seems to be that when switching platform verions, the libraries do not get switched as well?

PerennialNovice, 2024-09-04:
>The same problem exists in V4 when switching configuration (where the 2 configurations require different libraries).

Our projects use different libraries for AVR or STM32 target, but we never had an issue when switching. Maybe because we only had different "exclude from build" settings instead of relying on removing/adding libraries?

jantje, 2024-09-05:
The exclude from build is great but can not be used in the case of different versions of a board when using "hardware libraries" as the folder with the name of the lib can only point to one versions.

jantje, 2024-12-20:
Should be fixed in V5

closed 2024-12-20 by jantje