Sloeber/arduino-eclipse-plugin #1682: MacOS  installer incomplete or broken

asat232, 2024-09-15:
macOS Monterey Version 12.6.1 (MacBook Pro Mid 2015)
Download link:
https://github.com/Sloeber/arduino-eclipse-plugin/releases/download/V4_4_3_0/sloeber-ide-V4.4.3-macosx.cocoa.x86_64.zip
After downloading .zip and unzipping, when moving Sloeber.app to application folder having error message "Sloeber.app incomplete or damaged".
1. Tried to download both Safari and Chrome.
2. Tried before unzipping move zip from Download folder to another folder in my user root directory.
3. Used "sudo spctl --master-disable" to have Allow from "Anywhere" in security settings
4. Tried to install Version 4.4.2 on the same Mac and its installing without issues
https://github.com/Sloeber/arduino-eclipse-plugin/releases/download/V4_4_2_0/sloeber-ide-V4.4.2-macosx.cocoa.x86_64.zip
5. Inspecting content of Sloeber.app (V4.4.3) i see that Eclipse.app is damaged or broken.

So i understand the fast workaround can be to replace Eclipse.app in Sloeber.app
If its possible, provide me please with info from where Eclipse.app to take and version of it.

jantje, 2024-10-13:
To avoid installation problems you can always compile locally as explained here 
https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/readme.md

rfbarraza, 2024-11-05:
Howdy, seconding this to see if there are any updated for builds? 

4.3.2
4.3.3
4.4.3 

All seem to have this issue.

jantje, 2024-11-05:
As I do not own a mac I do not know nor understand what the problem is.

rfbarraza, 2024-11-05:
I do and would be more than happy to help. At least for other macOS folks who find themselves here.  :)

From a high-level perspective, it appears that the build process for Macs (at least aarch64 Macs) is not producing a valid build product.

When I attempt to run 4.4.3 I get the following error:
```
The application cannot be opened for an unexpected reason, error=Error Domain=NSOSStatusErrorDomain Code=-10827 "kLSNoExecutableErr: The executable is missing" UserInfo={_LSLine=4129, _LSFunction=_LSOpenStuffCallLocal}
```

Similarly, if I clone the repo and attempt to build locally by running: 
```
mvn clean verify -PSDK,mac64,latest -DskipTests=true && ./build_then_launch.sh
```
from HEAD of master, I see what seems to be a completely successful build result. But  get the following output from build_then_launch.sh:
```
Searching for the Eclipse executable (with our plugin pre-packaged) to launch
Did not find an Eclipse Arduino IDE executable built.....
```

Opening build_then_launch.sh, this is because line 3:
```
file_x86_64="./io.sloeber.product/target/products/io.sloeber.product/linux/gtk/x86_64/Sloeber/sloeber-ide"
```

is hardcoded for Linux builds. I was able to find the macOS build products in:
./io.sloeber.product/target/products/io.sloeber.product/macosx/cocoa/aarch64

So, while the official build product doesn't seem to be working, you can build it yourself by following the [aforementioned instructions](https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/readme.md).

jantje, 2024-11-05:
>I do and would be more than happy to help. At least for other macOS folks who find themselves here. :)

That sounds great

>mvn clean verify -PSDK,mac64,latest -DskipTests=true && ./build_then_launch.sh

That is not the right command.
the -PSDK says you want to build the SDK (that is the eclipse java development environment to help devellop Sloeber)
I do not use  ./build_then_launch.sh so I can not comment on it working or  not.

>from HEAD of master, I see what seems to be a completely successful build result. But get the following output from build_then_launch.sh:

As stated above that is because you didn't build io.sloeber.product but build io.sloeber.product.sdk (I know it is confusing)

rfbarraza, 2024-11-05:
> That is not the right command.
> the -PSDK says you want to build the SDK (that is the eclipse java development environment to help devellop Sloeber)
> I do not use ./build_then_launch.sh so I can not comment on it working or not.

Cool, (so assuming the following is correct) running:
```
mvn clean verify -Pmac64,latest,NOSDK -DskipTests=true
```

Yields a launch-able build. I can't seem to coerce the v4_4_3_0 tag to build, so I'm comparing HEAD on master to the build products downloaded from the sloeber website. But some quick differences I noticed are:

v4.4.3 is missing:
- Contents/Info.plist
- Contents/Resources/icons.icns

and has this vestigial directory:
- Contents/Eclipse.app

TrippyLighting, 2024-11-24:
Copying (and perhaps editing) the info.plist file from a previous version and copying the resource folder with the icon works in the sense that I can compile code for the ESP32-S3 with Sloeber 4.4.3. 

In general, the Mac versions of Sloeber seem quite "flaky" compared to the Windows versions.

1. I downloaded and installed 4.4.2  on my mac first (Intel MacBook Pro from 2017).
3. I unzipped it after the usual sudo xattr -r -d com.apple.quarantine Sloeber.app I moved Sloeber.app into the Applications folder.
4. I can compile and upload a simple Blink to the ESP32-S3
5. I can compile and upload my application, and the S3 communicates with TouchOSC on my iPhone  
6. However, either the indexing o discovery don't seem to work and given the issues I've read through here on Github, that seems not unusual.

On my Windows installation of 4.4.1, this all works flawlessly without having to use the command line or terminal to get anything out of quarantine.

Then I installed 4.4.1 on my mac using the same steps. Did not work. App could not be started. I had to go and find the sloeber-ide file and use chmod 755 to make it executable.
Then I was able to compile and upload a Blink and my main application to the ESP32-S3.

However, what all of those installations on the Mac share is that the indexing or discovery doesn't work.
I also have to create the file_opts file in order to successfully compile.

None of this is needed for my Windows Sloeber 4.4.1 installation. I can compile and upload, and libraries/includes are indexed/discovered.

jantje, 2024-11-24:
Many of the ESP32 problems are related to the specific version of the ESP32 platform (the indexer is version independent).
Are the ESP32 version the same on your Windows Sloeber 4.4.1 installation and the Windows Sloeber 4.4.2 installation?

FYI I just made a lot of changes to Sloeber V5 to make ESP32 version 3.0.7 work out of the box; I doubt Sloeber V4.X works with ESP V3.0.7.

TrippyLighting, 2024-11-24:
TL:DR -> versions 4.4.2 and 4.4.3 now work, and "stuff" has been discovered. I can build and upload my application with both.
I've used ESP32 platforms from 2.0.11, 2.0.14, and 2.0.17 to see if that would make a difference. My application compiles and works with either platform.
ESP43 platform 3.x.x does not work.

Longer version:
I ditched 4.4.1 on the Mac and focused on 4.4.2 and 4.4.3. So now I have a Sloeber442.app and a Sloeber443.app in the Applications folder.
I also have two separate workspaces, sloeber_workspace442 and sloeber_workspacce443. The latter is a copy of the first.
To ensure that I had the same settings and used the same libraries and ESP32 platforms, I had to start and close both versions a number of times. I also cleaned the project(s) and re-ran the indexer.
I noticed that after restarting the app (I believe both versions), the progress/status bar showed "finishing installation" at some point. Afterwards, things started working!

I believe some of this behavior has been described by other users, and what it comes down to is that Sloeber has to download stuff in the background but isn't particularly verbose about it. When it's finally done configuring itself, then things start to work seemingly magically.

Thank you for your relentless work on Sloeber!!!

jantje, 2024-11-25:
>I also have two separate workspaces, sloeber_workspace442 and sloeber_workspacce443. The latter is a copy of the first.

If you want to have identical copies you also need to copy the arduinplugin (which is where the "finishing installation" magic happens) from the sloeber app.

>I noticed that after restarting the app (I believe both versions), the progress/status bar showed "finishing installation" at some point. Afterwards, things started working!

The initial install is annoying and has to run fully for things to work properly (I fully understand why arduino IDE provides the avr platform with the IDE) but Sloeber V4.X and earlier also "finishing installation" at each Sloeber start. I finally got to fix that in V5 (as it makes testing annoying )

Note that you can create the SLOEBER_HOME environment variable ans Sloeber will place the arduinoplugin folder there.
This can be handy but running 2 Sloeber versions at the same time may cause problems if they have different workaround ID's (I'm talking about the sloeber.txt files which may be recreated all the time)

>Thank you for your relentless work on Sloeber!!!

Thanks

jantje, 2025-01-12:
Is this a real issue I can do something about?
Is there a issue in V5?