Sloeber/arduino-eclipse-plugin #1684: M5CoreS3 Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0

huwi, 2024-10-13:
**Describe your environment**   
Do you use the Sloeber product or plugin?   Sloeber Product
Sloeber version number  4.4.3
OS: [e.g. windows 10]   Linux Mint (Ubuntu)

**Describe the bug**   
I have spent some days to find the problem. with the arduino ide my sketch works find. when ich compile the same sketch with sloeber ide, the board will Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0.

I have tested all Board-Settings and Libraries.

**To Reproduce**   
Steps to reproduce the behavior:   
1. Create a new project
2. Use M5CoreS5 Board with M5GFX, M5Universe
3. Compiole a small Sketch
4. Download the Sketch 
5. Open the Console:

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376a35
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x109c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xb50
load:0x403cc700,len:0x2fe4
entry 0x403c98ac
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4200ce45  PS      : 0x00060f30  A0      : 0x820179ed  A1      : 0x3fceb2c0  
A2      : 0x00000001  A3      : 0x000000a8  A4      : 0x00000001  A5      : 0x00000004  
A6      : 0x3fc98044  A7      : 0x3fceb2a0  A8      : 0x00000000  A9      : 0x3fceb2a0  
A10     : 0x3fc95dd4  A11     : 0x000000ff  A12     : 0x000000ff  A13     : 0x00000023  
A14     : 0x3fc95e08  A15     : 0x3fcf1118  SAR     : 0x00000016  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0xffffffff  

Backtrace: 0x4200ce42:0x3fceb2c0 0x420179ea:0x3fceb2e0 0x40376727:0x3fceb310 0x403cdb0e:0x3fceb340 0x403cdea5:0x3fceb380 0x403c9919:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710

ELF file SHA256: a654e39091f38bcd

Sketch i used:

#include "Arduino.h"
#include "M5CoreS3.h"

void setup()
{
	auto cfg = M5.config();
	CoreS3.begin(cfg);

	Serial.begin(921600);
	while (!Serial) {delay(10);}

	Serial.println("Starting .....");
}

void loop()
{
	yield();
	delay(500);
	Serial.println("Running .....");
}

Board: M5CoreS3

Kind Regards, Hubert

jantje, 2024-10-13:
The only way I can think of to investigate this issue is by comparing the compile commands generated by Sloeber versus the Arduino IDE.

huwi, 2024-10-27:
Now i have done some testings.
If i only include the M5Unified.h file, nothing happens.
Then, when i include the M5CoreS3.h file, the problem arises.
I only in include the file, clean the project an upload the result.
I have increased the flash size ota_0 ota_1 to 3mb and the corecump areea.

jantje, 2024-10-27:
This may be related to #1625.