Sloeber/arduino-eclipse-plugin #1691: How to run menuconfig to update Kconfig for ESP32 ?

GitHubLionel, 2024-11-21:
Hi, 
I need to add some directives to my project (`CONFIG_ADC_CONTINUOUS_ISR_IRAM_SAFE` and `CONFIG_GDMA_ISR_IRAM_SAFE`).
It doesn't work if I add this directives with project/properties/Sloeber/"Compile options" with "append to C/C++" or Link.
I ask in [Espressif forum](https://github.com/espressif/esp-idf/issues/14893) and the answer is to use `menuconfig`.
How can do that with Sloeber ?
Best regards,
Lionel

jantje, 2024-11-21:
>project/properties/Sloeber/"Compile options" with "append to C/C++"

This should work. Can you provide what you have put there and the command from the console you and the command you want?

> the answer is to use menuconfig.

If they are not talking about the menu options is project/properties/Sloeber I have no clue what they are talking about.

GitHubLionel, 2024-11-22:
> > project/properties/Sloeber/"Compile options" with "append to C/C++"
> 
> This should work. Can you provide what you have put there and the command from the console you and the command you want?
> 
I add `-DCONFIG_ADC_CONTINUOUS_ISR_IRAM_SAFE -DCONFIG_GDMA_ISR_IRAM_SAFE` 
and I can see this in the log:
`"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\tools\esp-x32\2302/bin/xtensa-esp32-elf-gcc" -MMD -c "@L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-dc859c1e67\esp32/flags/c_flags" -Wall -Wextra -Os -DF_CPU=240000000L -DARDUINO=10812 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" -DARDUINO_VARIANT="\"esp32\"" -DARDUINO_PARTITION_ -DARDUINO_HOST_OS="\"FreeRTOS\"" -DARDUINO_FQBN="\"ESP32:esp32:esp32\"" -DESP32 -DCORE_DEBUG_LEVEL=5 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw -DARDUINO_USB_CDC_ON_BOOT=0  "@L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-dc859c1e67\esp32/flags/defines" "-IL:\Sloeber\workspace\ESP32_ADC" -iprefix "L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-dc859c1e67\esp32/include/" "@L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-dc859c1e67\esp32/flags/includes" "-IL:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-dc859c1e67\esp32/qio_qspi/include" -I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\variants\esp32" -I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\cores\esp32" -I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\libraries\EEPROM" -I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\libraries\EEPROM\src" -I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\libraries\Update\src" -I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\libraries\FS\src" -I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\libraries\LittleFS\src" -I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\libraries\WiFi\src" -I"L:\Sloeber\Library\Partition_utils" -I"L:\Sloeber\Library\Debug_utils" -I"L:\Sloeber\Library\RTCLocal" -I"L:\Sloeber\Library\Server_utils" -I"L:\Sloeber\Library\Fonts" -I"L:\Sloeber\Library\SSD1306" -I"L:\Sloeber\Library\SSD1327" -I"L:\Sloeber\Library\SH1107" -I"L:\Sloeber\Sloeber4\arduinoPlugin\libraries\NTPClient\3.2.1" -I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\libraries\Preferences\src" -I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\libraries\WebServer\src" -I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\libraries\HTTPUpdateServer\src" -I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\libraries\Network\src" -I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\libraries\SPIFFS\src" -I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\libraries\Wire\src" -MMD -MP -MF"core\core\esp32-hal-adc.c.d" -MT"core\core\esp32-hal-adc.c.o" -D__IN_ECLIPSE__=1 "@L:\Sloeber\workspace\ESP32_ADC\Release/file_opts" "L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\cores\esp32\esp32-hal-adc.c" -o "core\core\esp32-hal-adc.c.o" -DUSE_CONFIG_LIB_FILE -DCONFIG_ADC_CONTINUOUS_ISR_IRAM_SAFE -DCONFIG_GDMA_ISR_IRAM_SAFE`
So it is correct.

> > the answer is to use menuconfig.
> 
> If they are not talking about the menu options is project/properties/Sloeber I have no clue what they are talking about.

They are talking about the [Kconfig file](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/kconfig.html). 

The problem is that my program continue to crash with this message: 
`Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed). `
This message indicate that ADC functions are not in IRAM.

GitHubLionel, 2024-11-22:
EDIT : If I decompose the line in the log, you can see some mistakes on line 27 and 31: the quote " is in bad position.
```
"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\tools\esp-x32\2302/bin/xtensa-esp32-elf-gcc"
-MMD
-c
"@L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-dc859c1e67\esp32/flags/c_flags"
-Wall
-Wextra
-Os
-DF_CPU=240000000L
-DARDUINO=10812
-DARDUINO_ESP32_DEV
-DARDUINO_ARCH_ESP32
"-DARDUINO_BOARD=\"ESP32_DEV\""
-DARDUINO_VARIANT="\"esp32\""
-DARDUINO_PARTITION_
-DARDUINO_HOST_OS="\"FreeRTOS\""
-DARDUINO_FQBN="\"ESP32:esp32:esp32\""
-DESP32
-DCORE_DEBUG_LEVEL=5
-DARDUINO_RUNNING_CORE=1
-DARDUINO_EVENT_RUNNING_CORE=1
-DBOARD_HAS_PSRAM
-mfix-esp32-psram-cache-issue
-mfix-esp32-psram-cache-strategy=memw
-DARDUINO_USB_CDC_ON_BOOT=0

"@L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-dc859c1e67\esp32/flags/defines"
"-IL:\Sloeber\workspace\ESP32_ADC"
-iprefix
"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-dc859c1e67\esp32/include/"
"@L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-dc859c1e67\esp32/flags/includes"
"-IL:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-dc859c1e67\esp32/qio_qspi/include"
-I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\variants\esp32"
-I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\cores\esp32"
-I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\libraries\EEPROM"
-I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\libraries\EEPROM\src"
-I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\libraries\Update\src"
-I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\libraries\FS\src"
-I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\libraries\LittleFS\src"
-I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\libraries\WiFi\src"
-I"L:\Sloeber\Library\Partition_utils"
-I"L:\Sloeber\Library\Debug_utils"
-I"L:\Sloeber\Library\RTCLocal"
-I"L:\Sloeber\Library\Server_utils"
-I"L:\Sloeber\Library\Fonts"
-I"L:\Sloeber\Library\SSD1306"
-I"L:\Sloeber\Library\SSD1327"
-I"L:\Sloeber\Library\SH1107"
-I"L:\Sloeber\Sloeber4\arduinoPlugin\libraries\NTPClient\3.2.1"
-I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\libraries\Preferences\src"
-I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\libraries\WebServer\src"
-I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\libraries\HTTPUpdateServer\src"
-I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\libraries\Network\src"
-I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\libraries\SPIFFS\src"
-I"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\libraries\Wire\src"
-MMD
-MP
-MF"core\core\esp32-hal-adc.c.d"
-MT"core\core\esp32-hal-adc.c.o"
-D__IN_ECLIPSE__=1
"@L:\Sloeber\workspace\ESP32_ADC\Release/file_opts"
"L:\Sloeber\Sloeber4\arduinoPlugin\packages\esp32\hardware\esp32\3.0.3\cores\esp32\esp32-hal-adc.c"
-o
"core\core\esp32-hal-adc.c.o"
-DUSE_CONFIG_LIB_FILE
-DCONFIG_ADC_CONTINUOUS_ISR_IRAM_SAFE
-DCONFIG_GDMA_ISR_IRAM_SAFE
```
But I don't think this is the origin of the problem.

GitHubLionel, 2024-11-22:
NOTE2:
I also try with this directive: `-DCONFIG_ADC_CONTINUOUS_ISR_IRAM_SAFE=1 -DCONFIG_GDMA_ISR_IRAM_SAFE=1`
in "append to C/C++" and "Link".
And I also try in "C/C++ Build Environment" to modify the build.extra_flags.esp32 variable from `-DARDUINO_USB_CDC_ON_BOOT=0` to `-DARDUINO_USB_CDC_ON_BOOT=0 -DCONFIG_ADC_CONTINUOUS_ISR_IRAM_SAFE=1 -DCONFIG_GDMA_ISR_IRAM_SAFE=1`

GitHubLionel, 2024-11-22:
I try to directly add this two lines:
```
#define CONFIG_ADC_CONTINUOUS_ISR_IRAM_SAFE 1 
#define CONFIG_GDMA_ISR_IRAM_SAFE 1
```
in sdkconfig.h file in dio_qspi and qio_qspi include directory but with no more success.

In Kconfig.projbuild, I can read this paragraph:
```
config ARDUINO_ISR_IRAM
    bool "Run interrupts in IRAM"
    default "n"
    help
        Enabling this option will Attach all interrupts with the IRAm flag.
        It will also make some HAL function, like, digitalRead/Write and more
				be loaded into IRAM for access inside ISRs.
				Beware that this is a very dangerous setting. Enable it only if you
				are fully aware of the consequences.
```
I need a way to change the default "n" to "y". But how, that's the big question !

jantje, 2024-11-22:
>I need a way to change the default "n" to "y". But how, that's the big question !

That is a question for the esp community. If you know how to do it in the Arduino IDE we can see how to get it working in Sloeber.

GitHubLionel, 2024-11-23:
Ok, I found the response in [the FAQ of Espressif](https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/faq.html).
And this paragraph say that modify sdkconfig.h files is without effect:
`Note that modifying sdkconfig or sdkconfig.h files found in the arduino-esp32 project tree does not result in changes to these options. This is because ESP-IDF libraries are included into the arduino-esp32 project tree as pre-built libraries.`
So no way without ESP-IDF.

jantje, 2024-11-23:
>This is because ESP-IDF libraries are included into the arduino-esp32 project tree as pre-built libraries.

So the questions are:
1) Is the variant you want delivered/available (Third party library or so) in the arduino framework?
2) and if not: can you build the libraries from source in the arduino framework?

GitHubLionel, 2024-11-23:
>     1. Is the variant you want delivered/available (Third party library or so) in the arduino framework?
Not in the Arduino framework. In fact, it is strange that continuous ADC is provided but is unusable because the directive `ARDUINO_ISR_IRAM` is not activated.

>     2. and if not: can you build the libraries from source in the arduino framework?
I don't think so. There are some tools but it is out of my knowledge. We really need to use the "menuconfig" of ESP-IDF.

GitHubLionel, 2024-11-23:
Yes, not a Sloeber issue. Maybe a future enhancement but probably had to code ;)

closed 2024-11-23 by GitHubLionel