Sloeber/arduino-eclipse-plugin #1692: Linking fails with STM32Duino platform >2.7.1 when USB CDC is used

arthurfprecht, 2024-11-25:
**Describe your environment**   
Sloeber roduct 4.4.3   
OS: Linux Mint 22
Platform and Arduino configurations on section below

**Describe the bug**   
Linking fails with STM32Duino platform >2.7.1, due to changes in the USB CDC code structure when STM32duino updated from 2.7.1 to 2.8.0

**To Reproduce**   
Steps to reproduce the behavior:   
1. Download Sloeber Product 4.4.3
2. Add Platform from STM32Duino version >2.7.1
3. Try any code with USB Serial (USB support to CDC gerenic Serial Supersede USART)
4. Add the srcWrapper library and try to compile (Errors during linking will occur)
Snippets of the errors:
`Starting combiner`
`blink_f401.ino:10:(.text.setup+0x24): undefined reference to SerialUSB'`
`STMicroelectronics/hardware/stm32/2.8.0/libraries/SrcWrapper/src/stm32/hw_config.c:79:(.text.hw_config_init+0x20): undefined reference to USBD_CDC_init'`
`STMicroelectronics/hardware/stm32/2.8.0/cores/arduino/WSerial.cpp:66:(.text._Z14serialEventRunv+0x18): undefined reference to USBSerial::available()'`
5. (workaround attempt) Add USBDevice library, compile and run (USB Serial will not work)

platform:
[Arduino Core STM32](https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json)

screenshot of project properties->arduino:
[image]

**Expected behavior**   
The linking should work (i.e. the library should be added automatically) and no workaround (manually adding USBDevice.h) should be employed

**Data to back up your claim**   
Even if the workaround of manually adding the USBDevice.h causes the code to compile, the Serial does not work, as explained in a [similar issue that happened on Platformio](https://github.com/platformio/platform-ststm32/issues/793#issuecomment-2263282395)

jantje, 2024-11-27:
Thanks for the issue and the workaround.

I tried to reproduce this and I have very similar behavior in Sloeber V5 and stmDuino version 2.8.1
The only difference were: in step 4
1 )srcWrapper was automagically added after the first build.
2) I got a different error message during linking but also USB related.

Unfortunately the platformio issue does not contain any info on how they fixed it.

arthurfprecht, 2024-11-30:
Hello jantje, first thank you very much for replicating and adding the issue as documented. 

Today I tested a little more and I think I had another issue with the Serial happening alongside this one from USBDevice: 

The serial monitor view unfortunately is not working on my setup (described in the original post). 
If I use this very simple sketch and compile it with package version 2.7.1:
```cpp
#include "Arduino.h"

uint32_t lastBlink = 0;
bool previousLED = 0;

void setup()
{
	pinMode(LED_BUILTIN, OUTPUT);
	lastBlink = millis();
	Serial.begin(115200);
	while (!Serial);
}

void loop()
{
	if (millis()-lastBlink > 1000)
	{
		lastBlink = millis();
		previousLED = !previousLED;
		digitalWrite(LED_BUILTIN, previousLED);
		Serial.println("Something to test the serial\n");
	}
}
```

Then I upload and open the Serial monitor view, I get nothing:
[image]
But using other COM port visualizer, like minicom or picocom, I get the output: 
[image]

I went ahead and tested the newest versions. Applying the workaround (Adding USBDevice.h manually) is confirmed to work! :smile: 

Though now I need to figure out why the Serial monitor view is not working 🥴

jantje, 2024-11-30:
Strange that the serial is not working.

jantje, 2026-01-05:
To fully compare one also needs to look at the compile options (how many times I have added a option after plenty of hours looking for why things didn't work; forgot and made a new project, ....)
To avoid the extra work I started adding stuff like below to my code
```
#ifndef I_USE_GPS
#error define I_USE_GPS is missing
#endif
```
Also note that V4 and V5 are serious different beasts and there may be stuff working in V4 which does not work in V5 .

PerennialNovice, 2026-01-07:
Compile options has two differences:
Warning Level None vs All
and alternative size command Raw vs Arduino

jantje, 2026-01-07:
Those compile options should not result in differences on linking or USB-Serial is not working.

PerennialNovice, 2026-10-20:
Just dropping by to say that in newest 5.0.0.202610191859 and STM32duino 3.0.0 USBDevice is still not getting included when USB CDC is activated in the Arduino board selection menu.
Workaround to manually add still works