Sloeber/arduino-eclipse-plugin #1693: V5.0 testing C++ Perspective cannot be extended - esp. with build Configuration

ArminJo, 2024-11-28:
Is it intended to not support build configurations?
I enabled it here
[image]
but it gets unchecked after I apply and close the perspective setting window and open it again :-(

jantje, 2024-11-28:
This is a part of Eclipse CDT I didn't even know it existed.
So nothing is intended from my part.

ArminJo, 2024-11-28:
So I have hope, that this will be fixed in the future 😀.
For now I can test without it, but as you may remeber I use the configuration heavily for "production".
[image]

jantje, 2024-11-28:
V5 is supposed to be way better on behalf of configurations (especially config changes)
But I have no clue what is missing ;-)
I'll take a look.

jantje, 2024-11-28:
I've taken a look and I do not understand why configuration is not visible.
As a workaround you can select the c/c++ perspective to start from.
There is no real difference.

jantje, 2024-11-28:
Strange ... I went to the C/C++ perspective activated the "build configurations" played around went back to sloeber perspective and there is was.
[image]

jantje, 2024-12-14:
This clearly seems to be an eclipse issue that I can not be fix nor do I know who could.

ArminJo, 2024-12-15:
With CDT (also latest updates) I have no such problems...

jantje, 2024-12-15:
I understand but Build configuration is CDT stuff.
It looks as if part of CDT is not loaded and after looking at in in CDT perspective it gets loaded and is available.
(Eclipse uses lazy loading which mean stuff is only loaded when needed; it might very well be that this CDT gui part is not loaded when Sloeber starts and does not reference these perspective contributors)

closed 2025-03-31 by jantje