Sloeber/arduino-eclipse-plugin #1694: Sloeber V5 testing Windows

TrippyLighting, 2024-11-30:
Using the latest version of the "nightly" from [here](https://github.com/Sloeber/arduino-eclipse-plugin/actions/runs/12001163239):

- Windows 11 home 23H2. Ryzen 5950X 64Gb. 
- I also tested on a 2017 Intel MacBook Pro running macOS Ventura 13.7.1. 

Both share some UI observations/issues, which I report here. 
I'll create a separate issue for compilation/upload issues.

I tried to compile two different programs. 

1. One is an example of the [Freenove IR library for ESP 32](https://github.com/Freenove/Freenove_IR_Lib_for_ESP32). This uses the RMT hardware on the ESP32-S3 under test. It only works with the ESP IDF 5. As such, it does not compile using Sloeber 4.3.3. I did not use the .ino example but a .cpp and .h file. This was compiled and uploaded correctly with the Windows and macOS versions. Feedback from the board on received IR signals was reported through the serial monitor.

[image]

**Issues on both OS:**
1. After closing and restarting Sloeber, the Console window is blank. The tab is present but is missing UI elements. 

[image]

Closing that ab and selecting "Window->Show View->Other->Console brings up the console tab with the UI elements.

[image]

2. Sloeber "drops" the library selection when it is closed/restarted.
 The libraries are still in the project folder, but the previous selection when adding libraries to a project is lost.

The second application I tried to compile is my own code. This compiles and uploads with Sloeber 4.3.3 using ESP32 2.0.14 and 2.0.17.
2.0.14 was my "control" for this experiment as much of the application was developed with Sloeber 4.1.1 (Windows).

If it ain't broken don't fix it, so why am I trying to upgrade to Sloeber V5? 

I transitioned from using Teensy LC and 3.2 boards to ESP32-S3 (and perhaps ESP32-C6) due to the included WiFi and mDNS (Bonjour) abilities. The application allows me to remote control devices through OSC and TouchOSC. Now I want to IR control back to the application.

The app uses a bubnch of libraries:

TrippyLighting, 2024-11-30:
On macOS Slobber appears to have a problem indexing /discovering library content:
[image]

Compiling the code then results in this console output, which ends in 0 errors and 0 warnings found, but obvouldsy isnt a successful build. 

10:39:16 **** Incremental Build of configuration Release for project IR_Receive_ESP32_S3 ****
Arduino Toolchain (Sloeber edition)
Info: Internal Builder is used for build
1
/usr/bin/env bash -c "[ ! -f "/Users/peter/Documents/sloeber-workspace/IR_Receive_ESP32_S3/src"/partitions.csv ] || cp -f "/Users/peter/Documents/sloeber-workspace/IR_Receive_ESP32_S3/src"/partitions.csv "/Users/peter/Documents/sloeber-workspace/IR_Receive_ESP32_S3/bin/Release"/partitions.csv"
2
/usr/bin/env bash -c "[ -f "/Users/peter/Documents/sloeber-workspace/IR_Receive_ESP32_S3/bin/Release"/partitions.csv ] || [ ! -f "/Applications/Sloeber5.app/Contents/Eclipse/arduinoPlugin/packages/esp32/hardware/esp32/3.0.7/variants/esp32s3"/partitions.csv ] || cp "/Applications/Sloeber5.app/Contents/Eclipse/arduinoPlugin/packages/esp32/hardware/esp32/3.0.7/variants/esp32s3"/partitions.csv "/Users/peter/Documents/sloeber-workspace/IR_Receive_ESP32_S3/bin/Release"/partitions.csv"
3
/usr/bin/env bash -c "[ -f "/Users/peter/Documents/sloeber-workspace/IR_Receive_ESP32_S3/bin/Release"/partitions.csv ] || cp "/Applications/Sloeber5.app/Contents/Eclipse/arduinoPlugin/packages/esp32/hardware/esp32/3.0.7"/tools/partitions/default.csv "/Users/peter/Documents/sloeber-workspace/IR_Receive_ESP32_S3/bin/Release"/partitions.csv"
4
/usr/bin/env bash -c "[ -f "/Users/peter/Documents/sloeber-workspace/IR_Receive_ESP32_S3/src"/bootloader.bin ] && cp -f "/Users/peter/Documents/sloeber-workspace/IR_Receive_ESP32_S3/src"/bootloader.bin "/Users/peter/Documents/sloeber-workspace/IR_Receive_ESP32_S3/bin/Release"/IR_Receive_ESP32_S3.bootloader.bin || ( [ -f "/Applications/Sloeber5.app/Contents/Eclipse/arduinoPlugin/packages/esp32/hardware/esp32/3.0.7/variants/esp32s3"/bootloader.bin ] && cp "/Applications/Sloeber5.app/Contents/Eclipse/arduinoPlugin/packages/esp32/hardware/esp32/3.0.7/variants/esp32s3"/bootloader.bin "/Users/peter/Documents/sloeber-workspace/IR_Receive_ESP32_S3/bin/Release"/IR_Receive_ESP32_S3.bootloader.bin || "/Applications/Sloeber5.app/Contents/Eclipse/arduinoPlugin/packages/esp32/tools/esptool_py/4.6"/esptool --chip esp32s3 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o "/Users/peter/Documents/sloeber-workspace/IR_Receive_ESP32_S3/bin/Release"/IR_Receive_ESP32_S3.bootloader.bin "/Applications/Sloeber5.app/Contents/Eclipse/arduinoPlugin/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32s3"/bin/bootloader_qio_80m.elf )"
esptool.py v4.6
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
5
/usr/bin/env bash -c "[ ! -f "/Users/peter/Documents/sloeber-workspace/IR_Receive_ESP32_S3/src"/build_opt.h ] || cp -f "/Users/peter/Documents/sloeber-workspace/IR_Receive_ESP32_S3/src"/build_opt.h "/Users/peter/Documents/sloeber-workspace/IR_Receive_ESP32_S3/bin/Release"/build_opt.h"
6
/usr/bin/env bash -c "[ -f "/Users/peter/Documents/sloeber-workspace/IR_Receive_ESP32_S3/bin/Release"/build_opt.h ] || : > "/Users/peter/Documents/sloeber-workspace/IR_Receive_ESP32_S3/bin/Release"/build_opt.h"
7
/usr/bin/env bash -c ": > '/Users/peter/Documents/sloeber-workspace/IR_Receive_ESP32_S3/bin/Release/file_opts'"
8
/usr/bin/env bash -c "cp -f "/Applications/Sloeber5.app/Contents/Eclipse/arduinoPlugin/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32s3"/sdkconfig "/Users/peter/Documents/sloeber-workspace/IR_Receive_ESP32_S3/bin/Release"/sdkconfig"
No need to runStarting C++ compile: Esp.cpp.o
No need to runStarting C++ compile: FirmwareMSC.cpp.o
No need to runStarting C++ compile: FunctionalInterrupt.cpp.o
No need to runStarting C++ compile: HEXBuilder.cpp.o
No need to runStarting C++ compile: HWCDC.cpp.o
No need to runStarting C++ compile: HardwareSerial.cpp.o
No need to runStarting C++ compile: IPAddress.cpp.o
No need to runStarting C++ compile: MD5Builder.cpp.o
No need to runStarting C++ compile: MacAddress.cpp.o
No need to runStarting C++ compile: Print.cpp.o
No need to runStarting C++ compile: SHA1Builder.cpp.o
No need to runStarting C++ compile: Stream.cpp.o
No need to runStarting C++ compile: StreamString.cpp.o
No need to runStarting C++ compile: Tone.cpp.o
No need to runStarting C++ compile: USB.cpp.o
No need to runStarting C++ compile: USBCDC.cpp.o
No need to runStarting C++ compile: USBMSC.cpp.o
No need to runStarting C++ compile: WMath.cpp.o
No need to runStarting C++ compile: WString.cpp.o
No need to runStarting C++ compile: base64.cpp.o
No need to runStarting C++ compile: cbuf.cpp.o
Starting C++ compile: chip-debug-report.cpp.o
"/Applications/Sloeber5.app/Contents/Eclipse/arduinoPlugin/packages/esp32/tools/esp-xs3/2302/bin/xtensa-esp32s3-elf-g++" -MMD -c "@/Applications/Sloeber5.app/Contents/Eclipse/arduinoPlugin/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32s3/flags/cpp_flags" -Wall -Wextra -Os -Werror=return-type -DF_CPU=240000000L -DARDUINO=20302 -DARDUINO_ESP32S3_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD=\"ESP32S3_DEV\" -DARDUINO_VARIANT="esp32s3" -DARDUINO_PARTITION_default -DARDUINO_HOST_OS="macosx" -DARDUINO_FQBN="esp32 esp32 esp32s3 CDCOnBoot=cdc,CPUFreq=240,DebugLevel=verbose,DFUOnBoot=default,EraseFlash=none,EventsCore=1,FlashMode=qio,FlashSize=4M,JTAGAdapter=default,LoopCore=1,MSCOnBoot=default,Partitio [... cut ...]
<command-line>: error: 'hwcdc' was not declared in this scope
<command-line>: note: in definition of macro 'ARDUINO_FQBN'
<command-line>: error: 'ZigbeeMode' was not declared in this scope
<command-line>: note: in definition of macro 'ARDUINO_FQBN'
<command-line>: error: expected primary-expression before 'default'
<command-line>: note: in definition of macro 'ARDUINO_FQBN'

10:39:29 Build Finished. 0 errors, 0 warnings. (took 12s.966ms)

The warnings stem from a file called chip-debug-report.cpp in the ESP32 Arduino core:

[image]

jantje, 2024-11-30:
I'll take a closer look later but there is a huge difference in Sloeber between V4 and V5 which I think you are not aware of due to this 
[image]
All the folders you marked indeed should not be there (no idea why the are there now; need to look into it)
I have no idea what the applications folder is about.
Most important: there should not be a libraries folder in the root of the project. The libraries should be under arduino/[configname]/libraries

jantje, 2024-12-01:
>After closing and restarting Sloeber, the Console window is blank. The tab is present but is missing UI elements.

I can not reproduce this.

>Sloeber "drops" the library selection when it is closed/restarted.
The libraries are still in the project folder, but the previous selection when adding libraries to a project is lost.

Confirmed. I created issue #1695

>If it ain't broken don't fix it, so why am I trying to upgrade to Sloeber V5?

Well ESP32 support should be way better in V5 especially the indexer. 😉

>Although all of them seem to be properly included/indexed, Sloeber has discovery troubles with "stuff":

Can you provide the actual error? (problem view or hover should do the trick)

>The console shows a successful build that shows way too little feedback :

The console is lying to you. 😉
This is the dreaded file opts problem.
[image]
I could reproduce it here and it leads to one of my preferred issues https://github.com/Sloeber/arduino-eclipse-plugin/issues/1582#issuecomment-1836310745.
Note that the word "jankiness" comes from Arduino itself https://github.com/arduino/arduino-cli/issues/2369#issuecomment-1782464539
So this is a ESP32 issue (and they love creating special things without regard for existing things -absolutely different from pjrc-)

>Switching from ESP32 2.0.14 to 2.0.17 then reports a whole slew of errors:

The esp32 issue has only been fixed in 3.0 so 2.0.17 will fail for the same reason. (There seems to be a github issue (at the moment?) and I can't see the image right now)

>Also, Sloeber creates folders in the project I don't think should be present:

Apart from the Release folder which I have seen occasionally appear (racing conditions I think) I have no clue why these folder appeared. Are you sure this is not "old stuff"?

>On macOS Slobber appears to have a problem indexing /discovering library content:

The mac issue has something to do with quoting strings on the command line.
In the command line you see -DARDUINO_HOST_OS="macosx"
Then you get the error
```
report.cpp:211:52: note: in expansion of macro 'ARDUINO_HOST_OS'
211 | chip_report_printf(" Compile Host OS : %s\n", ARDUINO_HOST_OS);
```
The intention is that the " are in the expansion but apparently they are not.
The same goes for `DARDUINO_VARIANT="esp32s3"`
This should not be tricky as one can add the quotes in code but esp32 .... https://www.guyrutenberg.com/2008/12/20/expanding-macros-into-string-constants-in-c/
As 3.0.7 works on windows for this (it is in my test pack) this is very likely a mac/linux problem.
I created a issue to have a look on a linux system.

jantje, 2024-12-14:
>Switching from ESP32 2.0.14 to 2.0.17 then reports a whole slew of errors:

This seems to be related to Sloeber adding pde files to sloeber.ino.cpp from the OSC library.

```
#include "D:\arduinoPlugin\libraries\OSC\1.3.7\Applications\Processing\SLIPSerialToUDP\GUI.pde"
#include "D:\arduinoPlugin\libraries\OSC\1.3.7\Applications\Processing\SLIPSerialToUDP\SLIPSerialToUDP.pde"
#include "D:\arduinoPlugin\libraries\OSC\1.3.7\Applications\Processing\SLIPSerialToUDP\Serial.pde"
#include "D:\arduinoPlugin\libraries\OSC\1.3.7\Applications\Processing\SLIPSerialToUDP\UDP.pde"
#include "D:\arduinoPlugin\libraries\OSC\1.3.7\Applications\Processing\UDPReceiveBundle\UDPReceiveBundle.pde"
#include "D:\arduinoPlugin\libraries\OSC\1.3.7\Applications\Processing\UDPReceiveMessage\UDPReceiveMessage.pde"
#include "D:\arduinoPlugin\libraries\OSC\1.3.7\Applications\Processing\serialSend\serialSend.pde"
```

I'll make a new issue for this one.