Sloeber/arduino-eclipse-plugin #1695: V5.0 Sloeber "drops" the library selection when it is closed/restarted.

jantje, 2024-12-01:
The libraries are still in the project folder, but the previous selection when adding libraries to a project is lost.

As reported in issue #1694 by @TrippyLighting

[image]

Looking at this I also see in V5 there is a need for add to all configurations option.
I think the current code only adds to the current configuration.

closed 2024-12-14 by jantje

TrippyLighting, 2025-01-03:
I can confirm that this has been fixed and works as expected on macOS and Windows 11. Thank You!

jantje, 2025-01-03:
Thanks for confirming.

TrippyLighting, 2025-01-18:
On MacOS (I've not tried it on Windows), Sloeber V5 still drops some libraries! I have yet to find a pattern, which might be due to my not knowing how this works under the hood.

It does not drop all of the libraries, just some. I will possibly start to screen record sessions when I compile successfully, close Sloeber, and open it up again so I can share what is happening.