Sloeber/arduino-eclipse-plugin #1696: V5.0 ESP32 3.0.7 macro expansion of boardname does not conttain quotes.

jantje, 2024-12-01:
As reported by @TrippyLighting in issue #1694 

```
No need to runStarting C++ compile: cbuf.cpp.o
Starting C++ compile: chip-debug-report.cpp.o
"/Applications/Sloeber5.app/Contents/Eclipse/arduinoPlugin/packages/esp32/tools/esp-xs3/2302/bin/xtensa-esp32s3-elf-g++" -MMD -c "@/Applications/Sloeber5.app/Contents/Eclipse/arduinoPlugin/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32s3/flags/cpp_flags" -Wall -Wextra -Os -Werror=return-type -DF_CPU=240000000L -DARDUINO=20302 -DARDUINO_ESP32S3_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="ESP32S3_DEV" -DARDUINO_VARIANT="esp32s3" -DARDUINO_PARTITION_default -DARDUINO_HOST_OS="macosx" -DARDUINO_FQBN="esp32 esp32 esp32s3 CDCOnBoot=cdc,CPUFreq=240,DebugLevel=verbose,DFUOnBoot=default,EraseFlash=none,EventsCore=1,FlashMode=qio,FlashSize=4M,JTAGAdapter=default,LoopCore=1,MSCOnBoot=default,PartitionScheme=default,PSRAM=disabled,UploadMode=default,UploadSpeed=921600,USBMode=hwcdc,ZigbeeMode=default" -DESP32=ESP32 -DCORE_DEBUG_LEVEL=5 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=1 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0 "@/Applications/Sloeber5.app/Contents/Eclipse/arduinoPlugin/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32s3/flags/defines" "-I/Users/peter/Documents/sloeber-workspace/IR_Receive_ESP32_S3/src" -iprefix "/Applications/Sloeber5.app/Contents/Eclipse/arduinoPlugin/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32s3/include/" "@/Applications/Sloeber5.app/Contents/Eclipse/arduinoPlugin/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32s3/flags/includes" "-I/Applications/Sloeber5.app/Contents/Eclipse/arduinoPlugin/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32s3/qio_qspi/include" "-I/Applications/Sloeber5.app/Contents/Eclipse/arduinoPlugin/packages/esp32/hardware/esp32/3.0.7/cores/esp32" "-I/Applications/Sloeber5.app/Contents/Eclipse/arduinoPlugin/packages/esp32/hardware/esp32/3.0.7/variants/esp32s3" "-I/Applications/Sloeber5.app/Contents/Eclipse/arduinoPlugin/packages/esp32/hardware/esp32/3.0.7/libraries/ESP32/src" "-I/Users/peter/Documents/Arduino/libraries/Freenove_IR_Lib_for_ESP32/src" -MMD -MP "-MFarduino/Release/core/chip-debug-report.cpp.d" "-MTarduino/Release/core/chip-debug-report.cpp.o" -D__IN_ECLIPSE__=1 "@/Users/peter/Documents/sloeber-workspace/IR_Receive_ESP32_S3/bin/Release/file_opts" "/Applications/Sloeber5.app/Contents/Eclipse/arduinoPlugin/packages/esp32/hardware/esp32/3.0.7/cores/esp32/chip-debug-report.cpp" -o "arduino/Release/core/chip-debug-report.cpp.o"
/Applications/Sloeber5.app/Contents/Eclipse/arduinoPlugin/packages/esp32/hardware/esp32/3.0.7/cores/esp32/chip-debug-report.cpp: In function 'void printSoftwareInfo()':
: error: 'macosx' was not declared in this scope; did you mean 'acosl'?
/Applications/Sloeber5.app/Contents/Eclipse/arduinoPlugin/packages/esp32/hardware/esp32/3.0.7/cores/esp32/chip-debug-report.cpp:211:52: note: in expansion of macro 'ARDUINO_HOST_OS'
211 | chip_report_printf(" Compile Host OS : %s\n", ARDUINO_HOST_OS);
| ^~~~~~~~~~~~~~~
/Applications/Sloeber5.app/Contents/Eclipse/arduinoPlugin/packages/esp32/hardware/esp32/3.0.7/cores/esp32/chip-debug-report.cpp: In function 'void printBoardInfo()':
: error: 'esp32s3' was not declared in this scope
/Applications/Sloeber5.app/Contents/Eclipse/arduinoPlugin/packages/esp32/hardware/esp32/3.0.7/cores/esp32/chip-debug-report.cpp:221:52: note: in expansion of macro 'ARDUINO_VARIANT'
221 | chip_report_printf(" Arduino Variant : %s\n", ARDUINO_VARIANT);
| ^~~~~~~~~~~~~~~
: error: 'esp32' was not declared in this scope
/Applications/Sloeber5.app/Contents/Eclipse/arduinoPlugin/packages/esp32/hardware/esp32/3.0.7/cores/esp32/chip-debug-report.cpp:223:52: note: in expansion of macro 'ARDUINO_FQBN'
223 | chip_report_printf(" Arduino FQBN : %s\n", ARDUINO_FQBN);
| ^~~~~~~~~~~~
: error: 'CPUFreq' was not declared in this scope
: note: in definition of macro 'ARDUINO_FQBN'
```

ARDUINO_VARIANT ARDUINO_FQBN and ARDUINO_HOST_OS seem to be misquoted on the command line on mac (and linux?)

TrippyLighting, 2025-01-03:
I can confirm that this has been fixed and works as expected on macOS. Thank You!

jantje, 2025-01-15:
The fix has some kind of regression on windows.
That is: in the test there was only the uno and now there are leonardo, zero and ESP32S3.
Seems the leonardo and ESP32S3 fail on windows testing.

ESP32S3 error
```
"D:\arduinoPlugin\packages\esp32\tools\esp-x32\2405/bin/xtensa-esp32s3-elf-g++"  -MMD -c "@D:\arduinoPlugin\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-cfea4f7c-v1\esp32s3/flags/cpp_flags" -Wall -Wextra -Os -Werror=return-type -DF_CPU=240000000L -DARDUINO=20302 -DARDUINO_ESP32S3_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD=\"ESP32S3_DEV\" -DARDUINO_VARIANT="esp32s3" -DARDUINO_PARTITION_default -DARDUINO_HOST_OS="windows" -DARDUINO_FQBN="esp32:esp32:esp32s3:CDCOnBoot=default,CPUFreq=240,DebugLevel=none,DFUOnBoot=default,EraseFlash=none,EventsCore=1,FlashMode=qio,FlashSize=4M,JTAGAdapter=default,LoopCore=1,MSCOnBoot=default,PartitionScheme=default,PSRAM=disabled,UploadMode=default,UploadSpeed=921600,USBMode=hwcdc,ZigbeeMode=default" -DESP32=ESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1  -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=0 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0  "@D:\arduinoPlugin\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-cfea4f7c-v1\esp32s3/flags/defines" "-IR:\RegressionTest\issue1047_ESP32S3_Board_Names_Can_Be_used_as_Strings\src" -iprefix "D:\arduinoPlugin\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-cfea4f7c-v1\esp32s3/include/" "@D:\arduinoPlugin\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-cfea4f7c-v1\esp32s3/flags/includes" "-ID:\arduinoPlugin\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-cfea4f7c-v1\esp32s3/qio_qspi/include" "-ID:\arduinoPlugin\packages\esp32\hardware\esp32\3.1.1\variants\esp32s3" "-ID:\arduinoPlugin\packages\esp32\hardware\esp32\3.1.1\cores\esp32" -MMD -MP "-MFsrc\sloeber.ino.cpp.d" "-MTsrc\sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 "@R:\RegressionTest\issue1047_ESP32S3_Board_Names_Can_Be_used_as_Strings\bin\Release/file_opts" "..\..\src\sloeber.ino.cpp" -o "src\sloeber.ino.cpp.o"     
<command-line>: error: 'windows' was not declared in this scope
R:\RegressionTest\issue1047_ESP32S3_Board_Names_Can_Be_used_as_Strings\src\issue1047_ESP32S3_Board_Names_Can_Be_used_as_Strings.ino:13:18: note: in expansion of macro 'ARDUINO_HOST_OS'
   13 | char mychar4[] = ARDUINO_HOST_OS;
      |                  ^~~~~~~~~~~~~~~
<command-line>: error: 'esp32s3' was not declared in this scope
R:\RegressionTest\issue1047_ESP32S3_Board_Names_Can_Be_used_as_Strings\src\issue1047_ESP32S3_Board_Names_Can_Be_used_as_Strings.ino:16:18: note: in expansion of macro 'ARDUINO_VARIANT'
   16 | char mychar5[] = ARDUINO_VARIANT;
```

leonardo error
```
"D:\arduinoPlugin\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=20302 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 "-DUSB_MANUFACTURER="Unknown"" "-DUSB_PRODUCT=\"Arduino Leonardo\"" "-ID:\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\cores\arduino" "-ID:\arduinoPlugin\packages\arduino\hardware\avr\1.8.6\variants\leonardo" -MMD -MP "-MFsrc\sloeber.ino.cpp.d" "-MTsrc\sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 "..\..\src\sloeber.ino.cpp" -o "src\sloeber.ino.cpp.o"     
<command-line>:0:18: error: 'Unknown' was not declared in this scope
R:\RegressionTest\issue1047_Leonardo_Board_Names_Can_Be_used_as_Strings\src\issue1047_Leonardo_Board_Names_Can_Be_used_as_Strings.ino:7:18: note: in expansion of macro 'USB_MANUFACTURER'
 char mychar2[] = USB_MANUFACTURER;
```

Seems the quoting for USB_MANUFACTURER (for leonardo) ARDUINO_HOST_OS and ARDUINO_VARIANT are not correct on windows

jantje, 2025-01-17:
All works on windows now as well

commit cb3f47df (2024-12-28): Modify workaround to fix #1696 on linux

Unfortunately the builddefault ino on all boards fails to run on my
linux box due to out of memory
closed 2025-01-17 by jantje