Sloeber/arduino-eclipse-plugin #1698: STM32 Support for Sloeber?

DocAlex, 2024-12-10:
Hello,

I am using Sloeber frequently for arduino projects. Thank you for the tool.

Now I installed STM32Duino to support STM 32 boards. 

In the Arduino IDE its working perfectly (see image). I can generate code for the board an can upload it via the nucleo ST-link Programmer. 

[image]

I undestood that sloeber uses the Arduino environment and expected to find the same boards with STM32 processor as well. 

unfortunately that is not the case: (see image):

[image]

Did I miss anything? 

Thank you in advance

Alexander

jantje, 2024-12-10:
You will not find the STM32 boards in the arduino AVR boards.txt
[image]
And if you do not find the STM32 boards.txt ... did you install it using the sloeber boards manager?

All that being said; there are some problems with the STM32 boards and the indexer and Sloeber.

PerennialNovice, 2024-12-12:
Have a look over here at the STM32duino project:
[https://github.com/stm32duino/Arduino_Core_STM32](https://github.com/stm32duino/Arduino_Core_STM32)

In sloeber, from the top menubar choose "sloeber" -> "preferences" -> "Third party index urls" and add the url `https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json`
now you can choose STM32 under "sloeber" -> "preferences" -> "platforms and boards"

[image]

and click Apply and Close
have some patience until everything is downloaded and installed and after that you can select the STM32 platform in the project settings

closed 2024-12-15 by jantje