Sloeber/arduino-eclipse-plugin #1701: V5.0 Sloeber wrongfully adds pde from the library OSC to the sloeber.ino.cpp

jantje, 2024-12-14:
As reported in #1694

When including the library OSC (version 1.3.7 in my test) you will find these lines in sloeber.ino.cpp which should not be there.

```
#include "D:\arduinoPlugin\libraries\OSC\1.3.7\Applications\Processing\SLIPSerialToUDP\GUI.pde"
#include "D:\arduinoPlugin\libraries\OSC\1.3.7\Applications\Processing\SLIPSerialToUDP\SLIPSerialToUDP.pde"
#include "D:\arduinoPlugin\libraries\OSC\1.3.7\Applications\Processing\SLIPSerialToUDP\Serial.pde"
#include "D:\arduinoPlugin\libraries\OSC\1.3.7\Applications\Processing\SLIPSerialToUDP\UDP.pde"
#include "D:\arduinoPlugin\libraries\OSC\1.3.7\Applications\Processing\UDPReceiveBundle\UDPReceiveBundle.pde"
#include "D:\arduinoPlugin\libraries\OSC\1.3.7\Applications\Processing\UDPReceiveMessage\UDPReceiveMessage.pde"
#include "D:\arduinoPlugin\libraries\OSC\1.3.7\Applications\Processing\serialSend\serialSend.pde"
```
Due to sloeber thinking these files are part of your sketch other stuff is added to the sloeber.ino.cpp which causes " a whole slew of errors"

Workaround: exclude arduino/[your config probably Release]/libraries/OSC/Applications from the build by right clicking the Applications folder->resource configurations->exclude from build->select all the configs->ok

Note that the OSC library did not cause this problem in Sloeber V4 because Sloeber V4 only processed .ino/.pde files in the root of the project where V5 allows you to have .ino/.pde files anywhere in the project.

jantje, 2024-12-14:
This should be fixed

closed 2024-12-14 by jantje

TrippyLighting, 2025-01-03:
I can confirm that this has been fixed and works as expected. Thank You!

jantje, 2025-01-03:
Thanks for confirming.